espressif/esp-at #2: AT+CIPMODE=1 ..stop sending data with +++ proposal response_ok

ESP32DE, 2017-02-10:
hi
proposal:
if we stop sending data with '+++' in transparent mode ( AT+CIPMODE=1 ) 
there comes just in time no "OK\r\n"

if we can check for an "OK\r\n" 
we do not need wait "1 second" and try for next AT Command
we can check after sending stop signal ( +++ ) for an "OK\r\n" and 
then we can use next AT Command without a "wait"

[image]

best wishes
rudi ;-)

ESP32DE, 2017-02-23:
No interest - ok
have done in my repo
Close it

xcguang, 2017-02-23:
Thank you very much for your advice.We'll consider it,but it needs take some time to do with it.

ESP32DE, 2017-03-15:
you dont want this help and support
be happy with your AT example
we take other manufacturer in future 
for this.

closed 2017-03-15 by ESP32DE