espressif/esp-at #3: register server callback for act on received client data

ESP32DE, 2017-02-10:
hi

how we can create/register the server callback for act on client request/received data?
we get 

+IPD,0,n:xxxxxxxxxx		//	received	n	bytes,	data	=	xxxxxxxxxx
 
now we need a callback for act.

best wishes
rudi ;-)

ESP32DE, 2017-02-23:
No interest - ok
have done in my PC Repo
Close it

xcguang, 2017-02-24:
Thanks for your interest in esp32-at. Now there isn't server callback for  client request/received data in libat-core.a. Maybe you could parse "+IPD" in at_port_write_data,then do what you want to do.

ESP32DE, 2017-02-24:
> Maybe you could parse "+IPD" in at_port_write_data

thank you for your response.
yes this is the thing i do just in time 
( did in esp8266 too after the src was gone to lib )
https://youtu.be/WFo4Fs0p_NY?t=84

comfortable as parse would be,
can we use isr notify and  
can we use pointer to pvParameters / request data

the thing with "parse" is not professional :)
but ok for the most things to do as at processor with connected second µC ( microcontroller ) 

thank you 
best wishes
rudi ;-)

ESP32DE, 2017-03-15:
you do not want this help
so be happy....

closed 2017-03-15 by ESP32DE