espressif/esp-at #4: small typo mistake in AT docu

ESP32DE, 2017-02-12:
4.2.3
Site 19/53

AT+CIPSTA—Sets the IP Address of the ESP32 Station 

Query Command:  
AT+CIPSTA?
Function: to obtain the IP address of the ESP32 SoftAP.

think must renamed 
ESP32 SoftAP -> ESP32 Station
here too:

Set Command:  
AT+CIPSTA=<ip>,[,<gateway>,<netmask>] 
Function: to set the IP address of the ESP32 SoftAP. 

think must renamed 
ESP32 SoftAP -> ESP32 Station

[image]

best wishes
rudi ;-)

ESP32DE, 2017-02-23:
No interest - ok
there a few more mistake things in docu 
but ok - Close it

xcguang, 2017-02-23:
Sorry for the late reply and thanks for your report.We have fixed the mistake,but have not released it on github.

ESP32DE, 2017-03-15:
Pity about the wasted time
Will use my time now more meaningfully

closed 2017-03-15 by ESP32DE