espressif/esp-at #6: Guru Meditation Error: Core  0 panic'ed (abort)

ESP32DE, 2017-02-23:
> I (47) boot: ESP-IDF v2.0-rc1-116-g7dfb1c2 2nd stage bootloader
> I (47) boot: compile time 11:44:00
> I (47) boot: Enabling RNG early entropy source...
> I (67) boot: SPI Speed      : 40MHz
> I (80) boot: SPI Mode       : DIO
> I (92) boot: SPI Flash Size : 4MB
> I (105) boot: Partition Table:
> I (116) boot: ## Label            Usage          Type ST Offset   Length
> I (139) boot:  0 phy_init         RF data          01 01 0000f000 00001000
> I (162) boot:  1 ota_0            OTA app          00 10 00010000 000ea000
> I (185) boot:  2 nvs              WiFi data        01 02 000fa000 00016000
> I (209) boot:  3 ota_1            OTA app          00 11 00110000 000ea000
> I (232) boot:  4 otadata          OTA data         01 00 001fd000 00003000
> I (255) boot: End of partition table
> I (268) boot: Disabling RNG early entropy source...
> I (285) boot: Loading app partition at offset 00010000
> I (1449) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512)
> I (1450) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x19ac8 (105160) map
> I (1466) boot: segment 2: paddr=0x00039ad8 vaddr=0x3ffb0000 size=0x020ec (  8428) load
> I (1496) boot: segment 3: paddr=0x0003bbcc vaddr=0x40080000 size=0x00400 (  1024) load
> I (1520) boot: segment 4: paddr=0x0003bfd4 vaddr=0x40080400 size=0x1aad4 (109268) load
> I (1598) boot: segment 5: paddr=0x00056ab0 vaddr=0x400c0000 size=0x00034 (    52) load
> I (1599) boot: segment 6: paddr=0x00056aec vaddr=0x00000000 size=0x0951c ( 38172)
> I (1615) boot: segment 7: paddr=0x00060010 vaddr=0x400d0018 size=0x78490 (492688) map
> I (1643) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
> I (1665) heap_alloc_caps: At 3FFBA668 len 00025998 (150 KiB): DRAM
> I (1686) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAM
> I (1708) heap_alloc_caps: At 4009AED4 len 0000512C (20 KiB): IRAM
> I (1729) cpu_start: Pro cpu up.
> I (1740) cpu_start: Single core mode
> I (1753) cpu_start: Pro cpu start user code
> I (1814) cpu_start: Starting scheduler on PRO CPU.
> E (1820) uart: uart_wait_tx_done(703): uart driver error
> I (1820) uart: queue free spaces: 10
> heap:227152
> E (1825) wifi: esp_wifi_set_storage 1230 wifi is not init
> W (1831) wifi: wifi nvs_open fail ret=4353
> 
> E (1834) wifi: wifi_init 1191 ret=4353
> assertion "0 && "esp_wifi_init(&cfg)"" failed: file "/home/xcg/workspace/esp_iot_at/esp-at-application/components/esp-at-core/./at_port.c", line 560, function: initialise_wifi
> abort() was called at PC 0x400f732b
> Guru Meditation Error: Core  0 panic'ed (abort)
> 
> Backtrace: 0x40008155:0x3ffbbee0 0x40007d16:0x3ffbbf00 0x401114ff:0x3ffbbf30 0x40111caa:0x3ffbbf60 0x4010cd2c:0x3ffbbfa0 0x4010c85a:0x3ffbbfe0 0x400d0afd:0x3ffbc000
> 
> Rebooting...
> 

> ets Jun  8 2016 00:22:57
> 
> rst:0xc (SW_CPU_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)

xcguang, 2017-02-24:
Thanks for your interest in ESP32.
"esp_wifi_init" has been called in libat_core.a,and you shouldn't call it again.

yake827, 2017-02-24:
I also had this problem, i just discard esp_wifi_init function,and it fine:)

ESP32DE, 2017-02-24:
hi 
thank you for response.

> /home/xcg/workspace/esp_iot_at/esp-at-application/components/esp-at-core/./at_port.c", line 560, function: initialise_wifi

can not check/fix on line 560 it cause its your (lib_src) file ;-)

btw: 
if i use one earlier preview esp-idf version, the boot is ok.

best wishes
rudi ;-)

pcbreflux, 2017-02-24:
Hi Rudi,

can you tell us the esp-idf and xtensa version you successfully use?  
Tried dozen different esp-idf commit versions from jan till feb without success.
(shure the better way is to get a bugfixed version).

have a nice day
pcbreflux

ESP32DE, 2017-02-24:
Hi
not sure i understand right,
do you try to resolve the bug and can not reproduce 
or ist AT example not runing prinzipal?

runing AT example:
i use the esp-idf Commits on Feb 22, 2017 
thats the last before the path comes with Core/Task
Merge branch bugfix/app_tasks_start_on_wrong_cpu into master
https://github.com/espressif/esp-idf/commit/7dfb1c2e971b01f7c8e01307b6bed0a27e5bbdc7 

be sure you use the date right combine of wifi_lib to the esp-idf version.

compiler 
gcc version 5.2.0 (crosstool-NG crosstool-ng-1.22.0-61-gab8375a)

environment: msys32

hope this helps
best wishes
rudi ;-)

pcbreflux, 2017-02-27:
Thanks Rudi,

with the version of the esp-idf I was able to build the running firmware with out the error  
failed: file "/home/xcg/workspace/esp_iot_at/esp-at-application/components/esp-at-core/./at_port.c"

xcguang, 2017-03-04:
Hi Rudy,
We have reproduced the bug,and we will fix it as soon as possible.

gokr, 2017-03-10:
I was also bitten by this, I seemed to get it working by using tagged esp-idf (v2.0-rc1) - but I had to get the latest esptool otherwise flashing fails.

yake827, 2017-03-11:
Hi all, I solved the problem by pointing ESP_IDF to commit 7eaf1c76dc281c37075f8a8dac34cc3e09669f59, I guess the problem is that they add some parameters in the ESP_IDF of esp_wifi, resulting in the failure to compile.

ESP32DE, 2017-03-12:
yes now i am too ;(

> E (1825) wifi: esp_wifi_set_storage 1231 wifi is not init<\r><\n>
> E (1831) wifi: invalid magic number 2, call WIFI_INIT_CONFIG_DEFAULT to init config<\r><\n>
> E (1838) wifi: wifi_init 1230 ret=258<\r><\n>
> assertion "0 && "esp_wifi_init(&cfg)"" failed: file "/home/xcg/workspace/esp_iot_at/esp-at-application/components/esp-at-core/./at_port.c", line 560, function: initialise_wifi<\r><\n>
> abort() was called at PC 0x400f7757<\r><\n>
> Guru Meditation Error: Core 0 panic'ed (abort)<\r><\n>
>  
> 

espressif please:
open the AT lib
we can not develop on it if the AT goes closed to AT LIB
we can not see what is wrong in the line:

> "/home/xcg/workspace/esp_iot_at/esp-at-application/components/esp-at-core/./at_port.c", line 560, function: initialise_wifi

i know, there is no mistake with  initialise_wifi but how this is done, there is a mistake
-> 
> E (1831) wifi: invalid magic number 2, call WIFI_INIT_CONFIG_DEFAULT to init config<\r><\n>
> E (1838) wifi: wifi_init 1230 ret=258<\r><\n>

come on, 
open the lib, there is no secret in it, but we can develop on it better if open!

thank you 
best wishes
rudi ;-)

ESP32DE, 2017-03-15:
@xcguang 

> Hi Rudy,
> We have reproduced the bug,and we will fix it as soon as possible.

what is the problem to fix it not asap :)

we can't use the AT example since there are wifi param added
the light things to fix this asap would be
update the code in AT.c

better would be, you open the AT LIB for opensource
and we can fix it by self and must not wait

i am sry for allways stand still while the things fixex in a LIB

hey - we are in opensource 
not in LIB
:(

best wishes
rudi ;-)

ESP32DE, 2017-03-15:
You are to blame yourself if the attention and interest in it becomes less, just because we get always only libs get.

The same also applies to the wifi lib and bt lib
What is so secret?

open the lib - we are all adults and adults

ESP32DE, 2017-03-15:
honest?
I hate this silence of espressif if there are problems

donny681, 2017-03-15:
@ESP32DE I meet the same question.I have to wait and do nothing.I hope that espressif open the AT LIB for opensource and we can fix it by ourselves.

ESP32DE, 2017-03-15:
@donny681 
thats the same here - 
"Cooking emotions" 
Which can overflow a barrel,
Just always in productive time
When we need to answer

i do not understand this - why this fix need so long time if all espressif developers are in one building.  

@espressif
then open the things if you not can organice well to synchron it
then we can fix it by self.

ESP32DE, 2017-03-15:
Here a Workaround to fix the reboot with newer ESP-IDF Version

```
void at_task_init(void)
{
    uint8_t *version = (uint8_t *)malloc(64);
    esp_at_device_ops_struct esp_at_device_ops = {
        .read_data = at_port_read_data,
        .write_data = at_port_write_data,
        .get_data_length = at_port_get_data_length,
        .wait_write_complete = at_port_wait_write_complete,
    };
    
    esp_at_custom_ops_struct esp_at_custom_ops = {
        .status_callback = at_status_callback,
    };

    at_uart_init();

    sprintf((char*)version, "compile time:%s %s", __DATE__, __TIME__);
    esp_at_device_ops_regist (&esp_at_device_ops);
    esp_at_custom_ops_regist(&esp_at_custom_ops);
	
        // workaround
	// comment this line
	// esp_at_module_init (0, version);
    
    free(version);

    esp_at_custom_cmd_array_regist (at_custom_cmd, sizeof(at_custom_cmd)/sizeof(at_custom_cmd[0]));
    esp_at_port_write_data((uint8_t *)"\r\nready\r\n",strlen("\r\nready\r\n"));

}
```

you can then compile and run without reboot

right - we have then no AT Firmware
but the AT Firmware starts :) 
all other is not important or?
to fix this line we need the LIB in source

This is absolutely worthless
As an example with a lib
Which we can not fix ourselves
Makes this AT Example completely meaningless
When does espressif cooperate with the community at eye level?
Why does it make you feel so depressed?

My barrel is now full.
so my last words here for now: open the libs to source
i will never ask it again espressif, 
but you will loose me then as developer with esp32  in the future.

full of cooking emotion for your not responsing to the developer #1

ESP32DE, 2017-03-15:
i close it now - 
you don't want this help and support
have a good time with your things now.

closed 2017-03-15 by ESP32DE

xcguang, 2017-03-16:
@ESP32DE We have recompiled the AT lib,and fixed some bugs,maybe you can have a try using the latest code.Thanks.

rezasurmar, 2017-03-16:
@xcguang 
It's works, thanks for update.

ESP32DE, 2017-03-17:
@xcguang
AT Example now it runs like it must. thank you

best wishes
rudi ;-)

shirish47, 2017-07-22:
is this issue same ? https://github.com/tuanpmt/esp32-mqtt/issues/11

kmakhil90, 2017-10-31:
Hi,

Is there any working source code for AT firmware in esp32?