espressif/esp-at #8: Failed to startup after fresh flash.

robbym, 2017-03-10:
I am using the "Espressif ESP32 Development Board - Developer Edition" dev board to test it out, and I cloned this repo, the idf repo, and downloaded the toolchain as it says in the instructions. I do a "make flash" and keep the menuconfig as default. It flashes successfully, but when I connect to it via serial it keeps failing and rebooting. The first poweron_reset is due to me pressing the reset button, the rest is it rebooting itself. (Same stuff after the "Rebooting...") Have I done something wrong? Am I supposed to do some configuration voodoo to make this work? If so, can someone point me to the correct docs?
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:3464
load:0x40078000,len:7828
load:0x40080000,len:252
entry 0x40080034
[0;32mI (43) boot: ESP-IDF v2.0-rc1-257-g4745895 2nd stage bootloader[0m
[0;32mI (44) boot: compile time 14:16:11[0m
[0;32mI (44) boot: Enabling RNG early entropy source...[0m
[0;32mI (64) boot: SPI Speed      : 40MHz[0m
[0;32mI (77) boot: SPI Mode       : DIO[0m
[0;32mI (89) boot: SPI Flash Size : 4MB[0m
[0;32mI (102) boot: Partition Table:[0m
[0;32mI (113) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (136) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (159) boot:  1 ota_0            OTA app          00 10 00010000 000ea000[0m
[0;32mI (182) boot:  2 nvs              WiFi data        01 02 000fa000 00016000[0m
[0;32mI (205) boot:  3 ota_1            OTA app          00 11 00110000 000ea000[0m
[0;32mI (229) boot:  4 otadata          OTA data         01 00 001fd000 00003000[0m
[0;32mI (252) boot: End of partition table[0m
[0;32mI (265) boot: Disabling RNG early entropy source...[0m
[0;32mI (282) boot: Loading app partition at offset 00010000[0m
[0;32mI (1449) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512) [0m
[0;32mI (1450) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x19cf8 (105720) map[0m
[0;32mI (1467) boot: segment 2: paddr=0x00039d08 vaddr=0x3ffb0000 size=0x02144 (  8516) load[0m
[0;32mI (1497) boot: segment 3: paddr=0x0003be54 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (1520) boot: segment 4: paddr=0x0003c25c vaddr=0x40080400 size=0x1a70c (108300) load[0m
[0;32mI (1598) boot: segment 5: paddr=0x00056970 vaddr=0x400c0000 size=0x00034 (    52) load[0m
[0;32mI (1599) boot: segment 6: paddr=0x000569ac vaddr=0x00000000 size=0x0965c ( 38492) [0m
[0;32mI (1615) boot: segment 7: paddr=0x00060010 vaddr=0x400d0018 size=0x78e70 (495216) map[0m
[0;32mI (1643) heap_alloc_caps: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1665) heap_alloc_caps: At 3FFBA668 len 00025998 (150 KiB): DRAM[0m
[0;32mI (1686) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAM[0m
[0;32mI (1708) heap_alloc_caps: At 4009AB0C len 000054F4 (21 KiB): IRAM[0m
[0;32mI (1728) cpu_start: Pro cpu up.[0m
[0;32mI (1740) cpu_start: Single core mode[0m
[0;32mI (1753) cpu_start: Pro cpu start user code[0m
[0;32mI (1814) cpu_start: Starting scheduler on PRO CPU.[0m
[0;31mE (1820) uart: uart_wait_tx_done(703): uart driver error[0m
[0;32mI (1820) uart: queue free spaces: 10[0m
heap:227152

E (1825) wifi: esp_wifi_set_storage 1231 wifi is not init
E (1831) wifi: invalid magic number 2, call WIFI_INIT_CONFIG_DEFAULT to init config
E (1838) wifi: wifi_init 1230 ret=258
assertion "0 && "esp_wifi_init(&cfg)"" failed: file "/home/xcg/workspace/esp_iot_at/esp-at-application/components/esp-at-core/./at_port.c", line 560, function: initialise_wifi
abort() was called at PC 0x400f774f
Guru Meditation Error: Core  0 panic'ed (abort)

Backtrace: 0x40008155:0x3ffbbee0 0x40007d16:0x3ffbbf00 0x40111acf:0x3ffbbf30 0x4011227a:0x3ffbbf60 0x4010d288:0x3ffbbfa0 0x4010cdb2:0x3ffbbfe0 0x400d0b2d:0x3ffbc000

Rebooting...
```

yake827, 2017-03-11:
Hi, robbym, I had the same problem, and finally I solved the problem by pointing ESP_IDF to commit 7eaf1c76dc281c37075f8a8dac34cc3e09669f59, I guess the problem is that they add some parameters in the ESP_IDF of esp_wifi, resulting in the failure to compile.

robbym, 2017-03-11:
Hi yake827, pointing to that commit fixed the problem. Thank you!
However, after it boots up, it doesn't respond to AT commands:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:3488
load:0x40078000,len:7548
load:0x40080000,len:252
entry 0x40080034
[0;32mI (43) boot: ESP-IDF v2.0-rc1-82-g7eaf1c7 2nd stage bootloader[0m
[0;32mI (43) boot: compile time 17:51:18[0m
[0;32mI (44) boot: Enabling RNG early entropy source...[0m
[0;32mI (63) boot: SPI Speed      : 40MHz[0m
[0;32mI (76) boot: SPI Mode       : DIO[0m
[0;32mI (88) boot: SPI Flash Size : 4MB[0m
[0;32mI (101) boot: Partition Table:[0m
[0;32mI (112) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (135) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (158) boot:  1 ota_0            OTA app          00 10 00010000 000ea000[0m
[0;32mI (181) boot:  2 nvs              WiFi data        01 02 000fa000 00016000[0m
[0;32mI (205) boot:  3 ota_1            OTA app          00 11 00110000 000ea000[0m
[0;32mI (228) boot:  4 otadata          OTA data         01 00 001fd000 00003000[0m
[0;32mI (251) boot: End of partition table[0m
[0;32mI (264) boot: Disabling RNG early entropy source...[0m
[0;32mI (281) boot: Loading app partition at offset 00010000[0m
[0;32mI (1425) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512) [0m
[0;32mI (1426) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x19a50 (105040) map[0m
[0;32mI (1443) boot: segment 2: paddr=0x00039a60 vaddr=0x3ffb0000 size=0x020f0 (  8432) load[0m
[0;32mI (1473) boot: segment 3: paddr=0x0003bb58 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (1496) boot: segment 4: paddr=0x0003bf60 vaddr=0x40080400 size=0x1d210 (119312) load[0m
[0;32mI (1579) boot: segment 5: paddr=0x00059178 vaddr=0x400c0000 size=0x00034 (    52) load[0m
[0;32mI (1580) boot: segment 6: paddr=0x000591b4 vaddr=0x00000000 size=0x06e54 ( 28244) [0m
[0;32mI (1596) boot: segment 7: paddr=0x00060010 vaddr=0x400d0018 size=0x74d54 (478548) map[0m
[0;32mI (1624) heap_alloc_caps: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1646) heap_alloc_caps: At 3FFBA640 len 000259C0 (150 KiB): DRAM[0m
[0;32mI (1667) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAM[0m
[0;32mI (1689) heap_alloc_caps: At 4009D610 len 000029F0 (10 KiB): IRAM[0m
[0;32mI (1710) cpu_start: Pro cpu up.[0m
[0;32mI (1721) cpu_start: Single core mode[0m
[0;32mI (1734) cpu_start: Pro cpu start user code[0m
[0;31mE (2131) phy_init: PHY data partition validated[0m
[0;32mI (2494) phy: phy_version: 258, Nov 29 2016, 15:51:07, 0, 0[0m
[0;32mI (2495) cpu_start: Starting scheduler on PRO CPU.[0m
[0;31mE (2501) uart: uart_wait_tx_done(703): uart driver error[0m
[0;32mI (2507) uart: queue free spaces: 10[0m
heap:224496

E (2512) wifi: esp_wifi_set_storage 1218 wifi is not init
I (2518) wifi: frc2_timer_task_hdl:3ffc43ac, prio:22, stack:2048
I (2527) wifi: Init lldesc rx mblock:25
I (2527) wifi: Init lldesc rx ampdu len mblock:7
I (2531) wifi: Init lldesc rx ampdu entry mblock:4
I (2535) wifi: pp_task_hdl : 3ffd11bc, prio:23, stack:8192
I (2541) wifi: mode : softAP (24:0a:c4:05:b5:b9)
I (2547) wifi: mode : sta (24:0a:c4:05:b5:b8) + softAP (24:0a:c4:05:b5:b9)
I (2553) wifi: mode : softAP (24:0a:c4:05:b5:b9)
```

Notice the "[0;31mE (2501) uart: uart_wait_tx_done(703): uart driver error[0m".
Is this why? I am obviously able to connect to it via uart, but it throws an error?

yake827, 2017-03-11:
There may be internal conditions are not met, since it can be controlled through uart,  ignore it maybe a good choice. :)

closed 2017-03-11 by robbym

xcguang, 2017-03-16:
@robbym ,We have fixed the bug,and you could try it using the latest code.Thanks.