espressif/esp-at #10: wifi: esp_wifi_set_storage 1231 wifi is not init

gokr, 2017-03-16:
I would like https://github.com/espressif/esp32-at/issues/6 to be read and reopened since it is NOT solved at all. That issue was closed out of irritation. Please try to compile and run this firmware with the latest esp-idf and fix the bug.

xcguang, 2017-03-16:
Hi @gokr ,
We have fix it,please try it again.If you have any problem,please feel free to let us know.Thanks.

gokr, 2017-03-16:
Ok, I will try it, great!

gokr, 2017-03-17:
Ok, now I have tested this and... well, yes, now it works but I have two other problems. First of all, in order to get "ready" I had to make the changes for the UART described here:

https://www.esp32.com/viewtopic.php?f=2&t=1103&hilit=AT+commands&start=40

...then when I do it all and "make monitor" it shows everything up to "ready" but when I type in "AT" etc it doesn't answer anything back.

However, if I then switch IDF_PATH to the old 2.0-rc1 - then it works just fine. So I am guessing something is off with "make monitor" in the submodule esp-idf?

ESP32DE, 2017-03-17:
@gokr
you do not need to switch back to older IDF for run "monitor" 
 
if you use the actually IDF
monitor goes to simple_monitor
and the new monitor now is new

you can use
**make simple_monitor**
cause monitor is new and do just in time not support CR LF after send the command.
https://github.com/espressif/esp-idf/commit/02fdf8271de3a6db2ab9d4d6f023c160c84ebdbe
( scroll down to the comments ) 

@xcguang
AT Example now it runs like it must. thank you

best wishes
rudi ;-)

> $ make monitor
> MONITOR
> --- idf_monitor on COM27 115200 ---
> --- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
> ets Jun 8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> ets Jun 8 2016 00:22:57
> 
> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0x00
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0008,len:8
> load:0x3fff0010,len:3464
> load:0x40078000,len:7828
> load:0x40080000,len:252
> 0x40080000: _WindowOverflow4 at ??:?
> entry 0x40080034
> 0x40080034: _WindowOverflow4 at ??:?
> I (43) boot: ESP-IDF v2.0-rc1-296-g02fdf82 2nd stage bootloader
> I (44) boot: compile time 16:13:37
> I (44) boot: Enabling RNG early entropy source...
> I (64) boot: SPI Speed : 40MHz
> I (77) boot: SPI Mode : DIO
> I (89) boot: SPI Flash Size : 4MB
> I (101) boot: Partition Table:
> I (113) boot: ## Label Usage Type ST Offset Length
> I (135) boot: 0 phy_init RF data 01 01 0000f000 00001000
> I (159) boot: 1 ota_0 OTA app 00 10 00010000 000ea000
> I (182) boot: 2 nvs WiFi data 01 02 000fa000 00016000
> I (205) boot: 3 ota_1 OTA app 00 11 00110000 000ea000
> I (228) boot: 4 otadata OTA data 01 00 001fd000 00003000
> I (252) boot: End of partition table
> I (265) boot: Disabling RNG early entropy source...
> I (282) boot: Loading app partition at offset 00010000
> I (1450) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512)
> I (1450) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x19e78 (106104) map
> I (1467) boot: segment 2: paddr=0x00039e88 vaddr=0x3ffb0000 size=0x02190 ( 8592) load
> I (1497) boot: segment 3: paddr=0x0003c020 vaddr=0x40080000 size=0x00400 ( 1024) load
> 0x40080000: _WindowOverflow4 at ??:?
> I (1520) boot: segment 4: paddr=0x0003c428 vaddr=0x40080400 size=0x1a70c (108300) load
> I (1598) boot: segment 5: paddr=0x00056b3c vaddr=0x400c0000 size=0x00034 ( 52) load
> I (1599) boot: segment 6: paddr=0x00056b78 vaddr=0x00000000 size=0x09490 ( 38032)
> I (1615) boot: segment 7: paddr=0x00060010 vaddr=0x400d0018 size=0x78f08 (495368) map
> 0x400d0018: _stext at ??:?
> I (1642) cpu_start: Pro cpu up.
> I (1653) cpu_start: Single core mode
> I (1667) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
> I (1690) heap_alloc_caps: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM
> I (1711) heap_alloc_caps: At 3FFBA6B0 len 00025950 (150 KiB): DRAM
[... 95 lines cut ...]
> eMbeddedHome 2017
> [0;32mI (1869) uart: queue free spaces: 10[0m
> I (1907) wifi: wifi firmware version: 0b5b6e7
> I (1907) wifi: config NVS flash: enabled
> I (1907) wifi: config nano formating: disabled
> I (1912) wifi: Init dynamic tx buffer num: 32
> I (1913) wifi: wifi driver task: 3ffc3060, prio:23, stack:3584
> I (1918) wifi: Init static rx buffer num: 10
> I (1922) wifi: Init dynamic rx buffer num: 0
> I (1926) wifi: Init rx ampdu len mblock:7
> I (1929) wifi: Init lldesc rx ampdu entry mblock:4
> I (1934) wifi: wifi power manager task: 0x3ffc8408 prio: 21 stack: 2560
> I (1941) wifi: wifi timer task: 3ffc9488, prio:22, stack:3584
> [0;31mE (1946) phy_init: PHY data partition validated[0m
> [0;32mI (1969) phy: phy_version: 329, Feb 22 2017, 15:58:07, 0, 0[0m
> I (1970) wifi: Init ampdu: 1
> I (1970) wifi: Init ampdu: 1
> I (1970) wifi: mode : sta (24:0a:c4:04:5c:8c) + softAP (24:0a:c4:04:5c:8d)
> 
> ready
> AT+GMR
> AT version:0.10.0.0(Mar 16 2017 19:02:43)
> SDK version:v2.0-rc1-296-g02fdf82
> compile time:Mar 17 2017 13:03:48
> OK

gokr, 2017-04-04:
Works

closed 2017-04-04 by gokr