espressif/esp-at #11: Serious bug: Weird printout left in the code inserting debug info into data stream

gokr, 2017-04-04:
We are helping a customer using the AT processor in a hardware product incorporating the ESP32-WROOM module. Things work BUT there is at least one or two serious bugs in the AT-processor.

One bug manifests itself with a printout appearing sometimes in the data stream coming from the AT processor when we use the IP server facility (using the ESP32 as an AP):

"id:%d,Len:%d,dp:%p"

Now... since unfortunately you do not have the source published (this is a perfect case where it would have helped us a LOT) I haven't been able to verify this is from your code until now - I simply realized I could scan the archive library, and I discovered the above in at_ipCmd.o.

Needless to say this royally messes up our communication! :)

There may also be a second issue because at some occasions I have also observed an actual AT command being in the data stream coming from the ESP32. That feels like a buffer bug of some kind.

FayeY, 2017-04-05:
Hi, sorry that we can not duplicate your problem, could you please provide more details about your test steps and AT logs?

FayeY, 2017-04-07:
"id:%d,Len:%d,dp:%p" is a debug log which will be print from UART0 when sending data through network, and AT commands/responses are going through UART1. They are using different hardware cables. Why the "id:%d,Len:%d,dp:%p" will bother you?

gokr, 2017-04-07:
The problem is that we use UART0 for our AT commands, this is configurable using menuconfig!! So it definitely bothers us :)

xcguang, 2017-04-07:
Hi @gokr ，
You can output log via other uart or disable the log output. 
Maybe you need configure it in menuconfig.
```
Component config ---> ESP32-specific ---> UART for console output
```

gokr, 2017-04-19:
Thanks, we disabled the log output entirely and that removed it. Closing this one, but ... if I were you I would mention this as a potential "gotcha" in the README or something.

closed 2017-04-19 by gokr