espressif/esp-at #12: Guru Meditation Error: Core  0 panic'ed (abort)

JLospinoso, 2017-04-19:
Hello,

My ESP32 stopped working; I've tried several of the example scripts. All of them yield the following errors:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:2036
load:0x40078000,len:9988
load:0x40080000,len:252
entry 0x40080034
assertion "state == EntryState::WRITTEN || state == EntryState::EMPTY" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_page.cpp", line 315, function: esp_err_t nvs::Page::eraseEntryAndSpan(size_t)
abort() was called at PC 0x401040ff
Guru Meditation Error: Core  0 panic'ed (abort)

Backtrace: 0x40008155:0x3ffc9740 0x40007d16:0x3ffc9760 0x400f6b65:0x3ffc9790 0x400f7161:0x3ffc97d0 0x400f739a:0x3ffc9840 0x400f7a50:0x3ffc98a0 0x400f6284:0x3ffc9900 0x400f5f42:0x3ffc9950 0x400f5f8e:0x3ffc9970 0x400d1163:0x3ffc9990 0x4010406a:0x3ffc99b0 0x400d9489:0x3ffc99e0

CPU halted.
```

Thanks!

Josh

xcguang, 2017-04-19:
Hi @JLospinoso ,
Can you erase all flash and try it again please? If the crash persists,please let me know,thanks.

JLospinoso, 2017-04-19:
@xcguang Ok thanks, I'll try. I'm using a Sparkfun ESP32 Thing, any ideas on the easiest way to flash it?

xcguang, 2017-04-20:
You can try
```
python /esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset erase_flash
```

rmahajan95, 2017-05-04:
Hi! I got the exact same error. This solution worked for me. 
Thanks @xcguang ! :)

closed 2017-05-24 by FayeY

JLospinoso, 2017-06-14:
I can confirm that this worked for me as well. Any intuition about what might have happened?

masadnawaz, 2017-06-16:
It did not worked for me..can you please help me..??

xcguang, 2017-06-16:
@masadnawaz  Please try v0.10.0.0,the master branch maybe crash because of libat_core.a

masadnawaz, 2017-06-16:
is it because of some problem in the latest github version..??
I am not able to find v0.10.0.0, 
can you suggest me some other solution..?
Thanks..

xcguang, 2017-06-16:
@masadnawaz 
You can run this instruction to clone it
```
git clone -b v0.10.0.0 --recursive https://github.com/espressif/esp32-at.git
```

masadnawaz, 2017-06-16:
remote: Repository not found.
fatal: repository 'https://github.com/espressif/esp32/' not found

xcguang, 2017-06-16:
It is `https://github.com/espressif/esp32-at.git`,not `https://github.com/espressif/esp32/`.

masadnawaz, 2017-06-16:
with the older version, I m not able to even make the bin files, it start giving errors..
the same newer version is running perfectly on my team mate ESP32 IDF..we have match the menuconfig settings as well, but still i am getting the same error..
Thanks for your suggestions BTW..

freedaun, 2018-07-11:
I have this error also, on initializing the NVS in my app:
```
    assertion "state == EntryState::WRITTEN || state == EntryState::EMPTY" failed: file "C:/msys32/home/user/esp
    /esp-idf/components/nvs_flash/src/nvs_page.cpp", line 316, function: esp_err_t nvs::Page::eraseEntryAndSpan(size_t)
    abort() was called at PC 0x400dbc73 on core 0
```
This error is recurrent. Each day it happens after using the part for (development) for several hours. If I let it "cool" it starts working again, but not for a long. The next day it works again for quite some time. 

As said, this error is on my development board, which I have been using for just 6 weeks. In the application I have made sure that the NVS isn't written unnecessarily, so it can't be because of an excessive number of writes. Debug cycles are frequent, but also not that frequent as the debug cycle is close to a minute, and frequently much longer, so say 50x per day. 

So this worries me! This seems to be a temperature effect, and we'll be using the chip at a relatively high temperature (80 degrees).