espressif/esp-at #13: Spurious "SEND OK" in communication

gokr, 2017-04-25:
Since we managed to figure out https://github.com/espressif/esp32-at/issues/11 - things work much better, but there is still an issue with the text "SEND OK" appearing sometimes inside the payload, like this:

`+IPD,0,40:SEND OK ....`

It happens seldomly though, so may be some kind of timing thing. We are fully aware that this is the status response from CIPSEND (page 26 in docs) but we handle those properly and they shouldn't appear inside the payload from IPD. It seems like there are two different threads somehow getting interleaved here.

FayeY, 2017-04-28:
Thanks for your reporting, we will revise it in the future release.

gokr, 2017-04-28:
This makes our current code unstable - since we have no idea when/where this pops up inside the payload. So for us it is fairly important and it is a serious bug. Are you not going to look into it?

FayeY, 2017-04-28:
Hi gokr, we are trying to duplicate it and debug it now. I'll let you know if there is any update.

gokr, 2017-04-28:
Great! I can see if I can get more info from the guy on our side that has been debugging it.

gokr, 2017-04-28:
Just verified - when it shows up it seems to always be first in the payload. But it does not affect the size parameter (as expected) so the payload then ends up "too long" so to speak.

hs1956, 2017-05-09:
I have the same or a very similar problem.
Sending two request directly after another, I get:

```
0,CONNECT

+IPD,0,306:1,CONNECT

+IPD,1,285:GET /value HTTP/1.1
Host: 192.168.4.1:333
User-Agent: Mozilla/5.0 (Android 5.1.1; Tablet; rv:53.0) Gecko/53.0 Firefox/53.0
Accept: */*
Accept-Language: de,de-DE;q=0.8,en-US;q=0.5,en;q=0.3
Accept-Encoding: gzip, deflate
Referer: http://192.168.4.1:333/
Connection: keep-alive

PUT /unblock HTTP/1.1
Host: 192.168.4.1:333
User-Agent: Mozilla/5.0 (Android 5.1.1; Tablet; rv:53.0) Gecko/53.0 Firefox/53.0
Accept: */*
Accept-Language: de,de-DE;q=0.8,en-US;q=0.5,en;q=
Accept-Encoding: gzip, deflate
Referer: http://192.168.4.1:333/
Connection: keep-alive
Content-Length: 0

```
In the file esp-idf/components/driver/uart.c the function uart_tx_all uses a mutex, but probably this function is called twice: first for the IPD header, then for the IPD data, so other messages can be placed between header and data.
As the calling code probably is in components/at/lib/libat_core.a, I cannot analyze the problem further without access to the library source code.

xcguang, 2017-05-09:
Hi @hs1956,we have fixed it, but have not released it on github.

gokr, 2017-05-09:
Ok, good! Hoping for a release then :)

gokr, 2017-05-15:
@xcguang Any reason we can't get this fix in our hands? My customer is waiting for this and they have a deadline to meet.

FayeY, 2017-05-18:
Hi @gokr , we have updated it, please have a try.

closed 2017-05-24 by FayeY