espressif/esp-at #16: crash when power up

Nicholas3388, 2017-05-16:
the log as following:

[0;32mI (1681) cpu_start: Pro cpu up.[0m
[0;32mI (1693) cpu_start: Single core mode[0m
[0;32mI (1707) heap_alloc_caps: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1730) heap_alloc_caps: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM[0m
[0;32mI (1750) heap_alloc_caps: At 3FFBA7D0 len 00025830 (150 KiB): DRAM[0m
[0;32mI (1771) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (1793) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (1814) heap_alloc_caps: At 400936AC len 0000C954 (50 KiB): IRAM[0m
[0;32mI (1835) cpu_start: Pro cpu start user code[0m
[0;32mI (1892) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (1899) uart: queue free spaces: 10[0m
E (1901) wifi: invalid magic number 3f407776, call WIFI_INIT_CONFIG_DEFAULT to init config
E (1906) wifi: wifi_init 1304 ret=258
ESP_ERROR_CHECK failed: esp_err_t 0x102 at 0x4011959f
Guru Meditation Error: Core  0 panic'ed (abort)

Backtrace: 0x40008155:0x3ffafb50 0x40007d06:0x3ffafb70 0x40119d56:0x3ffafbb0 0x40114c6c:0x3ffafbf0 0x4011479e:0x3ffafc30 0x400d0659:0x3ffafc50

Rebooting...
ets Jun  8 2016 00:22:57

The bin files flashed as following:
bootloader.bin --> 0x1000
partitions_at.bin --> 0x8000
phy_init_data.bin --> 0xf000
esp-at.bin --> 0x10000
blank.bin --> 0xfa000
blank.bin --> 0x1fd000

I've try to erase flash using esptool, but it is useless.

FayeY, 2017-05-16:
Hi, ESP32-AT already has an ESP-IDF (NOT the latest one) as submodule, you need to use that one when compiling the project, otherwise, it may have compatibility problems. 
You can follow the [README](https://github.com/espressif/esp32-at) to compile it.

Nicholas3388, 2017-05-16:
@FayeY  Hi, when I clone the submodule and use it to compile, I got the following error:

I (44) boot: ESP-IDF v2.0-rc1-296-g02fdf82 2nd stage bootloader
I (44) boot: compile time 16:58:07
I (44) boot: Enabling RNG early entropy source...
I (64) boot: SPI Speed      : 40MHz
I (77) boot: SPI Mode       : DIO
I (89) boot: SPI Flash Size : 4MB
I (102) boot: Partition Table:
I (113) boot: ## Label            Usage          Type ST Offset   Length
I (136) boot:  0 phy_init         RF data          01 01 0000f000 00001000
I (159) boot:  1 ota_0            OTA app          00 10 00010000 000ea000
I (182) boot:  2 nvs              WiFi data        01 02 000fa000 00016000
I (206) boot:  3 ota_1            OTA app          00 11 00110000 000ea000
I (229) boot:  4 otadata          OTA data         01 00 001fd000 00003000
I (252) boot: End of partition table
I (265) boot: Disabling RNG early entropy source...
I (282) boot: Loading app partition at offset 00110000
E (300) esp_image: image at 0x110000 has invalid magic byte
W (319) esp_image: image at 0x110000 has invalid SPI mode 255
W (339) esp_image: image at 0x110000 has invalid SPI size 15
E (359) boot: Failed to verify app image @ 0x110000 (8194)
user code done

xcguang, 2017-05-16:
Please erase all flash,then download the bin.
Or download at.bin into 0x110000 and  0x10000

closed 2017-05-16 by Nicholas3388