espressif/esp-at #17: [Error] Can't boot

yanyang20069, 2017-05-18:
I didn't change anything. I flash the chip.

I (45) boot: ESP-IDF v2.0-rc1-813-g1ddf40fe 2nd stage bootloader
I (45) boot: compile time 16:08:38
I (46) boot: Enabling RNG early entropy source...
I (63) boot: SPI Speed      : 40MHz
I (76) boot: SPI Mode       : DIO
I (88) boot: SPI Flash Size : 4MB
I (101) boot: Partition Table:
I (112) boot: ## Label            Usage          Type ST Offset   Length
I (135) boot:  0 phy_init         RF data          01 01 0000f000 00001000
I (158) boot:  1 otadata          OTA data         01 00 00010000 00002000
I (181) boot:  2 nvs              WiFi data        01 02 00012000 0000e000
I (204) boot:  3 ble_data         unknown          40 00 00020000 00003000
I (228) boot:  4 ota_0            OTA app          00 10 00100000 00180000
I (251) boot:  5 ota_1            OTA app          00 11 00280000 00180000
I (274) boot: End of partition table
E (287) boot: ota data partition invalid and no factory, can't boot
user code done

The chip can't work. (Using ESP32 DevkitC)
Somebody can help me?

xcguang, 2017-05-18:
Hi @yanyang20069， please see https://github.com/espressif/esp32-at/issues/16

yanyang20069, 2017-05-18:
@xcguang I have tried, didn't work
[image]
Do you have other solution?

xcguang, 2017-05-18:
@yanyang20069 erase all flash,and download esp-at.bin into 0x100000 and 0x280000

yanyang20069, 2017-05-19:
@xcguang Another error comes.
[image]

FayeY, 2017-05-19:
# Compiling and flashing the project
1. You can clone the project into an empty directory using command:
```
git clone --recursive https://github.com/espressif/esp32-at.git
```
  * Note the `--recursive` option! The esp32-at project already has an ESP-IDF as a submodule, which should be used to compile the project, in order to avoid any compatibility issues (e.g. compile failure or running problems, etc.). 
  * If you have already cloned the esp32-at without this option, run another command to get all the submodules:
```shell
git submodule update --init --recursive
```  
2. Set `IDF_PATH` environment variable, for example:
```
export IDF_PATH=/path_of_your_esp32-at/esp-idf
```
3. `rm sdkconfig` to remove the old configuration.
4. Set the latest default configuration by `make defconfig`. Or you can call `make menuconfig` to configure the project.
5. `make flash` to compile the project and download it into the flash.
  * Or you can call `make` to compile it, and follow the printed instructions to download the bin files into flash by yourself.
  * `make print_flash_cmd` can be used to print the addresses of downloading.

yanyang20069, 2017-05-19:
@FayeY I have done all this. See the result that I have posted

FayeY, 2017-05-19:
Hi @yanyang20069 , I have tried it, it can work well.
According your log, you are using an old version ESP-IDF. Please check the Step 1 and Step 2 of the above post.

yanyang20069, 2017-05-19:
Hi, @FayeY ,I send you what i have done.
[image]
[image]
[image]
[image]

FayeY, 2017-05-19:
Hi @yanyang20069 , please download blank.bin to 0x10000.

yanyang20069, 2017-05-19:
Hi, @FayeY thanks a lot, it works

yanyang20069, 2017-05-19:
@FayeY 
[image]

closed 2017-05-21 by FayeY