espressif/esp-at #18: Can't get it to boot properly

gokr, 2017-05-25:
I can get this version to boot:

https://github.com/espressif/esp32-at/commit/80920103245b7dbcb0c27718bbbdf4cbca3ebbf2

...but newer versions seems to have problem booting. I suspect the commit right after - that changes the partition layout. Whatever I do I just don't get a ready prompt, I get the following:
```
gokr@yoda:~/evo/esp32/esp32-at$ make simple_monitor 
--- forcing DTR inactive
--- forcing RTS inactive
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:3464
load:0x40078000,len:7800
load:0x40080000,len:252
entry 0x40080034
�
```
Note the lack of "ready" at the end.

When it works I get this:
```
gokr@yoda:~/evo/esp32/esp32-at$ make simple_monitor 
--- forcing DTR inactive
--- forcing RTS inactive
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:3464
load:0x40078000,len:7800
load:0x40080000,len:252
entry 0x40080034
�
ready
```
The diff for my sdkconfig that works (for the older commit) compared to current `make defconfig` is this:
```
gokr@yoda:~/evo/esp32/esp32-at$ diff sdkconfig.works sdkconfig
81c81
< CONFIG_PARTITION_TABLE_CUSTOM_APP_BIN_OFFSET=0x10000
---
> CONFIG_PARTITION_TABLE_CUSTOM_APP_BIN_OFFSET=0x100000
84c84
< CONFIG_APP_OFFSET=0x10000
---
> CONFIG_APP_OFFSET=0x100000
269,271c269,271
< CONFIG_AT_UART_PORT=0
< CONFIG_AT_UART_PORT_RX_PIN=3
< CONFIG_AT_UART_PORT_TX_PIN=1
---
> CONFIG_AT_UART_PORT=1
> CONFIG_AT_UART_PORT_RX_PIN=16
> CONFIG_AT_UART_PORT_TX_PIN=17
```
Note old offsets obviously, since it works with that older commit. As you can see the only thing I change is to use UART0 instead of UART1.

FayeY, 2017-05-26:
Can you enable the log output (through UART0) in menuconfig for debugging?

FayeY, 2017-05-31:
Hi @gokr ,is this problem solved? 
Or you can check whether the hardware connection is correct:
The ESP32 Board sends AT commands through UART1 by default.
GPIO16 is RXD
GPIO17 is TXD
GPIO14 is RTS
GPIO15 is CTS

gokr, 2017-05-31:
I don't have time to fiddle with this this week - but I can probably take a look later next week. We are fine with the version that works that I mentioned above. The hw connections haven't changed - we have our ESP32s soldered onto cards :) And as I described we don't use UART1, we use UART0.

DeeZeng, 2017-07-04:
I'm using ESP32-DevKitC board.   UART0 is used to connect to PC.
So I changed  UART0 for AT command(default UART1) ,then i can see "ready" printed.
-->  make menuconfig -> Component config -> AT   -> 
  (0) uart port number for AT command
  (3) uart rx ...
  (1) uart tx...

// i don't know whether rts/cts works
  (13)  rts  （short R19/R20  DNI R21/R22 ）
  (15)  cts
  http://esp-idf.readthedocs.io/en/latest/get-started/establish-serial-connection.html

 then  make flash  & uart terminal to see  "ready". 
 but i can't enter comannd to uart terminal.

there is also a RED boot message 
"E (2599) phy_init: PHY data partition validated"
-------------------------------------
@FayeY 
Question 1: is RTS/CTS  must be correctlly wried for uart (I'd short R19/R20  DNI R21/R22 for test,but still can't enter commands )?
Question 2:  is the E (2599) error a issue?

DeeZeng, 2017-07-04:
Nothing happens when I type AT
https://github.com/espressif/esp32-at/issues/24

it seems like  enable "New Line Mode" on the serial tool can solve this problem.
but i didn't  find how to  set this in "New Line Mode" , I checked a "inplicit CR in every LF" option ,but still  the same . 

could anybody tell me how to set  "New Line Mode" in Putty?

thanks a lot

FayeY, 2017-07-04:
Hi @DeeZeng 
Q1. With the latest firmware, there is no limitation of RTS/CTS connection, RTS/CTS to be wired or not is totally depend on you.
Q2. It is not an issue, we will revise that log.
Q3. Maybe you can have a try with this serial tool [CoolTermWin]( http://freeware.the-meiers.org/CoolTermWin.zip)

FayeY, 2017-07-05:
Hi @DeeZeng 
About Q2, did you download phy_init_data.bin into flash?

DeeZeng, 2017-07-05:
Hi @FayeY

 Thank you, CoolTermWin works fine for AT commands

 About Q2.   I use method  from https://github.com/espressif/esp32-at to program flash
     make erase_flash  then make flash 

Do you mean i need use ESP32 DOWNLOAD tool to program phy_init_data.bin?
[image]

After I do this action , uart fails,   I need do "make flash" again to see the uart messages

(I'm a beginner to ESP32, now trying BLE demo)

FayeY, 2017-07-05:
make flash is fine. 
You can just ignore the log "E (2599) phy_init: PHY data partition validated", we will revise it in the future release.

closed 2017-07-05 by FayeY