espressif/esp-at #19: mbedtls stack overflow

temp4gh, 2017-06-06:
I tried to add some functions to AT code,but got the panic.
Log:
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:3464
load:0x40078000,len:7828
load:0x40080000,len:252
0x40080000: _WindowOverflow4 at ??:?
entry 0x40080034
0x40080034: _WindowOverflow4 at ??:?
I (44) boot: ESP-IDF v2.0-rc1-296-g02fdf827 2nd stage bootloader
I (44) boot: compile time 19:16:12
I (45) boot: Enabling RNG early entropy source...
I (65) boot: SPI Speed      : 40MHz
I (78) boot: SPI Mode       : DIO
I (90) boot: SPI Flash Size : 4MB
I (102) boot: Partition Table:
I (114) boot: ## Label            Usage          Type ST Offset   Length
I (136) boot:  0 phy_init         RF data          01 01 0000f000 00001000
I (160) boot:  1 otadata          OTA data         01 00 00010000 00002000
I (183) boot:  2 nvs              WiFi data        01 02 00012000 0000e000
I (206) boot:  3 ble_data         unknown          40 00 00020000 00003000
I (229) boot:  4 ota_0            OTA app          00 10 00100000 00180000
I (253) boot:  5 ota_1            OTA app          00 11 00280000 00180000
I (276) boot: End of partition table
I (289) boot: Disabling RNG early entropy source...
I (306) boot: Loading app partition at offset 00100000
I (1517) boot: segment 0: paddr=0x00100018 vaddr=0x00000000 size=0x0ffe8 ( 65512
)
I (1518) boot: segment 1: paddr=0x00110008 vaddr=0x3f400010 size=0x22108 (139528
) map
I (1534) boot: segment 2: paddr=0x00132118 vaddr=0x3ffb0000 size=0x02178 (  8568
) load
I (1565) boot: segment 3: paddr=0x00134298 vaddr=0x40080000 size=0x00400 (  1024
) load
0x40080000: _WindowOverflow4 at ??:?
I (1588) boot: segment 4: paddr=0x001346a0 vaddr=0x40080400 size=0x1a70c (108300
) load
I (1666) boot: segment 5: paddr=0x0014edb4 vaddr=0x400c0000 size=0x00034 (    52
) load
I (1667) boot: segment 6: paddr=0x0014edf0 vaddr=0x00000000 size=0x01218 (  4632
)
I (1683) boot: segment 7: paddr=0x00150010 vaddr=0x400d0018 size=0x8093c (526652
) map
0x400d0018: _stext at ??:?
I (1709) cpu_start: Pro cpu up.
I (1721) cpu_start: Single core mode
I (1735) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1758) heap_alloc_caps: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM
I (1778) heap_alloc_caps: At 3FFBA6B0 len 00025950 (150 KiB): DRAM
I (1799) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1821) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1842) heap_alloc_caps: At 4009AB0C len 000054F4 (21 KiB): IRAM
I (1863) cpu_start: Pro cpu start user code
I (1924) cpu_start: Starting scheduler on PRO CPU.
I (2166) uart: queue free spaces: 10
I (2168) wifi: wifi firmware version: 0b5b6e7
I (2168) wifi: config NVS flash: enabled
I (2168) wifi: config nano formating: disabled
I (2175) wifi: Init dynamic tx buffer num: 32
I (2176) wifi: wifi driver task: 3ffc2df8, prio:23, stack:3584
I (2181) wifi: Init static rx buffer num: 10
I (2185) wifi: Init dynamic rx buffer num: 0
I (2189) wifi: Init rx ampdu len mblock:7
I (2193) wifi: Init lldesc rx ampdu entry mblock:4
I (2197) wifi: wifi power manager task: 0x3ffc81a0 prio: 21 stack: 2560
I (2204) wifi: wifi timer task: 3ffc9220, prio:22, stack:3584
E (2209) phy_init: PHY data partition validated
I (2228) phy: phy_version: 329, Feb 22 2017, 15:58:07, 0, 0
I (2229) wifi: Init ampdu: 1
I (2229) wifi: mode : sta (24:0a:c4:05:8f:6c)
I (2231) wifi: Init ampdu: 1
I (2233) wifi: mode : sta (24:0a:c4:05:8f:6c) + softAP (24:0a:c4:05:8f:6d)
I (2242) wifi: mode : sta (24:0a:c4:05:8f:6c)
I (2365) wifi: n:1 1, o:1 0, ap:255 255, sta:1 1, prof:1
I (3022) wifi: state: init -> auth (b0)
I (3025) wifi: state: auth -> assoc (0)
I (3030) wifi: state: assoc -> run (10)
I (3044) wifi: connected with TP-LINK_M, channel 1
I (4823) event: ip: 10.0.0.116, mask: 255.255.255.0, gw: 10.0.0.1
I (13030) wifi: pm start, type:0

http get...
I (31200) example: Seeding the random number generator
I (31201) example: Loading the CA root certificate...
I (31204) example: Setting hostname for TLS session...
I (31207) example: Setting up the SSL/TLS structure...
I (31215) example: Connecting to www.howsmyssl.com:443...
I (32488) example: Connected.
I (32488) example: Performing the SSL/TLS handshake...
I (32488) mbedtls: ssl_tls.c:6306 => handshake

I (32492) mbedtls: ssl_cli.c:3254 client state: 0

I (32497) mbedtls: ssl_tls.c:2430 => flush output

I (32503) mbedtls: ssl_tls.c:2442 <= flush output

I (32508) mbedtls: ssl_cli.c:3254 client state: 1

I (32514) mbedtls: ssl_tls.c:2430 => flush output

I (32519) mbedtls: ssl_tls.c:2442 <= flush output

[0;32***ERROR*** A stack overflow in task at_process_task has been detected.
abort() was called at PC 0x40085c4c
0x40085c4c: vApplicationStackOverflowHook at F:/Code/ESP32/20170531/esp32-at/esp
-idf/components/esp32/panic.c:444
Guru Meditation Error: Core  0 panic'ed (abort)

Backtrace: 0x400824f0:0x3ffc9bf0 0x40082679:0x3ffc9c10 0x40082c46:0x3ffc9c30 0x4
0084a9b:0x3ffc9c50 0x40084a55:0x3ffc9c40
0x400824f0: uart_tx_char at F:/Code/ESP32/20170531/esp32-at/esp-idf/components/v
fs/vfs_uart.c:43 (discriminator 1)
0x40082679: uart_write at F:/Code/ESP32/20170531/esp32-at/esp-idf/components/vfs
/vfs_uart.c:61 (discriminator 2)
0x40082c46: xPortGetCoreID at F:/Code/ESP32/20170531/esp32-at/esp-idf/components
/freertos/tasks.c:4425
 (inlined by) vTaskSwitchContext at F:/Code/ESP32/20170531/esp32-at/esp-idf/comp
onents/freertos/tasks.c:2732
0x40084a9b: _frxt_dispatch at ??:?
0x40084a55: _frxt_int_exit at ??:?

Rebooting...

I dont know how to increase the stack size of mbedtls.Can someone provide any suggestions?Thanx!

FayeY, 2017-06-06:
Sorry that changing stack size is NOT supported now. We may add this new feature in the future. Sorry for the inconvenience, and thanks for your interest in ESP32!

xcguang, 2017-08-05:
Hi,maybe you can create a task to do it.

closed 2018-03-12 by xcguang