espressif/esp-at #22: +IPD design issue

kuldeepdhaka, 2017-06-12:
+IPD — Receives Network Data
this is a command (including packet-data) send by ESP32 in which the other side uC (or whatever) have no control of when will be send.

Since this is async (uC cannot control), this could cause problem to uC when comes at unexpected time.

The number of bytes that will be transmitted also isnt controlled at all by the uC.

Where as ESP32 AT+CIPSEND command have provision on number of bytes that can be transmitted (2KB max) and ESP32 get to send '>' to tell when it is ready to enter packet acceptance stage.

Due to the poor AT command design (+IPD), it will not be possible to use DMA properly with UART in uC side.

It would have been better that, the +IPD should have been similar to AT+CIPSEND.  the uC would poll at interval to fetch data.

something like would be better:
- uC send `AT+IPD=<link ID>,<len>`
- ESP32 send `<available-length>[,<remote IP>,<remote port>]`
- uC send `>`
- ESP32 send the `<packet data bytes....>` 

Though this is a complex negotiation but far better to use for uC interrupt driven processing and when bulk data is there uC can utilized DMA too.

MaJerle, 2017-07-14:
In general, with current (poor) approach, MCU DMA configuration is hard, but newer MCUs support IDLE line detection where you can also handle DMA by software.

In general, ESP should notify you than you have available data, specially when you are in server mode!

- ESP sends `+IPDREADY=<link ID>,<len>[,<remote IP>,<remote PORT>]
- uC send `AT+IPD=<link ID>,<desired len>`
- ESP response with `actual data here...`

In this case, MCU is aware that we have available data to read and will read it when need. If ESP got buffer overflow internally, this is then user's problem for slow read of data.

xcguang, 2017-07-15:
Hi,
The official demo code maybe cannot meet all requirements.
But I think it can be achieved by modifying at_port_write_data and customizing AT command in at_task.c. You can check the prefix `+IPD:` in at_port_write_data and put the date into  a buffer, The add a command like  `AT+IPD=<link ID>,<desired len>` to read it.

MaJerle, 2017-07-15:
There is a room for ESP improvements. I think Espressif should add this feature just like it is implemented in Sim800/900 where you have option to choose how to work.

akotowski, 2017-12-06:
I agree, for AT mode to be robust  it needs to support polling by the uc for receiving data.  Asking users to modify the AT code does not help because the purpose of the AT code is to avoid needing to program the 8266.  It would really help me to use the 8266 if this feature were implemented.