espressif/esp-at #24: Nothing happens when I type AT

kaustheta, 2017-06-16:
Hi,
I have the AT commands configured on UART0 and I have the following output: 

rst:0x1 (POWERON_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:3472
load:0x40078000,len:7804
load:0x40080000,len:252
entry 0x40080034
I (43) boot: ESP-IDF v2.0 2nd stage bootloader
I (43) boot: compile time 15:45:12
I (43) boot: Enabling RNG early entropy source...
I (59) boot: SPI Speed      : 40MHz
I (72) boot: SPI Mode       : DIO
I (84) boot: SPI Flash Size : 4MB
I (96) boot: Partition Table:
I (107) boot: ## Label            Usage          Type ST Offset   Length
I (130) boot:  0 phy_init         RF data          01 01 0000f000 00001000
I (153) boot:  1 otadata          OTA data         01 00 00010000 00002000
I (177) boot:  2 nvs              WiFi data        01 02 00012000 0000e000
I (200) boot:  3 ble_data         unknown          40 00 00020000 00003000
I (223) boot:  4 ota_0            OTA app          00 10 00100000 00180000
I (246) boot:  5 ota_1            OTA app          00 11 00280000 00180000
I (270) boot: End of partition table
I (283) boot: Disabling RNG early entropy source...
I (300) boot: Loading app partition at offset 00100000
I (1475) boot: segment 0: paddr=0x00100018 vaddr=0x00000000 size=0x0ffe8 ( 65512) 
I (1476) boot: segment 1: paddr=0x00110008 vaddr=0x3f400010 size=0x1c690 (116368) map
I (1492) boot: segment 2: paddr=0x0012c6a0 vaddr=0x3ffb0000 size=0x0215c (  8540) load
I (1523) boot: segment 3: paddr=0x0012e804 vaddr=0x40080000 size=0x00400 (  1024) load
I (1546) boot: segment 4: paddr=0x0012ec0c vaddr=0x40080400 size=0x1b028 (110632) load
I (1625) boot: segment 5: paddr=0x00149c3c vaddr=0x400c0000 size=0x00064 (   100) load
I (1626) boot: segment 6: paddr=0x00149ca8 vaddr=0x00000000 size=0x06360 ( 25440) 
I (1642) boot: segment 7: paddr=0x00150010 vaddr=0x400d0018 size=0x7a548 (501064) map
I (1670) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1692) heap_alloc_caps: At 3FFBA6C0 len 00025940 (150 KiB): DRAM
I (1713) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAM
I (1735) heap_alloc_caps: At 4009B428 len 00004BD8 (18 KiB): IRAM
I (1755) cpu_start: Pro cpu up.
I (1767) cpu_start: Single core mode
I (1780) cpu_start: Pro cpu start user code
I (1838) cpu_start: Starting scheduler on PRO CPU.
;32mI (2050) uart: queue free spaces: 10
I (2051) wifi: wifi firmware version: c604573
I (2052) wifi: config NVS flash: enabled
I (2052) wifi: config nano formating: disabled
I (2060) wifi: Init dynamic tx buffer num: 32
I (2060) wifi: wifi driver task: 3ffc4dac, prio:23, stack:3584
I (2065) wifi: Init static rx buffer num: 10
I (2069) wifi: Init dynamic rx buffer num: 0
I (2073) wifi: Init rx ampdu len mblock:7
I (2076) wifi: Init lldesc rx ampdu entry mblock:4
I (2081) wifi: wifi power manager task: 0x3ffca154 prio: 21 stack: 2560
I (2087) wifi: wifi timer task: 3ffcb1d4, prio:22, stack:3584
E (2093) phy_init: PHY data partition validated
I (2128) phy: phy_version: 329, Feb 22 2017, 15:58:07, 0, 0
I (2129) wifi: mode : softAP (30:ae:a4:03:6a:d9)
I (2132) wifi: mode : sta (30:ae:a4:03:6a:d8) + softAP (30:ae:a4:03:6a:d9)
I (2136) wifi: mode : softAP (30:ae:a4:03:6a:d9)

ready
AT

But nothing happens when I type AT and hit 'enter'. I tried using TeraTerm and SecureCRT but same results. Can someone please help? 

Thanks,
Kaustubh

FayeY, 2017-06-17:
AT commands end with CR LF (a new line). You should enable "New Line Mode" on the serial tool.

MHEtLive, 2017-06-17:
The AT Commands use u2TXD and  u2RXD?

kaustheta, 2017-06-19:
@FayeY perfect! Works like a charm. Thanks.

closed 2017-06-19 by kaustheta