espressif/esp-at #25: AT+CWQAP doesn't disconnect

AMEuser, 2017-06-29:
The command AT+CWQAP probably doesn't behave correctly. I expect a disconnection from the network. While it seems to disconnect and re-connect immediately after.
If I issue the command AT+CWJAP? it returns that it is connected.
Moreover I don't know what error is 
> +CWJAP:0
> 
> ERROR

Here the commands sequence:

> ready
> AT+CWLAP
> +CWLAP:(4,"Telecom-62716645",-45,"30:91:8f:a8:bc:e5",11)
> +CWLAP:(3,"DIRECT-q4M2070 Series",-52,"32:cd:a7:9f:35:c6",11)
> +CWLAP:(3,"",-66,"1c:b7:2c:83:55:20",2)
> +CWLAP:(3,"urbantrend2",-67,"1e:b7:1c:83:55:20",2)
> +CWLAP:(4,"TNCAPB4DAD1",-72,"c4:ea:1d:b4:da:d1",1)
> +CWLAP:(0,"FVGWiFi",-72,"6e:72:51:49:b9:3c",1)
> +CWLAP:(3,"AC66U",-84,"e0:3f:49:0b:8a:d0",6)
> +CWLAP:(3,"InfoPoint-Palmanova",-85,"ec:08:6b:9f:08:0c",10)
> +CWLAP:(0,"PTFVGPalmanova",-85,"68:72:51:80:7e:5e",11)
> +CWLAP:(3,"Private HDG",-87,"ee:94:f6:42:d8:a4",11)
> +CWLAP:(3,"TIM-27101510",-88,"b4:a5:ef:61:be:0b",1)
> +CWLAP:(0,"hotelaidogi",-88,"ea:94:f6:42:d8:a4",11)
> +CWLAP:(4,"fastweb-pd-cpe-st4233-1",-89,"00:19:70:b4:f5:99",6)
> +CWLAP:(4,"Digicom_AEC3",-89,"00:a0:a2:67:ae:c3",11)
> +CWLAP:(0,"CB-CLUB",-90,"90:f6:52:c2:cb:ca",8)
> +CWLAP:(4,"Demar Caffe Giancarlo",-90,"8c:59:c3:0b:ea:25",11)
> 
> OK
> AT+CWJAP="Telecom-62716645","*******************"
> +CWJAP:1
> 
> ERROR
> AT+CWJAP="Telecom-62716645","*******************"
> +CWJAP:0
> 
> ERROR
> AT+CWJAP="Telecom-62716645","*******************"
> 
> OK
> AT+CWQAP
> 
> OK
> AT+CWJAP?
> +CWJAP:"Telecom-62716645","30:91:8f:a8:bc:e5",11,-44
> 
> OK

And below the debug messages (------ highlights the point where disconnection command is given):

> ets Jun  8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> ets Jun  8 2016 00:22:57
> 
> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0x00
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0008,len:8
> load:0x3fff0010,len:3464
> load:0x40078000,len:7828
> load:0x40080000,len:252
> entry 0x40080034
> I (44) boot: ESP-IDF v2.0-rc1-257-g4745895 2nd stage bootloader
> I (44) boot: compile time 15:21:50
> I (44) boot: Enabling RNG early entropy source...
> I (64) boot: SPI Speed      : 40MHz
> I (77) boot: SPI Mode       : DIO
> I (89) boot: SPI Flash Size : 4MB
> I (102) boot: Partition Table:
> I (113) boot: ## Label            Usage          Type ST Offset   Length
> I (136) boot:  0 phy_init         RF data          01 01 0000f000 00001000
> I (159) boot:  1 ota_0            OTA app          00 10 00010000 000ea000
> I (182) boot:  2 nvs              WiFi data        01 02 000fa000 00016000
> I (205) boot:  3 ota_1            OTA app          00 11 00110000 000ea000
> I (229) boot:  4 otadata          OTA data         01 00 001fd000 00003000
> I (252) boot: End of partition table
> I (265) boot: Disabling RNG early entropy source...
> I (282) boot: Loading app partition at offset 00010000
> I (1449) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512)
> I (1450) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x19c84 (105604) map
> I (1467) boot: segment 2: paddr=0x00039c94 vaddr=0x3ffb0000 size=0x02150 (  8528) load
> I (1497) boot: segment 3: paddr=0x0003bdec vaddr=0x40080000 size=0x00400 (  1024) load
> I (1520) boot: segment 4: paddr=0x0003c1f4 vaddr=0x40080400 size=0x1a70c (108300) load
> I (1598) boot: segment 5: paddr=0x00056908 vaddr=0x400c0000 size=0x00034 (    52) load
> I (1599) boot: segment 6: paddr=0x00056944 vaddr=0x00000000 size=0x096c4 ( 38596)
> I (1615) boot: segment 7: paddr=0x00060010 vaddr=0x400d0018 size=0x78e60 (495200) map
> I (1643) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
> I (1665) heap_alloc_caps: At 3FFBA670 len 00025990 (150 KiB): DRAM
> I (1686) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAM
> I (1708) heap_alloc_caps: At 4009AB0C len 000054F4 (21 KiB): IRAM
> I (1728) cpu_start: Pro cpu up.
> I (1740) cpu_start: Single core mode
> I (1753) cpu_start: Pro cpu start user code
> I (1814) cpu_start: Starting scheduler on PRO CPU.
> I (1824) uart: queue free spaces: 10
> I (1860) wifi: wifi firmware version: 90b1b8b
> I (1861) wifi: config NVS flash: enabled
> I (1861) wifi: config nano formating: disabled
> I (1865) wifi: Init dynamic tx buffer num: 32
> I (1866) wifi: wifi driver task: 3ffc4d78, prio:23, stack:3584
> I (1871) wifi: Init static rx buffer num: 10
> I (1875) wifi: Init dynamic rx buffer num: 0
> I (1879) wifi: Init rx ampdu len mblock:7
> I (1883) wifi: Init lldesc rx ampdu entry mblock:4
> I (1887) wifi: wifi power manager task: 0x3ffca120 prio: 21 stack: 2560
> I (1894) wifi: wifi timer task: 3ffcb1a0, prio:22, stack:3584
> E (1899) phy_init: PHY data partition validated
> 
> I (1937) phy: phy_version: 329, Feb 22 2017, 15:58:07, 0, 0
> I (1938) wifi: Init ampdu: 1
> I (1938) wifi: mode : sta (24:0a:c4:03:f2:28)
> I (1940) wifi: Init ampdu: 1
> I (1941) wifi: mode : sta (24:0a:c4:03:f2:28) + softAP (24:0a:c4:03:f2:29)
> I (1951) wifi: mode : sta (24:0a:c4:03:f2:28)
> I (106752) wifi: n:11 0, o:1 0, ap:255 255, sta:11 0, prof:1
> I (107410) wifi: state: init -> auth (b0)
> I (107412) wifi: state: auth -> assoc (0)
> I (107416) wifi: state: assoc -> run (10)
> I (107446) wifi: connected with Telecom-62716645, channel 11
> I (108811) wifi: state: run -> init (0)
> I (108812) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
> I (113409) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
> I (113409) wifi: state: init -> auth (b0)
> I (113412) wifi: state: auth -> assoc (0)
> I (113416) wifi: state: assoc -> run (10)
> I (113445) wifi: connected with Telecom-62716645, channel 11
> I (114413) event: ip: 192.168.1.211, mask: 255.255.255.0, gw: 192.168.1.254
> I (123416) wifi: pm start, type:0
> 
> ------- here disconnection command AT+CWQAP is given -------
> 
> I (152825) wifi: state: run -> init (0)
> I (152826) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
> I (152827) wifi: pm stop, total sleep time: 0/29409876
> 
> I (152950) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
> I (152950) wifi: state: init -> auth (b0)
> I (152957) wifi: state: auth -> assoc (0)
> I (152961) wifi: state: assoc -> run (10)
> I (152997) wifi: connected with Telecom-62716645, channel 11
> I (154420) event: ip: 192.168.1.211, mask: 255.255.255.0, gw: 192.168.1.254
> I (162962) wifi: pm start, type:0

xcguang, 2017-06-29:
Hi,thanks for your report about the bug,and we have fixed it in the latest version.Please try it again,thanks.

AMEuser, 2017-06-29:
You are right, latest version works. Thank you.

closed 2017-06-29 by AMEuser