espressif/esp-at #27: How to connect  bluetooth to an Andoird phone?

DeeZeng, 2017-07-07:
I'm trying to set up the bluetooth demo.
 I'd read the esp32-at/docs/ESP32_AT_Commands_Set.md (AT COMMANDs for BLE)
   & esp32-at/tools/README.md (GattService.bin generation)
I tried t two ESP32-DevKitC board, one as Server ,another as Client . they can find and connect success

But when I use a Andoird phone to find ESP32 bluetooth server, it can't find it .
I'd installed  : https://dl.espressif.com/dl/EspBlufi-1.0.apk
EspBlufi  can find and connect to ESP32 bluetooth server , but nothing happens while wating at Prepare page.

I try to program the gattservice bin use Custom  partitions_at.csv

[image]
but no factory,can't boot ~

and I try to modify the partitions_two_ota.csv and add
ble_data,0x40,0,0x310000,12K
at last line

[image]
boot ok

then program the server.bin
[image]

boot ok
->
AT+BLEINIT=2
AT+BLEADVSTART
（also tried some commands like 
AT+BLEADVPARAM=50,150,0,0,7,0,0,"AC:C1:EE:96:90:1B"
   before ADV Start)

but the result is the same as not program the server.bin (Andoird phone not find the bluetooth, apk can but nothing happens)

--
Are there steps/things wrong when i try the bluetooth demo(or ignored)?

Could anybody tell me how to connect Bluetooth to Andoird phone?(some documents or links also ok )
And is Bluethooth can be set like wifi can auto cennect at power on?(like BLE auto runing as server and advertising ) 

I'm a beginner to ESP32 and bluetooth, hopen you can help me !  Thanks a lot!

FayeY, 2017-07-08:
ESPBlufi is not for BLE connect. You can have a try with the app "nRF" on your Android phone. Or use app "lightblue" on your apple devices (iphone or MAC).

DeeZeng, 2017-07-10:
Thanks @FayeY 
"nRF" can find the BLE server. 

And I'm trying to use no-AT examples.  Perhaps no-AT examples is more suit for me .

closed 2017-07-10 by FayeY