espressif/esp-at #28: Specify connections for server

MaJerle, 2017-07-14:
When device is in server mode, it can happen that at some point, all connections are used for server and there is no support to use client mode at that time.

What we should have option to set is which connections are available for server.

Correct me if I'm mistaken, but current server approach is like this:

- User enables server at specific port
- ESP creates new connections and listens at specific port
- When connection is accepted, ESP creates new empty connection to listen on same port
- And so on, which means there is always one connection active and ready to accept new connection, except if we reached max connections.

We should be able to limit the number of connections and for example only allow server to use them:

- uC sends `AT+CIPSERVERCONN=1,0`
- uC sends `AT+CIPSERVERCONN=1,1`
- uC sends `AT+CIPSERVERCONN=1,2`

These 3 commands would enable connections `0,1,2` only for server. If ESP got more than 3 connections at a time, they should wait before one of these 3 is available again.

If we then try `AT+CIPSTART=1,...` from uC, we should got an error because connection is reserved for server only! In this case, we can use connections `4,5` for client.

---
This allows us to have better control from uC, specially on devices with small amount of RAM.

xcguang, 2017-07-15:
Hi @MaJerle ,
Your opinion is a very good suggestion，and we will consider it in the next release.
Thanks very much.

MaJerle, 2017-07-15:
Hello @xcguang,

That's good to hear. I will post the same issue request to ESP8266 AT repository.

closed 2019-01-16 by FayeY