espressif/esp-at #30: AT+CIPSERVERMAXCONN

MaJerle, 2017-07-24:
Hello.

In last [commit](https://github.com/espressif/esp32-at/commit/1b1b98443697a28f98d3cfddfa5ba09bc916f2ff) is seen about AT+CIPSERVERMAXCONN. Can you provide more info?

Will it be possible to be server on multiple ports? For example CONN0 port 80, CONN2 port 400, etc?
Will this also work on ESP8266 AT software?

Thanks for response.

xcguang, 2017-07-24:
Hi @MaJerle ,
This command is configured with the maximum connection of the server accept. For example, if AT allows 5 connections, and I send `AT+CIPSERVERMAXCONN=3`, so there are two connection is reserved for client. By the way, there is a bug about this in idf, and we are debugging it. 
I think this satisfies #28. And if you have any question or request,just feel free to let me know.Thanks.

MaJerle, 2017-07-24:
This feature does not satisfie #28 because feature 28 allows you to be master to specify exact server connection you want for purpose (for example connection for port). This is allowed by LwIP and therefore should be implemented as new feature to support server on multiple connections each with different port.

xcguang, 2017-07-29:
Yes,you can set max connection when calling esp_at_module_init().