espressif/esp-at #32: AT+BLESCAN=1 NOT WORK

mxwang101, 2017-08-06:


mxwang101, 2017-08-06:
1) AT+BLESCAN=1 NOT WORK

AT+RST
OK
ready
AT+BLEINIT=1
OK
AT+BLESCAN=1
ERROR

xiewenxiang, 2017-08-07:
@mxwang101 
AT+BLESCAN＝<is_enable>,[\<duration time\>]

1. start scan: 
AT+BLESCAN=1,3 
if the duration time is 0, it will continue to scan
2. stop scan:
AT+BLESCAN=0

mxwang101, 2017-08-07:
Thanks, work!

 

Cloud you help me fix "AT+BLEGATTSSRVCRE, &AT+BLEGATTSSRVSTART, Both no work"? 

 

I did:

 

1 ) Download ' GattServiceExample.bin':

wangxin@wangxin-PC MINGW32 ~/esp/esp32-at/esp32-at/tools
$ esptool --port COM8 write_flash 0x21000 GattServiceExample.bin
esptool.py v2.0.1
Connecting........__
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 0)
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 2704 bytes to 225...
Wrote 2704 bytes (225 compressed) at 0x00021000 in 0.0 seconds (effective 618.0 kbit/s)...
Hash of data verified.

2.       Partition Table:�

�[0;32mI (97) boot: Partition Table:�[0m
�[0;32mI (108) boot: ## Label Usage Type ST Offset Length�[0m
�[0;32mI (130) boot: 0 phy_init RF data 01 01 0000f000 00001000�[0m
�[0;32mI (154) boot: 1 otadata OTA data 01 00 00010000 00002000�[0m
�[0;32mI (177) boot: 2 nvs WiFi data 01 02 00012000 0000e000�[0m
�[0;32mI (200) boot: 3 ble_data unknown 40 01 00021000 00003000�[0m
�[0;32mI (223) boot: 4 server_cert unknown 40 02 00024000 00002000�[0m
�[0;32mI (247) boot: 5 server_key unknown 40 03 00026000 00002000�[0m
�[0;32mI (270) boot: 6 server_ca unknown 40 04 00028000 00002000�[0m
�[0;32mI (293) boot: 7 fatfs Unknown data 01 81 00070000 00090000�[0m
�[0;32mI (316) boot: 8 ota_0 OTA app 00 10 00100000 00180000�[0m
�[0;32mI (340) boot: 9 ota_1 OTA app 00 11 00280000 00180000�[0m
�[0;32mI (363) boot: End of partition table�[0m

3.       Try ATs still get ERRORs:

AT+RST
OK
ready

AT+BLEINIT=2
OK

AT+BLEGATTSSRVCRE
ERROR

AT+BLEGATTSSRVSTART
ERROR

Cloud you please help me fix it?

Thanks!

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 xiewenxiang
发送时间: 2017年8月7日 21:53
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] AT+BLESCAN=1 NOT WORK 

 

@mxwang101 <https://github.com/mxwang101> 
AT+BLESCAN＝<is_enable>,[<duration time>]

1.	start scan:
AT+BLESCAN=1,3
if the duration time is 0, it will continue to scan
2.	stop scan:
AT+BLESCAN=0

mxwang101, 2017-08-07:
Thanks, All work now.

I think we do need more doc's about BLE_AT how to use, and  ' ESP32_AT_Commands_Set.md' do need to be fix.

xcguang, 2017-11-14:
Thanks for your advice, we will continue to improve the documentation.

closed 2017-11-14 by xcguang