espressif/esp-at #33: AT+BLEGATTSSRVCRE, &AT+BLEGATTSSRVSTART, Both no work

mxwang101, 2017-08-06:


FayeY, 2017-08-06:
Have you downloaded a service bin? [How to generate a service bin](https://github.com/espressif/esp32-at/tree/master/tools)

mxwang101, 2017-08-06:
No, I don't know how to download "ATBleService.bin" , " GattServiceExample.bin " to where, and how to. Cloud you please give more info? Thanks.

 

I can made "ATBleService.bin" , " GattServiceExample.bin " OK

 

Also, I try to make 'at-pki.bin', but got IOError :

 

wangxin@wangxin-PC MINGW32 ~/esp/esp32-at/esp32-at/tools

$ python AtPKI.py generate_bin -b at-pki.bin cert Cert1.pem key PrivateKey1.pem ca CA1.pem ca CA2.pem cert Cert2.pem key PrivateKey2.pem ca CA3.pem

Traceback (most recent call last):

  File "AtPKI.py", line 231, in <module>

    main()

  File "AtPKI.py", line 218, in main

    data = FileFormat.to_bytes(args.pki_list)

  File "AtPKI.py", line 136, in to_bytes

    pki_items.extend([PKIItem.to_bytes(pki_type, i, x) for i, x in enumerate(pki_list)])

  File "AtPKI.py", line 52, in to_bytes

    with open(file_name, "rb") as _file:

IOError: [Errno 2] No such file or directory: 'Cert1.pem'

 

 

Cloud you please help me to fix it?

 

Thanks!

 

 

 

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 FayeY
发送时间: 2017年8月6日 16:20
收件人: espressif/esp32-at
抄送: mxwang101; Author
主题: Re: [espressif/esp32-at] AT+BLEGATTSSRVCRE, &AT+BLEGATTSSRVSTART, Both no work (#33)

 

Have you downloaded a service bin? How to generate <https://github.com/espressif/esp32-at/tree/master/tools>  a service bin

mxwang101, 2017-08-07:
To All，

Download ' GattServiceExample.bin', Not Fix.

1 ) Download ' GattServiceExample.bin':

wangxin@wangxin-PC MINGW32 ~/esp/esp32-at/esp32-at/tools
$ esptool --port COM8 write_flash 0x21000 GattServiceExample.bin
esptool.py v2.0.1
Connecting........__
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 0)
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 2704 bytes to 225...
Wrote 2704 bytes (225 compressed) at 0x00021000 in 0.0 seconds (effective 618.0 kbit/s)...
Hash of data verified.

2) Partition Table:

[0;32mI (97) boot: Partition Table:[0m
[0;32mI (108) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (130) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (154) boot:  1 otadata          OTA data         01 00 00010000 00002000[0m
[0;32mI (177) boot:  2 nvs              WiFi data        01 02 00012000 0000e000[0m
[0;32mI (200) boot:  3 ble_data         unknown          40 01 00021000 00003000[0m
[0;32mI (223) boot:  4 server_cert      unknown          40 02 00024000 00002000[0m
[0;32mI (247) boot:  5 server_key       unknown          40 03 00026000 00002000[0m
[0;32mI (270) boot:  6 server_ca        unknown          40 04 00028000 00002000[0m
[0;32mI (293) boot:  7 fatfs            Unknown data     01 81 00070000 00090000[0m
[0;32mI (316) boot:  8 ota_0            OTA app          00 10 00100000 00180000[0m
[0;32mI (340) boot:  9 ota_1            OTA app          00 11 00280000 00180000[0m
[0;32mI (363) boot: End of partition table[0m

3) Try ATs still get ERRORs:

AT+RST
OK
ready

AT+BLEINIT=2
OK

AT+BLEGATTSSRVCRE
ERROR

AT+BLEGATTSSRVSTART
ERROR

Cloud you please help me fix it?

Thanks!

mxwang101, 2017-08-07:
Just find，

1）Send:

AT+BLEINIT=2
OK

AT+BLEGATTSSRVCRE
ERROR

2）Get info:

[0;32mI (15618) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (16179) BLE_AT_GATTS: EVT 0, gatts if 3
[0m
partation fail

any help?

xiewenxiang, 2017-08-07:
@mxwang101 
Hi mxwang101,
Do you have downloaded at_customer.bin to 0x20000?

xcguang, 2017-08-07:
Hi @mxwang101 ,
Please don't modify `partitions_at.csv`. And you can run `python esp-idf/components/partition_table/gen_esp32part.py -q at_customize.csv at_customize.bin` to generate at_customize.bin, then download `at_customize.bin` to 0x20000 and `GattServiceExample.bin` to 0x21000.

mxwang101, 2017-08-07:
Yes，I downloaded，use：

$ esptool --port COM8 write_flash 0x20000 at_customer.bin

Is it right way to download .bin file?

and still not work:

AT+BLEINIT=2
OK
AT+BLEGATTSSRVCRE
ERROR

I renamed ATBleService.bin to at_customer.bin. Is it right file?

how to get at_customer.bin? I can't find any info about it please help

here is Partition Table:
 [0;32mI (97) boot: Partition Table:[0m
[0;32mI (108) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (130) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (154) boot:  1 otadata          OTA data         01 00 00010000 00002000[0m
[0;32mI (177) boot:  2 nvs              WiFi data        01 02 00012000 0000e000[0m
[0;32mI (200) boot:  3 at_customize     unknown          40 00 00020000 000e0000[0m
[0;32mI (223) boot:  4 ota_0            OTA app          00 10 00100000 00180000[0m
[0;32mI (247) boot:  5 ota_1            OTA app          00 11 00280000 00180000[0m
[0;32mI (270) boot: End of partition table[0m

xcguang, 2017-08-07:
`partitions_at.bin` is the primary partitions table,which is not allowed to modify; `at_customize.bin` is custom partitions table,and you can edit it. You can find the partition named `ble_data` in `at_customize.csv`, and you can change its `Offset` and `Size`, but ' GattServiceExample.bin' must be downloaded to the "Offset" address.

mxwang101, 2017-08-07:
Thanks !  Work now.

mxwang101, 2017-08-07:
Thanks, Works now

 

I think we do need more documents about BLE_AT how to use, and  '  <https://github.com/espressif/esp32-at/blob/master/docs/ESP32_AT_Commands_Set.md> ESP32_AT_Commands_Set.md'need to be fix.

 

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 xcguang
发送时间: 2017年8月7日 23:39
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] AT+BLEGATTSSRVCRE, &AT+BLEGATTSSRVSTART, Both no work (#33)

 

partitions_at.bin is the primary partitions table,which is not allowed to modify; at_customize.bin is custom partitions table,and you can edit it. You can find the partition named ble_data in at_customize.csv, and you can change its Offset and Size, but ' GattServiceExample.bin' must be downloaded to the "Offset" address.

xcguang, 2017-08-07:
The  AtPKI.py is to generate cert,private_key and ca for AT SSL server,and if you need not CA,please ignore it. At first, you should have cert,private_key and ca.Then you can run
`python AtPKI.py generate_bin -b at-cert.bin cert Cert.pem`
`python AtPKI.py generate_bin -b at-key.bin key PrivateKey.pem`
`python AtPKI.py generate_bin -b at-ca.bin key CA.pem`.
By default,you can download at-cert.bin to 0x24000, at-key.bin to 0x26000 and at-ca.bin to 0x28000.

### Notice: If the `CA.pem` is from the third party, not Certificate Authority, you should run `python AtPKI.py generate_bin -b at-ca.bin key Cert.pem` to generate at-ca.bin.

mxwang101, 2017-08-08:
Thanks for info about ATPKI, I will test it later.  I have problem about " AT+BLEGATTSNTFY"
 
Here is log:

ready

AT+BLEINIT=2
OK

AT+BLEGATTSSRVCRE
OK

AT+BLEGATTSSRVSTART
OK

AT+BLEADVSTART
OK

+BLECONN:0,"6b:03:e5:83:d7:2d"

AT+BLEGATTSCHAR?
+BLEGATTSCHAR:"char",1,1,0xC300
+BLEGATTSCHAR:"desc",1,1,1
+BLEGATTSCHAR:"char",1,2,0xC301
+BLEGATTSCHAR:"desc",1,2,1
OK

AT+BLEGATTSNTFY=1,1,"303132" 
ERROR

How to fix it?

Also, AT+BLECONN and AT+BLEDISCONN, Both Not work. please help fix.

Thanks

xiewenxiang, 2017-08-08:
@mxwang101
AT+BLEGATTSNTFY=<conn_index>,<srv_index>,<char_index>,<length>
after you input the command, it will enter input mode
AT+BLEGATTSNTFY=0,1,1,3
>
you can input  012 now。

AT+BLECONN and AT+BLEDISCONN only work in client mode
AT+BLECONN=<conn_index>,<address>   //conn_index must be 0, it was reserved for future used
AT+BLEDISCONN=<conn_index>

mxwang101, 2017-08-08:
AT+BLEGATTSNTFY works thanks.

AT+BLECONN and AT+BLEDISCONN only work in client mode?  
So, server mode can not scan, & connect to client?

Our projects need the ble server to scan ble clients, and pick up one of them to connect，then pass data to the server by wifi.

I try to use ESP32 since it has both BLE & Wifi functions.  Do you think we can use ESP32-AT+a CPU do this job?

Thanks!

xcguang, 2017-08-08:
There is a mcu in esp32, you can develop base on it. If the product function is simple,maybe only esp32 will work well.

xiewenxiang, 2017-08-08:
Server can be scanned, but in esp32 ble-at, this is banned,it will return error.
But the server can not send connection request, The AT+BLECONN only work in client mode

mxwang101, 2017-08-08:
Yes，this is the best way to do. Do you have some example code which can be installed with esp32-at same times, and we can run it with call the at commands. For examples, we can use ble_at get the device data, after simple prosesed, use wifi _at send the data out.
On 08/08/2017 14:07, xcguang wrote:

There is a mcu in esp32, you can develop base on it. If the product function is simple,maybe only esp32 will work well.

xcguang, 2017-08-08:
Sorry, now we haven't one. Maybe we can develop a simple demo for you, but it will take a little time.

mxwang101, 2017-08-08:
Thanks, We do need this kind of simple demo!  waiting for ...............

mxwang101, 2017-08-09:
Also,  in Example：AT+BLEGATTCPRIMSRV? don't work, please help

	AT+BLEINIT=1   // role: client
	AT+BLECONN="24:12:5f:9d:91:98"
	AT+BLEGATTCPRIMSRV?
	AT+BLEGATTCCHAR=3 // set a specific index according to the result of the previous command
	AT+BLEGATTCWR=3,3,,"30313233" // set a specific index according to the result of the previous command

Log:
AT+BLEINIT=1
OK
AT+BLECONN=0,"c0:35:5e:95:85:e3"
OK
AT+BLEGATTCPRIMSRV?
ERROR

mxwang101, 2017-08-09:
Also, in file ESP32_AT_Commands_Set.md, some AT commands about BLE need to updated to right formats,   Can I get some info about there, even before office release? we do need to make it work as so as possible.
Thanks.

mxwang101, 2017-08-09:
Also we do need "# 6. BLE AT Example " section works, if the example works, we can move to our project.

## 6. BLE AT Example  
Below is an example of using two ESP32 modules, one as a BLE server (hereafter named "ESP32 Server"), the other one as a BLE client (hereafter named "ESP32 Client"). The example shows how to use BLE functions with AT commands.  
***Notice:***

xiewenxiang, 2017-08-09:
we will update the AT docs today. 
```
AT+BLEINIT=1   // role: client
AT+BLECONN=0,"24:12:5f:9d:91:98"
AT+BLEGATTCPRIMSRV=0
AT+BLEGATTCCHAR=0,3 // set a specific index according to the result of the previous command
AT+BLEGATTCWR=0,3,3,,4 // set a specific index according to the result of the previous command
>    //waiting for input
OK
```

mxwang101, 2017-08-09:
Thanks lot! Please let me kown when released.
On 08/09/2017 09:58, xiewenxiang wrote:

we will update the AT docs today.

AT+BLEINIT=1   // role: client
AT+BLECONN=0,"24:12:5f:9d:91:98"
AT+BLEGATTCPRIMSRV=0
AT+BLEGATTCCHAR=0,3 // set a specific index according to the result of the previous command
AT+BLEGATTCWR=0,3,3,,4 // set a specific index according to the result of the previous command

>    //waiting for input
OK

xcguang, 2017-08-09:
Hi, `ESP32_AT_Commands_Set.md` has been updated, please refer to the latest document,thanks.

mxwang101, 2017-08-09:
Got, Thanks, Try it!

xcguang, 2017-08-09:
[at_task.c.zip](https://github.com/espressif/esp32-at/files/1210574/at_task.c.zip)
Hi, this is a very simple demo. 
The task app_test_task periodically sends command ```AT+GMR\r\n```, and the task app_main_task processes response.Maybe you need to parse the response in app_main_task.

mxwang101, 2017-08-09:
Get，thanks
On 08/09/2017 10:24, xcguang wrote:

Hi, ESP32_AT_Commands_Set.md has been updated, please refer to the latest document,thanks.

mxwang101, 2017-08-09:
Thanks，I will try it later.
On 08/09/2017 15:01, xcguang wrote:

at_task.c.zip
Hi, this is a very simple demo.
The task app_test_task periodically sends command AT+GMR\r\n, and the task app_main_task processes response.Maybe you need to parse the response in app_main_task.

mxwang101, 2017-08-09:
Thx. Try it but still have some problems:

 

1) The documents 6.
<https://github.com/espressif/esp32-at/blob/master/docs/ESP32_AT_Commands_Se
t.md#BLE-AT> BLE AT Example

.....................

AT+BLEGATTCCHAR=0,3

 

 Response:

 +BLEGATTCCHAR:"char",0,3,1,0xC300,2

 +BLEGATTCCHAR:"desc",0,3,1,1,0x2901

 +BLEGATTCCHAR:"char",0,3,2,0xC301,2

 +BLEGATTCCHAR:"desc",0,3,2,1,0x2901

 +BLEGATTCCHAR:"char",0,3,3,0xC302,8

 +BLEGATTCCHAR:"desc",0,3,3,1,0x2901

 +BLEGATTCCHAR:"char",0,3,4,0xC303,4

 +BLEGATTCCHAR:"desc",0,3,4,1,0x2901

 +BLEGATTCCHAR:"char",0,3,5,0xC304,8

 +BLEGATTCCHAR:"char",0,3,6,0xC305,16

 +BLEGATTCCHAR:"desc",0,3,6,1,0x2902

 +BLEGATTCCHAR:"char",0,3,7,0xC306,32

 +BLEGATTCCHAR:"desc",0,3,7,1,0x2902

 OK  

 

but I run same AT, only get:

 

AT+BLEGATTCCHAR=0,3

+BLEGATTCCHAR:"char",0,3,1,0xC300,0x02

+BLEGATTCCHAR:"desc",0,3,1,1,0x2901

+BLEGATTCCHAR:"char",0,3,2,0xC301,0x02

+BLEGATTCCHAR:"desc",0,3,2,1,0x2901

 

OK

 

Total different！do we user different ‘GattServiceExample.csv’？ or others?
If you are use different one, please set me, so I can test all functions
frist.

 

Also, could you tell me, what is " perm" means, and how to pickup right one?

 

Thanks,

 

2) In the document 

 

## 6. [BLE AT Example](#BLE-AT)

..........................

 

2. Establish BLE connection:  

 

    ESP32 Server:   

    (1) Query the BLE address. For example, its address is
"24:0a:c4:03:f4:d6".

 

        Command:

        AT+BLEADDR?                              // get server's BLE address

         

        Response:

        +BLEADDR:24:0a:c4:03:f4:d6

        OK

    (2) Start advertising.
//???????????????????????????,,,have problem????

 

        Command:

        AT+BLEADDR?                              // get server's BLE address

        

        Response:

        +BLEADDR:24:0a:c4:03:f4:d6

        OK

....................................................

AND

 

3. Read/Write a characteristic:  

    ESP32 Server:  
    (1) Create services.  
    
        AT+BLEGATTSSRVCRE
        
        Response:
        OK
    (2) Start services.  
    
        AT+BLEGATTSSRVSTART
        
        Response:
        OK
    (3) Discover characteristics.  

 

before, Start advertising. I find, after Start advertising, you can not
Start services. ? 

 

how to fix?

 

Thanks

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com
[mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 xcguang
发送时间: 2017年8月9日 10:25
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] AT+BLEGATTSSRVCRE, &AT+BLEGATTSSRVSTART, Both
no work (#33)

 

Hi, ESP32_AT_Commands_Set.md has been updated, please refer to the latest
document,thanks.

—
<https://github.com/espressif/esp32-at/issues/33#issuecomment-321134314>  it
on GitHub, or mute
aArvNxRks5sWRhlgaJpZM4Ounif>  the thread.
<https://github.com/notifications/beacon/AZZHqoAZyd9NLgF2wIBNZceGhBMgzZfaks5
sWRhlgaJpZM4Ounif.gif>

xiewenxiang, 2017-08-09:
@mxwang101 
1). Because the GattServiceExample.csv is different.
2). Do you can't start adv? I have tested it ,it works
3). After Start advertising, you still can Start services
```
AT+BLEINIT=2

OK
AT+BLEADVSTART

OK
AT+BLEGATTSSRVCRE

OK
AT+BLEGATTSSRVSTART

OK
```

mxwang101, 2017-08-09:
Yes, work now. I will try the examples again, if get right GattServiceExample.csv

 

Cloud you please sent me the GattServiceExample.csv which used in the example?

 

Also, the means of  " perm", in the GattServiceExample.csv?

 

Thanks,

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 xiewenxiang
发送时间: 2017年8月9日 20:06
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] AT+BLEGATTSSRVCRE, &AT+BLEGATTSSRVSTART, Both no work (#33)

 

@mxwang101 <https://github.com/mxwang101> 
1). Because the GattServiceExample.csv is different.
2). Do you can't start adv? I have tested it ,it works
3). After Start advertising, you still can Start services

AT+BLEINIT=2
 
OK
AT+BLEADVSTART
 
OK
AT+BLEGATTSSRVCRE
 
OK
AT+BLEGATTSSRVSTART
 
OK

xcguang, 2018-08-20:
Hi, 
If have any question, please create the new issue, and I will close this issue firstly.

closed 2018-08-20 by xcguang

mxwang101, 2018-08-20:
OK, Thx

 

发件人: noreply@github.com [mailto:noreply@github.com] 代表 xcguang
发送时间: 2018年8月20日 10:03
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] AT+BLEGATTSSRVCRE, &AT+BLEGATTSSRVSTART, Both no work (#33)

 

Hi,
If have any question, please create the new issue, and I will close this issue firstly.