espressif/esp-at #35: How to setup GattServiceExample.csv for test?

mxwang101, 2017-08-10:


mxwang101, 2017-08-10:
In the tools, have a GattServiceExample.csv, but can not get right config for the test, no ideal how to setup:

          CHAR:3,5,0xC304,8
        +CHAR:3,6,0xC305,16
        +CHAR:3,7,0xC306,32
        
Discover characteristics.  
    
        AT+BLEGATTCCHAR=3
        
        Response:
        +CHAR:3,1,0xC300,2
        +DESC:3,1,1,0x2901
        +CHAR:3,2,0xC301,2
        +DESC:3,2,1,0x2901
        +CHAR:3,3,0xC302,8
        +DESC:3,3,1,0x2901
        +CHAR:3,4,0xC303,4
        +DESC:3,4,1,0x2901
        +CHAR:3,5,0xC304,8
        +CHAR:3,6,0xC305,16
        +DESC:3,6,1,0x2902
        +CHAR:3,7,0xC306,32
        +DESC:3,7,1,0x2902
        OK

mxwang101, 2017-08-10:
Cloud we get right GattServiceExample.csv for  ' ESP32_AT_Commands_Set.md'?
Or tell us how to set up GattServiceExample.csv for to like '+CHAR:3,7,0xC306,32' , how to setup '32'(Indicate)?  Thx

mxwang101, 2017-08-10:
HELP!  AT+BLEGATTCRD=0,3,1  NOT WORK!!! 

AT+BLEGATTCCHAR=0,3
+BLEGATTCCHAR:"char",0,3,1,0xC300,0x02
+BLEGATTCCHAR:"desc",0,3,1,1,0x2901
+BLEGATTCCHAR:"char",0,3,2,0xC301,0x02
+BLEGATTCCHAR:"desc",0,3,2,1,0x2901
+BLEGATTCCHAR:"char",0,3,3,0xC302,0x16
+BLEGATTCCHAR:"desc",0,3,3,1,0x2902
+BLEGATTCCHAR:"char",0,3,4,0xC303,0x32
+BLEGATTCCHAR:"desc",0,3,4,1,0x2902

OK
AT+BLEGATTCRD=0,3,1

ERROR

mxwang101, 2017-08-11:
Thank @FayeY !

mxwang101, 2017-08-11:
Hi Faye, 

 

I installed new GattServiceExample.csv. It is OK!

 

But try to send AT+BLEGATTCRD, still get ERROR, How to fix it? Please help. Thanks!

 

log file:

 

AT+BLEGATTCCHAR=0,3

+BLEGATTCCHAR:"char",0,3,1,0xC300,0x02

+BLEGATTCCHAR:"desc",0,3,1,1,0x2901

+BLEGATTCCHAR:"char",0,3,2,0xC301,0x02

+BLEGATTCCHAR:"desc",0,3,2,1,0x2901

+BLEGATTCCHAR:"char",0,3,3,0xC302,0x08

+BLEGATTCCHAR:"desc",0,3,3,1,0x2901

+BLEGATTCCHAR:"char",0,3,4,0xC303,0x04

+BLEGATTCCHAR:"desc",0,3,4,1,0x2901

+BLEGATTCCHAR:"char",0,3,5,0xC304,0x08

+BLEGATTCCHAR:"char",0,3,6,0xC305,0x10

+BLEGATTCCHAR:"desc",0,3,6,1,0x2902

+BLEGATTCCHAR:"char",0,3,7,0xC306,0x20

+BLEGATTCCHAR:"desc",0,3,7,1,0x2902

 

OK

AT+BLEGATTCRD=0,3,1

 

ERROR

 

Try to read 0x2901, 0x2902, both OK:

 

AT+BLEGATTCRD=0,3,1,1

+BLEGATTCRD:0,1,

 

OK

AT+BLEGATTCRD=0,3,1,1

+BLEGATTCRD:0,1,

 

OK

AT+BLEGATTCRD=0,3,7,1

+BLEGATTCRD:0,2,01

 

OK

AT+BLEGATTCRD=0,3,6,1

+BLEGATTCRD:0,2,01

 

OK

AT+BLEGATTCRD=0,3,4,1

+BLEGATTCRD:0,0,

 

OK

 

Try to read 0xc301, 0xc302,both not ok:

 

OK

AT+BLEGATTCRD=0,3,1

 

ERROR

AT+BLEGATTCRD=0,3,2

 

ERROR

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 FayeY
发送时间: 2017年8月11日 10:04
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] How to setup GattServiceExample.csv for test? (#35)

 

Hi @mxwang101 <https://github.com/mxwang101>  , the GattServiceExample.csv has been updated, and the guide is in the README.md <https://github.com/espressif/esp32-at/tree/master/tools> . Thanks for your interest in ESP32!

xiewenxiang, 2017-08-11:
@mxwang101 
we will fix it this afternoon

mxwang101, 2017-08-11:
Thanks, when fix, lat me konw

mxwang101, 2017-08-18:
fixed or not?

On 08/11/2017 11:58, xiewenxiang wrote:

@mxwang101
we will fix it this afternoon

xiewenxiang, 2017-08-18:
@mxwang101 
It was fixed, we will release it after test.

mxwang101, 2017-08-18:
THANKS! Waiting for .....

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 xiewenxiang
发送时间: 2017年8月18日 19:49
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] How to setup GattServiceExample.csv for test? (#35)

 

@mxwang101 <https://github.com/mxwang101> 
It was fixed, we will release it after test.

xiewenxiang, 2017-08-25:
@mxwang101 
we have fixed it ,please update your code to latest version.

mxwang101, 2017-08-25:
Thanks！
On 08/25/2017 11:40, xiewenxiang wrote:

@mxwang101
we have fixed it ,please update your code to latest version.

mxwang101, 2017-08-30:
TO ALL：WORK。THANKS！ 

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 xiewenxiang
发送时间: 2017年8月25日 11:41
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] How to setup GattServiceExample.csv for test? (#35)

 

@mxwang101 <https://github.com/mxwang101> 
we have fixed it ,please update your code to latest version.

closed 2017-09-14 by FayeY