espressif/esp-at #36: NTP Timezone works the other way arounf

diegoherranz, 2017-09-22:


luc-github, 2017-09-29:
Hi  as reported alsohere  https://github.com/espressif/arduino-esp32/issues/403#issuecomment-305183116
I also confirm current behavior on ESP32 is  for time zone +1 need to set -1
`configTime(-3600, 0, "my.ntp.server", NULL, NULL);`
and for time zone -1 need to set +1
`configTime(3600, 0, "my.ntp.server", NULL, NULL);`

on ESP8266 the behaviour is correct : for time zone +1 need to use +1
`configTime(3600, 0, "my.ntp.server", NULL, NULL);`

I did a script to compare same code on both platform : https://gist.github.com/luc-github/9511ed8e4bf8adb0f6cf40e81381aa53

daveyboy17, 2017-12-01:
I am in New Zealand (UTC+13) and to get the timezone into the right day I have to set AT+CIPSNTPCFG=1,-11 As -11 is the -ve limit I an still 2 hours behind the real time.
Tested with:
AT version:1.0.0.0(870285c - Nov 14 2017 07:26:38)
SDK version:v2.1-65-g8bca703
compile time:Nov 17 2017 07:53:25
Bin version(Wroom32):1.0.0

diegoherranz, 2018-04-14:
Tested again today with current HEAD (1605786fedf8ef2a03bddbff913b143a66303d93) and not fixed yet.

By the way, it seems the API changed since the original comment (see details [here](https://www.espressif.com/sites/default/files/documentation/esp32_at_instruction_set_and_examples_en.pdf)) :
`AT+CIPSNTPCFG=-2,"us.pool.ntp.org","cn.ntp.org.cn","ntp.sjtu.edu.cn"`
is now:
`AT+CIPSNTPCFG=1,-2,"us.pool.ntp.org","cn.ntp.org.cn","ntp.sjtu.edu.cn"`

xcguang, 2020-01-09:
We tested the latest branch, it looks well, if you still have any question, please reopen the issue, thanks.

closed 2020-01-09 by xcguang

diegoherranz, 2020-01-11:
I've tested 7cac7ad (tag v2.0.0.0_esp32) and it works properly now.
Thanks!