espressif/esp-at #38: Custom AT cmd calling a web server function with ssl

santanapablo1975, 2017-10-03:
Hi All

I am creating a custom AT comand in order to call a function, but the function is a web server with ssl security, but this functions does not work when it is integrated with the at lib, I have teste to call a web server with no SSL and it works.

I would like to as if the AT lib some how is calling or creting a socket with SSL features?

Regards

Juan

xcguang, 2017-10-09:
Hi,
Does web server use the certification?
Could you provide the AT command when connecting SSL web server?

santanapablo1975, 2017-10-09:
Hi

Yes I am using the open SSL certificate.

I call my custom made AT cmd like this:

static esp_at_cmd_struct at_custom_cmd[] = {
    {"+UART", NULL, at_queryCmdUart, at_setupCmdUartDef, NULL},
    {"+UART_CUR", NULL, at_queryCmdUart, at_setupCmdUart, NULL},
    {"+UART_DEF", NULL, at_queryCmdUartDef, at_setupCmdUartDef, NULL},
    {"+CIUPDATE", NULL, NULL, NULL, at_exeCmdCipupdate},
    {".BLK", NULL, NULL, NULL, at_exeCmdBlk},
    {".BLKOFF", NULL, NULL, NULL, at_exeCmdBlkoff},
    {".UART2", NULL, NULL, NULL, at_exeCmdUart2},
    {".OPENSSLSERVER", NULL, NULL, NULL, at_exeCmdOpenSslServer},
    {".OPENSSLSERVERNOSLL", NULL, NULL, NULL, at_exeCmdOpenNOSslServer}
};

static uint8_t at_exeCmdOpenSslServer(uint8_t *cmd_name)//add get station
ip and ap ip
{
            char* at_write_port;
            at_write_port="\r\nOpen SSL Server very simple example!\r\n";
            esp_at_port_write_data((uint8_t
*)at_write_port,strlen(at_write_port));
                openssl_client_init((void *)at_write_port);

                return ESP_AT_RESULT_CODE_OK;
}

the function openssl_client_init is the one in the idf examples, oof course
with out the wifi set up, because it is done using the AT cmd

I guess (just assuming" the AT lib is already starting a kind of web server
process.

do you think that this kibd of integrations would be possible or not using
the AT lib?

Regards

Juan

xcguang, 2017-10-10:
Hi，
Could you provide any log about fail? In fact, AT lib can support ssl server, but not ssl web server. Maybe you can try it, and for more information, please refer to the command ```AT+CIPSERVER``` and how to generate PKI Bin is described in the document ```tools/README.md```.

santanapablo1975, 2017-10-10:
Hi

I am sending 2 logs:

1) the one with the SSL web server and no AT lib, it is running

2)the second one with the SSL web server and WITH AT lib, it is not running

My guess is with the AT lib compiled some how other SSL functions are
reserved or blocked?

Do you think that this kind of integration would be possible in the near
future?

REgards

Juan

ESP-Coco, 2017-10-11:
Hi, I have tested openssl_client_init and openssl_server_init integrated with the at lib, and I found the result is exactly the same as the IDF demo. Could you provide some logs about fail to help us analyze it?

santanapablo1975, 2017-10-12:
Hi,

Thanks for the good news!!!, then I decided to test all again, it means:

1) downloaded the whole AT lib and the proper IDF, it works!!
2)add only openssl_client_init, it works!!
3) then I decided to compare the sdkconfig files (my project vs the AT lib)
and I have found that my project had some differences in some resources,
then I have updated my project according the AT lib and now it works!!!!

Thanks in any case, I think we can close the case!!

Regarsds!!

Juan

closed 2017-10-12 by santanapablo1975