espressif/esp-at #41: Compile error following README instructions

tmedicci, 2017-10-09:
Hello, I've installed ESP-IDF just now and I really compiled and flashed containing examples. However, I tried to compile ESP32-at environment and I've got the following error:

I'm using Ubuntu 16.04.3 64-bit on a VM. The ESP-IDF path from esp32-at (cloned from this repository) was added to PATH environment, as mentioned on instructions.

`WARNING: Toolchain version is not supported: crosstool-NG crosstool-ng-1.22.0-73-ge28a011
Expected to see version: crosstool-NG crosstool-ng-1.22.0-61-gab8375a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Toolchain version is not supported: crosstool-NG crosstool-ng-1.22.0-73-ge28a011
Expected to see version: crosstool-NG crosstool-ng-1.22.0-61-gab8375a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
CC src/efuse.o
CC src/bootloader_flash.o
CC src/secure_boot_signatures.o
CC src/secure_boot.o
CC src/flash_partitions.o
CC src/esp_image_format.o
CC src/flash_encrypt.o
CC src/bootloader_random.o
AR libbootloader_support.a
CC log.o
AR liblog.a
CC spi_flash_rom_patch.o
AR libspi_flash.a
CC micro-ecc/uECC.o
AR libmicro-ecc.a
CC esp32/rtc_pm.o
CC esp32/cpu_util.o
CC esp32/brownout.o
CC esp32/rtc_init.o
CC esp32/rtc_sleep.o
CC esp32/rtc_time.o
CC esp32/rtc_clk.o
AR libsoc.a
CC bootloader_start.o
CC flash_qio_mode.o
AR libmain.a
LD bootloader.elf
esptool.py v2.0
building AT customized partition table at_customize.csv
building AT customized partition table succeed
generating AT customized partition bins to /home/tmedicci/esp/esp32-at/build/customized_partitions
generating ble_data.bin: python /home/tmedicci/esp/esp32-at/tools/BLEService.py -t /home/tmedicci/esp/esp32-at/build/customized_partitions/ble_data.bin /home/tmedicci/esp/esp32-at/components/customized_partitions/raw_data/ble_data/example.csv
AT customized partitions successfully generated: ble_data
generating server_cert.bin: python /home/tmedicci/esp/esp32-at/tools/AtPKI.py generate_bin -b /home/tmedicci/esp/esp32-at/build/customized_partitions/server_cert cert /home/tmedicci/esp/esp32-at/components/customized_partitions/raw_data/server_cert/example.crt
AT customized partitions successfully generated: server_cert
generating server_key.bin: python /home/tmedicci/esp/esp32-at/tools/AtPKI.py generate_bin -b /home/tmedicci/esp/esp32-at/build/customized_partitions/server_key key /home/tmedicci/esp/esp32-at/components/customized_partitions/raw_data/server_key/example.key
AT customized partitions successfully generated: server_key
generating server_ca.bin: python /home/tmedicci/esp/esp32-at/tools/AtPKI.py generate_bin -b /home/tmedicci/esp/esp32-at/build/customized_partitions/server_ca ca /home/tmedicci/esp/esp32-at/components/customized_partitions/raw_data/server_ca/example.crt
AT customized partitions successfully generated: server_ca
AT customized bin generation finished
Building partitions from /home/tmedicci/esp/esp32-at/partitions_at.csv...
CC esp_ota_ops.o
AR libapp_update.a
CC src/efuse.o
CC src/bootloader_flash.o
CC src/secure_boot_signatures.o
CC src/secure_boot.o
CC src/flash_partitions.o
CC src/esp_image_format.o
CC src/flash_encrypt.o
CC src/bootloader_random.o
AR libbootloader_support.a
CC bt.o
CC bluedroid/bta/dm/bta_dm_api.o
CC bluedroid/bta/dm/bta_dm_sco.o
CC bluedroid/bta/dm/bta_dm_ci.o
CC bluedroid/bta/dm/bta_dm_act.o
CC bluedroid/bta/dm/bta_dm_pm.o
CC bluedroid/bta/dm/bta_dm_main.o
CC bluedroid/bta/dm/bta_dm_cfg.o
[... 553 lines cut ...]
/home/tmedicci/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:776:1: error: expected ',' or ';' before '{' token
 {
 ^
/home/tmedicci/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:786:42: error: '__gthread_recursive_mutex_init_function' declared as an 'inline' variable
 __gthread_recursive_mutex_init_function (__gthread_recursive_mutex_t *__mutex)
                                          ^
/home/tmedicci/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:786:42: error: '__gthread_recursive_mutex_t' was not declared in this scope
/home/tmedicci/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:786:71: error: '__mutex' was not declared in this scope
 __gthread_recursive_mutex_init_function (__gthread_recursive_mutex_t *__mutex)
                                                                       ^
/home/tmedicci/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:787:1: error: expected ',' or ';' before '{' token
 {
 ^
In file included from /home/tmedicci/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/ext/atomicity.h:35:0,
                 from /home/tmedicci/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/basic_string.h:39,
                 from /home/tmedicci/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/string:52,
                 from /home/tmedicci/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/random:40,
                 from /home/tmedicci/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_algo.h:66,
                 from /home/tmedicci/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/algorithm:62,
                 from /home/tmedicci/esp/esp32-at/esp-idf/components/cxx/./cxx_guards.cpp:20:
/home/tmedicci/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr.h:151:27: error: expected declaration before end of line
/home/tmedicci/esp/esp32-at/esp-idf/make/component_wrapper.mk:210: recipe for target 'cxx_guards.o' failed
make[1]: *** [cxx_guards.o] Error 1
/home/tmedicci/esp/esp32-at/esp-idf//make/project.mk:386: recipe for target 'cxx-build' failed
make: *** [cxx-build] Error 2`

xcguang, 2017-10-09:
Hi,
Please use the old ESP32 toolchain  to try again,which can be downloaded from  https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-61-gab8375a-5.2.0.tar.gz.

tmedicci, 2017-10-09:
Ok! That was a problem related about toolchain version. Thanks!

closed 2017-10-09 by tmedicci