espressif/esp-at #44: compilation problem libat_core.a undefined reference

skxo, 2017-10-15:
Hi, 
I try to compile esp32-at by myself and I have these errors (see log below). 
Everything si ok for other project compilations like hello_world.....
If people can help me on this, it would be wonderful. Thanks a lot!
sk

LOG:
LD build/esp-at.elf
/home/xisco/esp/esp32-at/components/at/lib/libat_core.a(at_bleCmd.o):(.literal.gattc_profile_event_handler+0x50): undefined reference to `esp_ble_gattc_get_descriptor'
/home/xisco/esp/esp32-at/components/at/lib/libat_core.a(at_bleCmd.o):(.literal.gattc_profile_event_handler+0x54): undefined reference to `esp_ble_gattc_get_characteristic'
/home/xisco/esp/esp32-at/components/at/lib/libat_core.a(at_bleCmd.o):(.literal.gattc_profile_event_handler+0x58): undefined reference to `esp_ble_gattc_get_included_service'
/home/xisco/esp/esp32-at/components/at/lib/libat_core.a(at_bleCmd.o):(.literal.at_setupCmdBleCfgMtu+0x0): undefined reference to `esp_ble_gattc_config_mtu'
/home/xisco/esp/esp32-at/components/at/lib/libat_core.a(at_bleCmd.o):(.text.gattc_profile_event_handler+0x41f): undefined reference to `esp_ble_gattc_get_descriptor'
/home/xisco/esp/esp32-at/components/at/lib/libat_core.a(at_bleCmd.o):(.text.gattc_profile_event_handler+0x446): undefined reference to `esp_ble_gattc_get_characteristic'
/home/xisco/esp/esp32-at/components/at/lib/libat_core.a(at_bleCmd.o):(.text.gattc_profile_event_handler+0x51f): undefined reference to `esp_ble_gattc_get_descriptor'
/home/xisco/esp/esp32-at/components/at/lib/libat_core.a(at_bleCmd.o):(.text.gattc_profile_event_handler+0x6e7): undefined reference to `esp_ble_gattc_get_included_service'
/home/xisco/esp/esp32-at/components/at/lib/libat_core.a(at_bleCmd.o): In function `at_setupCmdBleCfgMtu':
(.text.at_setupCmdBleCfgMtu+0x42): undefined reference to `esp_ble_gattc_config_mtu'
/home/xisco/esp/esp32-at/components/at/lib/libat_core.a(at_bleCmd.o): In function `at_setupCmdBleGattcPrimSrv':
(.text.at_setupCmdBleGattcPrimSrv+0x10f): undefined reference to `esp_ble_gattc_get_characteristic'
/home/xisco/esp/esp32-at/components/at/lib/libat_core.a(at_bleCmd.o): In function `at_setupCmdBleGattcInclSrv':
(.text.at_setupCmdBleGattcInclSrv+0x9e): undefined reference to `esp_ble_gattc_get_included_service'
collect2: error: ld returned 1 exit status
/home/xisco/esp/esp-idf/make/project.mk:366: recipe for target '/home/xisco/esp/esp32-at/build/esp-at.elf' failed
make: *** [/home/xisco/esp/esp32-at/build/esp-at.elf] Error 1

xcguang, 2017-10-16:
Hi,
Do you enable ble function when ```make menuconfig```?
Maybe you can use sdkconfig.defaults in esp32-at, and then ```make defconfig```.

skxo, 2017-10-17:
hi, 
thanks for your help.
unfortunately, BLE is enabled and make defconfig using sdkconfig.defaults does not compile well (same error at LD build/esp-at.elf)
Regards

xcguang, 2017-10-30:
Which esp-idf do you use?please try https://github.com/espressif/esp-idf/tree/e2e46f4243fa8f54c8fdf272829637d49e11bf4a.

skxo, 2017-10-31:
Hi xcguang,

thanks for your answer. I used the latest version.
Then, with your link, I use the 2.1 version now. 
I had to modify the toolchain following http://esp-idf.readthedocs.io/en/v2.1/get-started/index.html
It compiles fine.
Issue closed and thanks again !
sk

closed 2017-11-14 by xcguang