espressif/esp-at #45: Pairing problem

MaciekJanicki, 2017-11-07:
Hello,
I currently try to use the esp32-at project to establish a connection with blood pressure monitor. The problem is that although I can connect I can't pair with the device which leads to not being able to get access to read/write the needed GATT characteristics. The commands used are the same as presented in examples. 
Please help, would appreciate any hint.

Best regards

xiewenxiang, 2017-11-14:
Hi MaciekJanicki,

The BLE pairing and bonding commands was not support in this version, we will add these in next release.

Best regards.

xcguang, 2018-01-25:
@MaciekJanicki  
Now the latest AT project supports this function, please try it if possible.

MaciekJanicki, 2018-01-30:
@xcguang 
Thanks for the update! I will try it in some spare time.

kealist, 2018-03-02:
What are the AT Pairing and Bonding commands?  I don't see them in the documentation

kmakhil90, 2018-06-20:
Hi,

Anybody having sequence of AT commands to make a pairing with other BLE device?

kealist, 2018-06-20:
@kmakhil90 The only thing you can do is use `AT+BLESECPARAM` to set security parameters with ESP_LE_AUTH_REQ_SC_BOND, or MITM if you have a IO Capability beyond NONE.   That's probably the best you can get to have a paring mode, but we have no io capability on our device, so anything can pair at any time.

xiewenxiang, 2018-06-21:
@kmakhil90  [doc](https://github.com/espressif/esp32-at/blob/master/docs/ESP32_AT_Commands_Set.md#cmd-BLESMPPAR)

kmakhil90, 2018-06-21:
hi @xiewenxiang, @kealist,

Thanks for the information shared. But i got the commands related to pairing. but i would like to know if anybody know the sequence of command execution to acheive this?

FayeY, 2018-06-22:
Hi @kmakhil90 ,

Please try this:
1. AT+BLESECPARAM=1,1,16,3,3 // set parameters of pairing
2. Establish a BLE connection
3. AT+BLEENC=0,3 // start pairing request
// message  +BLESECNTFYKEY:<key> will prompt
// input the same <key> on phone app to pair the ESP32
// message "+BLEAUTHCMPL" means that the pairing is completed.

Others:
User can use commands below to unbind BLE pairing bonded device, if it is needed.
AT+BLEENCDEV? to list all bonded devices.
AT+BLEENCCLEAR=0 to unbind the device of which index is 0.

Thanks.

closed 2018-06-22 by FayeY

kmakhil90, 2018-06-26:
Hi @FayeY,

I have tried this but getting ERROR always. Do we need to give any extra commands before this?

alexdecei, 2019-07-09:
Hi, I find this method very disturbing because it is not secure at all: we allow the connection, **then** we ask for permission, it should be the opposite, isn't it ?
It's like building a house with locks in every indoors doors... But without entrance door.

It seems like I missunderstood something ?

#
For the moment, we can't send` AT+BLEENC `if no device is connected (I get "ERROR").
When I connect my phone to the ESP32, I send ` AT+BLEENC `  ` AT+BLEENCRSP `   `AT+BLEKEYREPLY` 
 and a pairing request is asked.
I can ignore that pairing request during10 seconds and during this time, I can read and write to my ESP.

That is problematic

#
Here is the entire code:
```
AT+BLEINIT=2
LESECPARAM=4,1,16,3,3
```
_waiting for connection..._
```
AT+BLEENC=0,3
AT+BLEENCRSP=0,1
AT+BLEKEYREPLY=0,123456
```
_Pairing request..._

And here I can do what I want without any pairing/permssion/passkey

#
I tried differents parameters for BLESECPARAM and BLEENC.
Did I miss something ?

Thank you.

alexdecei, 2019-07-11:
I have resolved my problem.

In fact, ble secure needs 2 configurations to work, I didn't think that it would need an AT security configuration **plus** a service.csv perm configuration:

> index,uuid_len,uuid,perm,val_max_len,val_cur_len,value
> 0,16,0x2800,0x01,16,16,ED2B4E3A2820492F9507DF165285E831
> 1,16,0x2803,**0x44**,1,1,8
> 2,128,0xED2B4E3B2820492F9507DF165285E831,**0x44**,512,1,30
> 3,16,0x2803,**0x44**,1,1,10
> 4,128,0xED2B4E3C2820492F9507DF165285E831,**0x44**,1,1,20
> 5,16,0x2902,**0x44**,2,2,3031

So if I put 0x11 insted of 0x44, I will never be able to secure my BLE connection:

- Read 0x01
- Read Encrypted 0x02
- Read ENC MIMT 0x04
- Write 0x10
- Write Encrypted 0x20
- Write ENC MIMT 0x40
- Read ENC SIGNED 0x80
- Read ENC SIGNED MIMT 0x100

Maher9494, 2019-11-13:
But what if the ESP32 is the client and you want to establish pairing with a server which has a PIN. 
I think that I will need to execute this command after establishing the connection. Am I right ?
`AT+BLEKEYREPLY=0,123456`