espressif/esp-at #46: No module named yaml

Chunghyon, 2017-11-11:
Hello,
I wanted to use ESP32 BLE Server module and followed the instruction in esp32-at/tools/readme.md.
But... got the error message as below.

*****************************************
Charlie Choi@Charlie MINGW32 /c/esp32/projects/esp32-at/tools
$ python BLEService.py -t GattServiceExample.bin GattServiceExample.csv
Traceback (most recent call last):
  File "BLEService.py", line 6, in <module>
    import yaml
ImportError: No module named yaml
****************************************

PyYAML and xlrd are aleady installed as below...
****************************************
Charlie Choi@Charlie MINGW32 /c/esp32/projects/esp32-at/tools
$ pip install pyyaml xlrd
Requirement already satisfied: pyyaml in c:/msys32/mingw32/lib/python2.7/site-packages
Requirement already satisfied: xlrd in c:/msys32/mingw32/lib/python2.7/site-packages
Charlie Choi@Charlie MINGW32 /c/esp32/projects/esp32-at/tools
$ pip freeze
pyserial==3.4
PyYAML==3.12
xlrd==1.1.0**
****************************************

Could somebody help me to use BLE server moudule via at-command?

Chunghyon, 2017-11-13:
I found the problem doesn't happen when I excute the python in "msys2.exe". 
It happens in "mingw32.exe".
What's the difference?

ustccw, 2017-11-21:
@Chunghyon 
Hi,sorry for the late reply.
msys2.exe is the upgraded version of mingw32.exe, to be compatible with more attributes.
In fact, it also can install and work well on "mingw32.exe" in a right install way.
- for linux:
$ sudo apt-get install python-yaml
- for windows:[maybe a little complex]
following the [readme ](https://github.com/espressif/esp32-at/tree/master/tools)
1. install python 2.7
2. install pip
    - maybe you should set some pip attributes by the guidence
3. install the following python packages with pip: pip install pyyaml xlrd

if failed, try:  // maybe different system configuration would result in the install fail.
1. $ easy_install rosinstall
2. $ easy_install python-yaml
3. $ easy_install xlrd

in my  windows 7 [64 bit] workspace, it works well under the mingw32.exe when i run `make` command.

mws-rmain, 2017-11-29:
I just downloaded the current Windows pre-built toolchain, pulled the current image of esp32-at with git, tried to compile.

Running 'make', I receive the following warning:
`GENCONFIG
WARNING: Toolchain version is not supported: 1.22.0-75-gbaf03c2
Expected to see version: 1.22.0-73-ge28a011
`
and the following error:
`generating AT customized partition bins to /home/Admin/esp32-at/build/customized_partitions
generating ble_data.bin: python /home/Admin/esp32-at/tools/BLEService.py -t /home/Admin/esp32-at/build/customized_partitions/ble_data.bin /home/Admin/esp32-at/components/customized_partitions/raw_data/ble_data/example.csv
Traceback (most recent call last):
  File "K:/msys32/home/Admin/esp32-at/tools/BLEService.py", line 6, in <module>
    import yaml
ImportError: No module named yaml
make: *** [/home/Admin/esp32-at/components/customized_partitions/Makefile.projbuild:43: /home/Admin/esp32-at/build/at_customize.bin] Error 1
`
If 'yaml' is required by Python to complete the build, I'm surprised it's not part of the pre-built toolchain (Python 2.7 _IS_ already installed as part of this toolchain).

From the warning shown above, it appears the pre-built toolchain is newer than the current esp32-at code expects. Possibly this is related to the eventual failure 

???

Or have I missed some step?

ustccw, 2017-11-30:
@mws-rmain 
hi,
tbh, python 2.7 would not pre-install yaml. 
you can lookup it by visit [pyyaml.org](http://pyyaml.org/wiki/PyYAMLDocumentation)
> Starting from the 3.08 release, PyYAML and LibYAML bindings provide a complete support for Python 3
you had better install yaml module firstly according your python and pc environment.

Best regards

mws-rmain, 2017-11-30:
I guess the point I would try to make from my experience is:
Either YAML is required by many of the ESP32 projects, or it is unique to the ESP32-AT project.
- If it is required by all/many/some of the ESP32 projects, then it's not unreasonable to expect that it should be included in the pre-built toolchain that is supposedly configured for ESP32 development.
- if it is required ONLY for ESP32-AT, then I'd suggest it would be appropriate to mention this 'unique' additional requirement in the build instructions, such as on the home page of the project. This information would be equally useful to developers who choose a pre-built toolchain, and developers who assemble their toolchain from scratch.

Since the pre-built toolchain is now more advanced than the ESP32-AT expects and causes the warning I mentioned about 'unsupported toolchain', I expect I may well have to go ahead and assemble a toolchain from scratch.

ustccw, 2017-12-01:
thanks for your suggestion.

Best regards

closed 2018-01-05 by FayeY