espressif/esp-at #47: +++ to terminate UART-WiFi passthrough transmission

Chunghyon, 2017-11-13:
Hello,
According th the at-command instruction, +++ will terminate the passthrough transmission.
However it doesn't seem to work neither in TCP nor in UDP mode.
Please let me know how I can exit from the passthrough mode.

Thank you.

xcguang, 2017-11-14:
Hi, please type `+++` without `\r\n`.

Chunghyon, 2017-11-14:
Yes, I did but passthrough mode is not terminated.

MaJerle, 2017-11-14:
Did you wait for 1second after you sent +++? You have to wait before you can start sending commands.

Chunghyon, 2017-11-16:
hi, MaJerle
Yes... I waited for more than several tens seconds but no change....

pawelzwronek, 2018-01-21:
Did you send it as fast as possible? In manual you can read typing '+++' from keyboard might not be fast enough. Try paste it from clipboard.

xcguang, 2018-03-24:
Hi @Chunghyon ,
Does it work well?

Chunghyon, 2018-03-26:
I made a mistake in my coding. It wroks well.
Sorry for the confusion.

closed 2018-03-26 by Chunghyon