espressif/esp-at #48: esp32-at compile errors

lucaguazzaroni, 2017-11-14:
I just clone this repository and followed the instructions in README.md. But when i run "make all" the log shows this errors. I am on a Windows  environment.

First , compilation starts with this warning about  crosstool.

>$ make flash
WARNING: Toolchain version is not supported: crosstool-NG crosstool-ng-1.22.0-73-ge28a011
Expected to see version: crosstool-NG crosstool-ng-1.22.0-61-gab8375a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Toolchain version is not supported: crosstool-NG crosstool-ng-1.22.0-73-ge28a011
Expected to see version: crosstool-NG crosstool-ng-1.22.0-61-gab8375a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
CC src/bootloader_random.o
CC src/flash_encrypt.o

.
.
.

>CC port/coap_io_socket.o
AR libcoap.a
CXX cxx_guards.o
In file included from c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\xtensa-esp32-elf\bits\gthr.h:148:0,
                 from c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\ext\atomicity.h:35,
                 from c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\basic_string.h:39,
                 from c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\string:52,
                 from c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\random:40,
                 from c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algo.h:66,
                 from c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\algorithm:62,
                 from C:/Users/pasante/Documents/ESP32/esp32-at/esp-idf/components/cxx/cxx_guards.cpp:20:
c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\xtensa-esp32-elf\bits\gthr-default.h:47:9: error: 'pthread_t' does not name a type
 typedef pthread_t __gthread_t;
         ^
c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\xtensa-esp32-elf\bits\gthr-default.h:48:9: error: 'pthread_key_t' does not name a type
 typedef pthread_key_t __gthread_key_t;
         ^
c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\xtensa-esp32-elf\bits\gthr-default.h:49:9: error: 'pthread_once_t' does not name a type
 typedef pthread_once_t __gthread_once_t;
         ^
c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\xtensa-esp32-elf\bits\gthr-default.h:50:9: error: 'pthread_mutex_t' does not name a type
 typedef pthread_mutex_t __gthread_mutex_t;
         ^
c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\xtensa-esp32-elf\bits\gthr-default.h:51:9: error: 'pthread_mutex_t' does not name a type
 typedef pthread_mutex_t __gthread_recursive_mutex_t;
         ^
c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\xtensa-esp32-elf\bits\gthr-default.h:52:9: error: 'pthread_cond_t' does not name a type
 typedef pthread_cond_t __gthread_cond_t;
         ^

.
. more errors
.

> c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\xtensa-esp32-elf\bits\gthr-default.h:787:1: error: expected ',' or ';' before '{' token
 {
 ^
In file included from c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\ext\atomicity.h:35:0,
                 from c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\basic_string.h:39,
                 from c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\string:52,
                 from c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\random:40,
                 from c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algo.h:66,
                 from c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\algorithm:62,
                 from C:/Users/pasante/Documents/ESP32/esp32-at/esp-idf/components/cxx/cxx_guards.cpp:20:
c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\xtensa-esp32-elf\bits\gthr.h:151:27: error: expected declaration before end of line
make[1]: *** [/c/Users/pasante/Documents/ESP32/esp32-at/esp-idf/make/component_wrapper.mk:211: cxx_guards.o] Error 1
make: *** [/c/Users/pasante/Documents/ESP32/esp32-at/esp-idf/make/project.mk:386: cxx-build] Error 2

lucaguazzaroni, 2017-11-14:
In linux compiles ok crosstool
 https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-73-ge28a011-5.2.0.tar.gz

xamix, 2017-11-20:
Hi,

I get the exact same problem with fresh install on Windows 7 64bits.
Any solution for build on Windows?

Regards.

ustccw, 2017-11-21:
@lucaguazzaroni @xamix 
Hi, 
this is because the compiler is not compatible. at the beginning of your compile log, it has warning info:
```
WARNING: Toolchain version is not supported: crosstool-NG crosstool-ng-1.22.0-73-ge28a011
Expected to see version: crosstool-NG crosstool-ng-1.22.0-61-gab8375a
```
so you must use the old compiler, called  **crosstool-NG crosstool-ng-1.22.0-61-gab8375a**
you can download the total toolchain  from https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20170330.zip
the old compiler is under the /opt/ directory, you can copy the compiler to your workpath and export it, of course, for convenience, you can work in this workspace.
you will see the following log if you compile the project successfully:
[image]

xamix, 2017-11-22:
Hello,

Thank you,
For information it is now compiling without any error since the last push with new IDF

ustccw, 2017-11-22:
thanks for your attetion to `esp32-at`.

Best regards

closed 2018-01-05 by FayeY

ahmedawais, 2018-05-10:
hi
i am getting the following error while compling in ubuntu 
Description	Resource	Path	Location	Type
make: *** [component-main-build] Error 2	myapp		 	C/C++ Problem
make[1]: *** [main.o] Error 1	myapp		 	C/C++ Problem
fatal error: esp_gap_ble_api.h: No such file or directory	main.c	/myapp/main	line 28	C/C++ Problem
recipe for target 'component-main-build' failed	myapp		line 467, external location: /home/awais/esp32/esp-idf/make/project.mk	C/C++ Problem
recipe for target 'main.o' failed	myapp		line 285, external location: /home/awais/esp32/esp-idf/make/component_wrapper.mk	C/C++ Problem
#warning "This header is deprecated, please use functions defined in esp_bt.h instead." [-Wcpp]	myapp		line 2, external location: /home/awais/esp32/esp-idf/components/bt/include/bt.h	C/C++ Problem

does anybody know the reason of this ?