espressif/esp-at #49: compile failed - Include err

orshefi, 2017-11-16:
Hi All,

I've been trying to compile the esp32_at example with no success.
All the paths have been set properly.
I have all the necessary tools to compile (I think) because I have managed to compile examples from the esp_idf repo before changing the IDF_PATH to the esp_idf that embedded in the esp32_at directory.

The error is as follows:

`                 from /relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/ext/atomicity.h:35,
                 from /relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/basic_string.h:39,
                 from /relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/string:52,
                 from /relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/random:40,
                 from /relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_algo.h:66,
                 from /relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/algorithm:62,
                 from /relevant_path/esp/esp32-at/esp-idf/components/cxx/./cxx_guards.cpp:20:
/relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:47:9: error: 'pthread_t' does not name a type
 typedef pthread_t __gthread_t;
         ^
/relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:48:9: error: 'pthread_key_t' does not name a type
 typedef pthread_key_t __gthread_key_t;
         ^
/relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:49:9: error: 'pthread_once_t' does not name a type
 typedef pthread_once_t __gthread_once_t;
         ^
/relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:50:9: error: 'pthread_mutex_t' does not name a type
 typedef pthread_mutex_t __gthread_mutex_t;
         ^
/relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:51:9: error: 'pthread_mutex_t' does not name a type
 typedef pthread_mutex_t __gthread_recursive_mutex_t;
         ^
/relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:52:9: error: 'pthread_cond_t' does not name a type
 typedef pthread_cond_t __gthread_cond_t;
         ^
/relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:101:1: error: 'pthread_once' was not declared in this scope
 __gthrw(pthread_once)
 ^
/relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:102:1: error: 'pthread_getspecific' was not declared in this scope
 __gthrw(pthread_getspecific)
 ^
/relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:103:1: error: 'pthread_setspecific' was not declared in this scope
 __gthrw(pthread_setspecific)
 ^
/relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:105:1: error: 'pthread_create' was not declared in this scope
 __gthrw(pthread_create)
 ^
/relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:106:1: error: 'pthread_join' was not declared in this scope
 __gthrw(pthread_join)
 ^
/relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:107:1: error: 'pthread_equal' was not declared in this scope
 __gthrw(pthread_equal)
 ^
/relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:108:1: error: 'pthread_self' was not declared in this scope
 __gthrw(pthread_self)
 ^
/relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:109:1: error: 'pthread_detach' was not declared in this scope
 __gthrw(pthread_detach)
 ^
/relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:111:1: error: 'pthread_cancel' was not declared in this scope
 __gthrw(pthread_cancel)
 ^
/relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:113:1: error: 'sched_yield' was not declared in this scope
 __gthrw(sched_yield)
 ^
/relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:115:1: error: 'pthread_mutex_lock' was not declared in this scope
 __gthrw(pthread_mutex_lock)
 ^
/relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:116:1: error: 'pthread_mutex_trylock' was not declared in this scope
 __gthrw(pthread_mutex_trylock)
 ^
/relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:120:1: error: 'pthread_mutex_unlock' was not declared in this scope
[... 213 lines cut ...]
/relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:786:42: error: '__gthread_recursive_mutex_init_function' declared as an 'inline' variable
 __gthread_recursive_mutex_init_function (__gthread_recursive_mutex_t *__mutex)
                                          ^
/relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:786:42: error: '__gthread_recursive_mutex_t' was not declared in this scope
/relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:786:71: error: '__mutex' was not declared in this scope
 __gthread_recursive_mutex_init_function (__gthread_recursive_mutex_t *__mutex)
                                                                       ^
/relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr-default.h:787:1: error: expected ',' or ';' before '{' token
 {
 ^
In file included from /relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/ext/atomicity.h:35:0,
                 from /relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/basic_string.h:39,
                 from /relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/string:52,
                 from /relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/random:40,
                 from /relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_algo.h:66,
                 from /relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/algorithm:62,
                 from /relevant_path/esp/esp32-at/esp-idf/components/cxx/./cxx_guards.cpp:20:
/relevant_path/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/bits/gthr.h:151:27: error: expected declaration before end of line
/relevant_path/esp/esp32-at/esp-idf/make/component_wrapper.mk:210: recipe for target 'cxx_guards.o' failed
make[1]: *** [cxx_guards.o] Error 1
/relevant_path/esp/esp32-at/esp-idf/make/project.mk:386: recipe for target 'cxx-build' failed
make: *** [cxx-build] Error 2
`

I'll much appriciate your reply

ustccw, 2017-11-21:
@orshefi @xamix
the same issue to [issue 48](https://github.com/espressif/esp32-at/issues/48).
Hi, 
this is because the compiler is not compatible. at the beginning of your compile log, it has warning info:
```
WARNING: Toolchain version is not supported: crosstool-NG crosstool-ng-1.22.0-73-ge28a011
Expected to see version: crosstool-NG crosstool-ng-1.22.0-61-gab8375a
```
so you must use the old compiler, called  **crosstool-NG crosstool-ng-1.22.0-61-gab8375a**
you can download the total toolchain  from https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20170330.zip
the old compiler is under the /opt/ directory, you can copy the compiler to your workpath and export it, of course, for convenience, you can work in this workspace.
you will see the following log if you compile the project successfully:
[image]

orshefi, 2017-11-21:
> esp32_win32_msys2_environment_and_toolchain-20170330.zip

Thank you, the toolchain attached is for windows version, can you attach
the linux toolchain? (sorry for asking I could not find it by myself)

ustccw, 2017-11-21:
@orshefi 

- for 64-bit Linux:

  https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-61-gab8375a-5.2.0.tar.gz

- for 32-bit Linux:

  https://dl.espressif.com/dl/xtensa-esp32-elf-linux32-1.22.0-61-gab8375a-5.2.0.tar.gz

you could see more information in the esp32-at/esp-idf/docs/get-started/linux-setup.rst

closed 2018-01-05 by FayeY