espressif/esp-at #50: wifi: invalid magic number 1 -> at_port.c line 589

ESP32DE, 2017-11-18:
[image]

please check your file!

> /builds/rd/esp-at-core/components/at_core/ .at_port.c

best wishes
rudi ;-)

ESP32DE, 2017-11-21:
again no interesst in this and want no community help.
after few days now i closed this problem and will never touch your AT lib again until you do not open source the code that we can help by self.

closed 2017-11-21 by ESP32DE

xcguang, 2017-11-21:
Hi, 
sorry for the late reply because we are busy in off-site testing these days. 
Which esp-idf do you use? Is it the esp32-at/esp-idf? Do you clone it by ```git clone --recursive https://github.com/espressif/esp32-at.git```?

ustccw, 2017-11-29:
@ESP32DE

you would encounter this error if you set a unmatched ESP-IDF path when compile your esp-at.
the matched esp-idf is under directory `esp32-at/esp-idf`, current commid is 7e8c2a9c00a6fb05cd5da306b62c4474a999b1a2
**Notes:**
#### you should make sure that you have the right commitID, especially, should not use the latest esp-idf 

for set right esp-idf path, you should set as the following:
```
export IDF_PATH=~/esp-at/esp-idf
```
tbh, you need not to set esp-idf in latest esp-at.

Best regards