espressif/esp-at #51: AT command defining end of line

xamix, 2017-11-22:
Hello,

I'm trying to use the actual AT command with Putty but it seem that the AT command interpreter is waiting for '\r\n' to be send before interpreting command.

However, it seem not possible to send it with Putty (ENTER key send only '\r'), so is there a solution to change the end of line for AT command interpreter from '\r\n' to '\r'?

I  tryed by changing CONFIG_NEWLIB_STDIN_LINE_ENDING_LF to 1 but it seem to not using this.
Maybe it's inside the closed source "libat_core.a" ?

Regards.

xcguang, 2017-11-23:
Hi, yes,it is defined in "libat_core.a". Could you try other serial tools, for example, cutecom, SecureCRT?

xamix, 2017-11-23:
Hi,

I changed and used TeraTerm.
I think that the 'libat_core' closed source will be limiting for future dev on my project.
Can we know the reason why this part of the project if closed source?

Anyway it's a good start for easyly test feature, but it seem that implementing my own communication interface will be the way to go...

Regards.

xcguang, 2017-11-24:
Thanks for your advice, We'll consider it in the future.

xcguang, 2018-01-17:
Hi @xamix ,
We have added custom command line terminator, but it just supports one character.
Please try it, you can configure it by `make menuconfig` --> `Components config` --> `AT` --> `AT command terminator support`, then set `AT command terminator`.

xcguang, 2018-03-12:
Hi @xamix ,
Does it work well?

xamix, 2018-03-19:
Hi @xcguang,

I tested it and I confirm that it works well.

Thank you.

closed 2018-03-20 by xcguang