espressif/esp-at #52: wifi: invalid magic number 1,

redheadedrod, 2017-11-28:
I noticed a closed issue had this same problem but wasn't resolved. 

This is a brand new install of ESP-IDF  in past week. I followed the directions for the installing of ESP-AT. 

Here is a partial screen shot of what I am getting.
[image]
After this it gives a Backtrace and reboots.

A couple of items. Since this board is an XBEE format board running on top of an Arduino board the only UART available to the Arduino is UART0 so everything is done on UART0. 

Not sure about what I SHOULD be setting everything to I found a data sheet that looks like the same as the processor on the schematics for this Xbee board. So I used the following settings:
[image]

The other item is that I have the console stuff still going to UART0 and is turned on. Not sure if this is a collision in the sw and not allowed for both the console and the communications to go through the same UART?

xcguang, 2017-11-28:
Hi，How do you clone esp32-at? Is it `git clone --recursive https://github.com/espressif/esp32-at.git?`
If so, please try `cd esp32-at`, then `make defconfig` and `make`.

redheadedrod, 2017-11-28:
Yes, I used git clone --recursive for both esp-idf and for esp32-at. I will do as requested and report my result.

redheadedrod, 2017-11-28:
Compiled and only changed the uart to uart 0 and flashed... Same error. My Issue MAY be related to getting the uart settings right but not sure what they should be? I am assuming the pin numbers are the GPIO pins?

ustccw, 2017-11-29:
＠redheadedrod

ustccw, 2017-11-29:
@redheadedrod 
you would encounter this error if you set a unmatched ESP-IDF path when compile your esp-at.
the matched esp-idf is under directory `esp32-at/esp-idf`, current commid is 7e8c2a9c00a6fb05cd5da306b62c4474a999b1a2
**Notes:**
#### you should make sure that you have the right commitID, especially, should not use the latest esp-idf 

for set right esp-idf path, you should set as the following:
```
export IDF_PATH=~/esp-at/esp-idf
```
tbh, you need not to set esp-idf in latest esp-at.

Best regards

redheadedrod, 2017-11-29:
Yes I am using latest ESP-IDF and IDF_PATH is set to that one.

Your last line:

> tbh, you need not to set esp-idf in latest esp-at.

This confuses me. If I change or remove the path should I be fine then? 
If I read between the lines I would guess that if the path is there it will be used, if not then it will default to using the one included?

ustccw, 2017-11-30:
if you did not set a default path in your shell, the esp-at would adopt the following esp-idf, which defined at `esp32-at/Makefile`
```
export IDF_PATH ?= $(PWD)/esp-idf
```
of course, you also can set `IDF_PATH=~/esp-at/esp-idf`  in your shell again.
#### the only point you need care is that you should not set the IDF_PATH to the latest esp-idf, please know.

ustccw, 2017-11-30:
@redheadedrod

redheadedrod, 2017-11-30:
Thank you for the help and rapid response. 
I set the IDF_PATH to the one within the ESP32-AT directory and flashed that. No errors and I get a ready prompt. So it looks like everything is working. 

Thanks again!

Now to actually work with it and get it to do things.

ustccw, 2017-11-30:
thank you for your attention to esp32-at.

Best regards

johnny0099, 2017-12-24:
I have the same issue, i'm setting IDF path like this:

export IDF_PATH=/c/esp32/esp32-at/esp-idf (that is the IDF in the esp-at folder, not the last one).

Here my steps:

git clone --recursive https://github.com/espressif/esp32-at.git   (final path on my disk is c:/esp32/esp32-at/)

download and install the following toolchain:
esp32_win32_msys2_environment_and_toolchain-20171123.zip

Using mingw32.exe do the following:

rm sdkconfig  (remove config)
make defconfig  (load default config)
make menuconfig  (changing UART pin assignment and COM for flash tool)
make flash  (complile and flash)

I'm getting ESP-WROOM32 in reset loop:

Rebooting...
ets Jun  8 2016 00:22:57
 
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5728
ho 0 tail 12 room 4
load:0x40078000,len:0
load:0x40078000,len:13804
entry 0x40079030
.[0;32mI (74) boot: Detected ESP32.[0m
.[0;32mI (34) boot: ESP-IDF v3.0-rc1-2-g0ebae99a-dirty 2nd stage bootloader.[0m
.[0;32mI (34) boot: compile time 03:42:09.[0m
.[0;32mI (34) boot: Enabling RNG early entropy source....[0m
.[0;32mI (40) boot: SPI Speed      : 40MHz.[0m
.[0;32mI (45) boot: SPI Mode       : DIO.[0m
.[0;32mI (49) boot: SPI Flash Size : 4MB.[0m
.[0;32mI (53) boot: Partition Table:.[0m
.[0;32mI (56) boot: ## Label            Usage          Type ST Offset   Length.[0m
.[0;32mI (63) boot:  0 phy_init         RF data          01 01 0000f000 00001000.[0m
.[0;32mI (71) boot:  1 otadata          OTA data         01 00 00010000 00002000.[0m
.[0;32mI (78) boot:  2 nvs              WiFi data        01 02 00012000 0000e000.[0m
.[0;32mI (86) boot:  3 at_customize     unknown          40 00 00020000 000e0000.[0m
.[0;32mI (93) boot:  4 ota_0            OTA app          00 10 00100000 00180000.[0m
.[0;32mI (101) boot:  5 ota_1            OTA app          00 11 00280000 00180000.[0m
.[0;32mI (108) boot: End of partition table.[0m
.[0;32mI (113) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x20414 (132116) map.[0m
.[0;32mI (167) esp_image: segment 1: paddr=0x0012043c vaddr=0x3ffc0000 size=0x03320 ( 13088) load.[0m
.[0;32mI (173) esp_image: segment 2: paddr=0x00123764 vaddr=0x40080000 size=0x00400 (  1024) load.[0m
.[0;32mI (175) esp_image: segment 3: paddr=0x00123b6c vaddr=0x40080400 size=0x0c4a4 ( 50340) load.[0m
.[0;32mI (204) esp_image: segment 4: paddr=0x00130018 vaddr=0x400d0018 size=0xd464c (869964) map.[0m
.[0;32mI (506) esp_image: segment 5: paddr=0x0020466c vaddr=0x4008c8a4 size=0x077b0 ( 30640) load.[0m
.[0;32mI (519) esp_image: segment 6: paddr=0x0020be24 vaddr=0x400c0000 size=0x00064 (   100) load.[0m
.[0;32mI (531) boot: Loaded app from partition at offset 0x100000.[0m
.[0;32mI (531) boot: Disabling RNG early entropy source....[0m
.[0;32mI (532) cpu_start: Pro cpu up..[0m
.[0;32mI (535) cpu_start: Single core mode.[0m
.[0;32mI (540) heap_init: Initializing. RAM available for dynamic allocation:.[0m
.[0;32mI (547) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM.[0m
.[0;32mI (553) heap_init: At 3FFD4258 len 0000BDA8 (47 KiB): DRAM.[0m
.[0;32mI (559) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM.[0m
.[0;32mI (565) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM.[0m
.[0;32mI (572) heap_init: At 40094054 len 0000BFAC (47 KiB): IRAM.[0m
.[0;32mI (578) cpu_start: Pro cpu start user code.[0m
.[0;32mI (596) cpu_start: Starting scheduler on PRO CPU..[0m
.[0;32mI (687) uart: queue free spaces: 10.[0m
E (689) wifi: invalid magic number 1, call WIFI_INIT_CONFIG_DEFAULT to init config
E (690) wifi: wifi_init 1395 ret=258
ESP_ERROR_CHECK failed: esp_err_t 0x102 at 0x40133ef3
file: "/builds/rd/esp-at-core/components/at_core/./at_port.c" line 589
func: initialise_wifi
expression: esp_wifi_init(&cfg)
 
Backtrace: 0x4008a71c:0x3ffd6bc0 0x4008aa89:0x3ffd6be0 0x40133ef3:0x3ffd6c00 0x40134763:0x3ffd6c60 0x401156a7:0x3ffd6ca0 0x4011500b:0x3ffd6ce0 0x400d0902:0x3ffd6d00
 
Rebooting...
ets Jun  8 2016 00:22:57

Any suggestion? What I'm dooing wrong?

mxwang101, 2017-12-24:
how to check "commid is 7e8c2a9c00a6fb05cd5da306b62c4474a999b1a2" ? please help. thanks

mxwang101, 2017-12-24:
And I have same problem like johnny0099 commented. how to fix?

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5724
load:0x40078000,len:0
load:0x40078000,len:13804
entry 0x40079030
[0;32mI (68) boot: Detected ESP32[0m
[0;32mI (32) boot: ESP-IDF v3.0-rc1-2-g0ebae99a 2nd stage bootloader[0m
[0;32mI (32) boot: compile time 01:17:03[0m
[0;32mI (32) boot: Enabling RNG early entropy source...[0m
[0;32mI (38) boot: SPI Speed      : 40MHz[0m
[0;32mI (42) boot: SPI Mode       : DIO[0m
[0;32mI (46) boot: SPI Flash Size : 4MB[0m
[0;32mI (50) boot: Partition Table:[0m
[0;32mI (54) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (61) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (69) boot:  1 otadata          OTA data         01 00 00010000 00002000[0m
[0;32mI (76) boot:  2 nvs              WiFi data        01 02 00012000 0000e000[0m
[0;32mI (83) boot:  3 at_customize     unknown          40 00 00020000 000e0000[0m
[0;32mI (91) boot:  4 ota_0            OTA app          00 10 00100000 00180000[0m
[0;32mI (98) boot:  5 ota_1            OTA app          00 11 00280000 00180000[0m
[0;32mI (106) boot: End of partition table[0m
[0;31mE (110) boot: ota data partition invalid and no factory, will try all partitions[0m
[0;32mI (118) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x20414 (132116) map[0m
[0;32mI (173) esp_image: segment 1: paddr=0x0012043c vaddr=0x3ffc0000 size=0x03320 ( 13088) load[0m
[0;32mI (179) esp_image: segment 2: paddr=0x00123764 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (180) esp_image: segment 3: paddr=0x00123b6c vaddr=0x40080400 size=0x0c4a4 ( 50340) load[0m
[0;32mI (210) esp_image: segment 4: paddr=0x00130018 vaddr=0x400d0018 size=0xd464c (869964) map[0m
[0;32mI (512) esp_image: segment 5: paddr=0x0020466c vaddr=0x4008c8a4 size=0x077b0 ( 30640) load[0m
[0;32mI (525) esp_image: segment 6: paddr=0x0020be24 vaddr=0x400c0000 size=0x00064 (   100) load[0m
[0;32mI (537) boot: Loaded app from partition at offset 0x100000[0m
[0;32mI (537) boot: Disabling RNG early entropy source...[0m
[0;32mI (537) cpu_start: Pro cpu up.[0m
[0;32mI (541) cpu_start: Single core mode[0m
[0;32mI (545) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (552) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (558) heap_init: At 3FFD4258 len 0000BDA8 (47 KiB): DRAM[0m
[0;32mI (564) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (571) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (577) heap_init: At 40094054 len 0000BFAC (47 KiB): IRAM[0m
[0;32mI (583) cpu_start: Pro cpu start user code[0m
[0;32mI (602) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (611) uart: queue free spaces: 10[0m
E (614) wifi: invalid magic number 1, call WIFI_INIT_CONFIG_DEFAULT to init config
E (614) wifi: wifi_init 1395 ret=258
ESP_ERROR_CHECK failed: esp_err_t 0x102 at 0x40133ef3
file: "/builds/rd/esp-at-core/components/at_core/./at_port.c" line 589
func: initialise_wifi
expression: esp_wifi_init(&cfg)

Backtrace: 0x4008a71c:0x3ffd6bc0 0x4008aa89:0x3ffd6be0 0x40133ef3:0x3ffd6c00 0x40134763:0x3ffd6c60 0x401156a7:0x3ffd6ca0 0x4011500b:0x3ffd6ce0 0x400d0902:0x3ffd6d00

Rebooting...
ets Jun  8 2016 00:22:57

johnny0099, 2017-12-24:
Crazy!!! Finally solved this problem, just wasting 3 days. Hope that the Espressif team take seriously in charge this situation!!!

There is a problem with ESP-IDF 3.0 and/or TOOLCHAIN 20171123 and/or latest ESP-AT source files.
I was able to compile and having ESP-AT working with the following configuration:

Source files esp32-at-1.0.0.0
Toolchain 20170330
Using IDF v2.1.1
IDF_PATH=/your local path/esp-idf-v2.1.1  (should point to idf-v2.1.1 path)

HERE WHERE TO DOWNLOAD

ESP32 AT v1.0.0.0
https://github.com/espressif/esp32-at/archive/v1.0.0.0.zip

TOOLCHAIN 20170330 (crosstool-NG crosstool-ng-1.22.0-61-gab8375a)
https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20170330.zip

ESP-IDF v2.1.1
https://github.com/espressif/esp-idf/releases/download/v2.1.1/esp-idf-v2.1.1.zip
https://github.com/espressif/esp-idf/archive/v2.1.1.zip

Install Using Git (Preferred for dependecies)
git clone https://github.com/espressif/esp-idf.git esp-idf-v2.1.1
cd esp-idf-v2.1.1/
git checkout v2.1.1
git submodule update --init --recursive

mxwang101, 2017-12-24:
Yes，You are right. The old IDF 2.1.1 work.  But IDF 3.0 not work.  Hope esp32-at team can fix as soon. 

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 johnny0099
发送时间: 2017年12月25日 4:02
收件人: espressif/esp32-at
抄送: mxwang101; Comment
主题: Re: [espressif/esp32-at] wifi: invalid magic number 1, (#52)

 

Crazy!!! Finally solved this problem, just wasting 3 days. Hope that the Espressif team take seriously in charge this situation!!!

There is a problem with ESP-IDF 3.0 and/or TOOLCHAIN 20171123 and/or latest ESP-AT source files.
I was able to compile and having ESP-AT working with the following configuration:

Source files esp32-at-1.0.0.0
Toolchain 20170330
Using IDF v2.1.1
IDF_PATH=/your local path/esp-idf-v2.1.1 (should point to idf-v2.1.1 path)

HERE WHERE TO DOWNLOAD

ESP32 AT v1.0.0.0
https://github.com/espressif/esp32-at/archive/v1.0.0.0.zip

TOOLCHAIN 20170330 (crosstool-NG crosstool-ng-1.22.0-61-gab8375a)
https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20170330.zip

ESP-IDF v2.1.1
https://github.com/espressif/esp-idf/releases/download/v2.1.1/esp-idf-v2.1.1.zip
https://github.com/espressif/esp-idf/archive/v2.1.1.zip

Install Using Git (Preferred for dependecies)
git clone https://github.com/espressif/esp-idf.git esp-idf-v2.1.1
cd esp-idf-v2.1.1/
git checkout v2.1.1
git submodule update --init --recursive

FayeY, 2017-12-28:
Hi, so sorry for the inconvenience. We have updated the [README](https://github.com/espressif/esp32-at).

> Since we updated the toolchain recently, it is not compatible with the old version. Please use the toolchain we provided in the esp32-at/esp-idf/docs/get-started/linux-setup.rst and esp32-at/esp-idf/docs/get-started/windows-setup.rst to build your ESP32 AT project.

closed 2017-12-28 by FayeY

mxwang101, 2017-12-29:
NOT FIX! SAME ERROR!!

 

Follow "new" readme, download the  <https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20171123.zip> https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20171123.zip,

and install, and got same error“E (701) wifi: invalid magic number 1,”

 

..............................

entry 0x40079030

_[0;32mI (74) boot: Detected ESP32_[0m

_[0;32mI (34) boot: ESP-IDF v3.0-rc1-2-g0ebae99a-dirty 2nd stage bootload

er_[0m

..............................

 

E (701) wifi: invalid magic number 1, call WIFI_INIT_CONFIG_DEFAULT to init config

E (702) wifi: wifi_init 1395 ret=258

ESP_ERROR_CHECK failed: esp_err_t 0x102 at 0x40133fd3

file: "/builds/rd/esp-at-core/components/at_core/./at_port.c" line 589

func: initialise_wifi

expression: esp_wifi_init(&cfg)

 

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 FayeY
发送时间: 2017年12月28日 11:33
收件人: espressif/esp32-at
抄送: mxwang101; Comment
主题: Re: [espressif/esp32-at] wifi: invalid magic number 1, (#52)

 

Closed #52 <https://github.com/espressif/esp32-at/issues/52> .

FayeY, 2017-12-29:
Hi @mxwang101 , 

1. Are you using the toolchain we provided in the [esp32-at/esp-idf/docs/get-started/linux-setup.rst](https://github.com/espressif/esp-idf/blob/0ebae99ab072fb91fd5223869d786f319db2e274/docs/get-started/linux-setup.rst) and [esp32-at/esp-idf/docs/get-started/windows-setup.rst](https://github.com/espressif/esp-idf/blob/0ebae99ab072fb91fd5223869d786f319db2e274/docs/get-started/windows-setup.rst)?

2. Did you clone the AT software by command `git submodule update --init --recursive`?

The most possible reason is the second one, please clone the project according to the [README](https://github.com/espressif/esp32-at).

mxwang101, 2017-12-29:
Yes, I do " git submodule update --init --recursive", but nothing change like:

 

 

wangxin@wangxin-PC MINGW32 /c/esp/esp32-at

$ git submodule update --init --recursive

 

wangxin@wangxin-PC MINGW32 /c/esp/esp32-at

$

 

and same error again.

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 FayeY
发送时间: 2017年12月29日 11:44
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] wifi: invalid magic number 1, (#52)

 

Hi @mxwang101 <https://github.com/mxwang101>  ,

1.  Are you using the toolchain we provided in the esp32-at/esp-idf/docs/get-started/linux-setup.rst <https://github.com/espressif/esp-idf/blob/0ebae99ab072fb91fd5223869d786f319db2e274/docs/get-started/linux-setup.rst>  and esp32-at/esp-idf/docs/get-started/windows-setup.rst <https://github.com/espressif/esp-idf/blob/0ebae99ab072fb91fd5223869d786f319db2e274/docs/get-started/windows-setup.rst> ?

2.  Did you clone the AT software by command git submodule update --init --recursive?

The most possible reason is the second one, please clone the project according to the README <https://github.com/espressif/esp32-at> .

xcguang, 2017-12-29:
Hi @mxwang101 ,
Please change directory to `esp32-at/esp-idf`, and check commit id by `git log`. 
The correct commit id should be `0ebae99`.

redheadedrod, 2017-12-29:
I can assure you that if you do as suggested and insure your using the right esp-idf it does compile and install. If your getting that error it is due to it using the wrong version.

mxwang101, 2017-12-29:
how to check commit id

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 xcguang
发送时间: 2017年12月29日 13:08
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] wifi: invalid magic number 1, (#52)

 

Hi @mxwang101 <https://github.com/mxwang101>  ,
Please change directory to esp32-at/esp-idf, and check commit id by git log.
The correct commit id should be 0ebae99.

mxwang101, 2017-12-29:
OK，commit 0ebae99ab072fb91fd5223869d786f319db2e274 (HEAD)

 

Same Error

 

wangxin@wangxin-PC MINGW32 /c/esp/esp32-at/esp-idf

 

$ git log

 

commit 0ebae99ab072fb91fd5223869d786f319db2e274 (HEAD)

 

Merge: 043ef326 6e0f9057

Author: Angus Gratton <angus@espressif.com>

Date:   Mon Dec 18 06:43:17 2017 +0800

 

    Merge branch 'bugfix/windows_10_idf_monitor_redux_v30' into 'release/v3.0'

 

    idf_monitor: Fix Windows 10 bug in cases where second console write also fails (v3.0 backport)

 

    See merge request !1676

 

 

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 xcguang
发送时间: 2017年12月29日 13:08
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] wifi: invalid magic number 1, (#52)

 

Hi @mxwang101 <https://github.com/mxwang101>  ,
Please change directory to esp32-at/esp-idf, and check commit id by git log.
The correct commit id should be 0ebae99.

mxwang101, 2017-12-29:
Is this right one ? It does compile OK, BUT Same error.

 

commit 0ebae99ab072fb91fd5223869d786f319db2e274 (HEAD)

 

 

 

wangxin@wangxin-PC MINGW32 /c/esp/esp32-at/esp-idf

$ git log

commit 0ebae99ab072fb91fd5223869d786f319db2e274 (HEAD)

Merge: 043ef326 6e0f9057

Author: Angus Gratton <angus@espressif.com>

Date:   Mon Dec 18 06:43:17 2017 +0800

 

    Merge branch 'bugfix/windows_10_idf_monitor_redux_v30' into 'release/v3.0'

 

    idf_monitor: Fix Windows 10 bug in cases where second console write also fails (v3.0 backport)

 

    See merge request !1676

 

commit 6e0f905761f6185ef27c46f8b374f68265a2db68

Author: Angus Gratton <angus@espressif.com>

Date:   Fri Dec 15 16:45:44 2017 +1100

 

    idf_monitor: Fix Windows 10 bug in cases where second console write also fails

 

    Although in my tests the second write always passes, people have reported

    different results.

 

    Closes https://github.com/espressif/esp-idf/issues/1136 (again)

 

commit 043ef32651fada9f66362949deecfe101c636e6c (tag: v3.0-rc1)

Author: Tian Hao <tianhao@espressif.com>

Date:   Thu Dec 7 21:48:27 2017 +0800

 

    component/btdm : change bt.h name to esp_bt.h to form the name prefix

 

commit eae3b45170193d89b8d929361b0a8e4fc15472ae

Author: krzychb <krzychb@gazeta.pl>

Date:   Mon Dec 4 20:22:24 2017 +0100

 

    LEDC Driver: Added back original definitions of 'duty_resolution' and 'clock_divider'. This update is to provide backward compatibility with ESP-IDF 2.1.

 

commit a1c79bbc7bbcfc5dd87f0f86c8ba85b36b5c6114

Author: krzychb <krzychb@gazeta.pl>

Date:   Mon Dec 4 06:10:50 2017 +0100

 

    Pin descriptions, overview diagram and board dimensions carried over from ESP32-PICO-KIT_Datasheet_EN.pdf. The datasheet looks redundant and will not be used.

 

commit cc46b5054ab3cb74aa7f8fabf5cc43acf9c455a3

Author: zhangyanjiao <zhangyanjiao@espressif.com>

Date:   Mon Dec 4 19:23:43 2017 +0800

 

    when netconn created directly,netconn_delete() will not call netconn_free(),

:

 

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 redheadedrod
发送时间: 2017年12月29日 13:10
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] wifi: invalid magic number 1, (#52)

 

I can assure you that if you do as suggested and insure your using the right esp-idf it does compile and install. If your getting that error it is due to it using the wrong version.

mxwang101, 2017-12-29:
Done everything still get: No Fix。

 

Switch back to old one。 

 

_[0;32mI (698) uart: queue free spaces: 10_[0m

E (701) wifi: invalid magic number 1, call WIFI_INIT_CONFIG_DEFAULT to init config

E (702) wifi: wifi_init 1395 ret=258

ESP_ERROR_CHECK failed: esp_err_t 0x102 at 0x40133ef3

file: "/builds/rd/esp-at-core/compon

ents/at_core/./at_port.c" line 589

func: initialise_wifi

expression: esp_wifi_init(&cfg)

 

Backtrace: 0x4008a71c:0x3ffd6bc0 0x4008aa89:0x3ffd6be0 0x40133ef3:0x3ffd6c00 0x40134763:0x3ffd6c60 0x401156a7:0x3ffd6ca0 0x4011500b:0x3ffd6ce0 0x400d0902:0x3ffd6d00

 

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 redheadedrod
发送时间: 2017年12月29日 13:10
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] wifi: invalid magic number 1, (#52)

 

I can assure you that if you do as suggested and insure your using the right esp-idf it does compile and install. If your getting that error it is due to it using the wrong version.

FayeY, 2017-12-29:
Hi, sorry that we cannot reproduce your problem, could you provide your entire project for debugging? Or at least the AT lib you used. Thanks.

mxwang101, 2017-12-29:
OK, here is lib. Please check if OK?. 

 

I am no ideal what is  "invalid magic number 1" and where is " file: "/builds/rd/esp-at-core/compon

ents/at_core/./at_port.c""

 

Thanks

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 FayeY
发送时间: 2017年12月29日 16:22
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] wifi: invalid magic number 1, (#52)

 

Hi, sorry that we cannot reproduce your problem, could you provide your entire project for debugging? Or at least the AT lib you used. Thanks.

FayeY, 2017-12-29:
Hi @mxwang101 , have you attached the lib?

mxwang101, 2017-12-29:
Yes，in the zip file. send again.

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 FayeY
发送时间: 2017年12月29日 18:00
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] wifi: invalid magic number 1, (#52)

 

Hi @mxwang101 <https://github.com/mxwang101>  , have you attached the lib?

FayeY, 2017-12-29:
Hi @mxwang101 , please visit the issue page https://github.com/espressif/esp32-at/issues/52, there is no attachment..

mxwang101, 2017-12-29:
OK, how did I can send the zip file to you? I send by email attachment..

mxwang101, 2017-12-29:
Could I send you email as attachment?

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 FayeY
发送时间: 2017年12月29日 18:16
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] wifi: invalid magic number 1, (#52)

 

Hi @mxwang101 <https://github.com/mxwang101>  , please visit the issue page #52 <https://github.com/espressif/esp32-at/issues/52> , there is no attachment..

mxwang101, 2017-12-30:
at lab  attachment as at.rar. 

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 FayeY
发送时间: 2017年12月29日 18:16
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] wifi: invalid magic number 1, (#52)

 

Hi @mxwang101 <https://github.com/mxwang101>  , please visit the issue page #52 <https://github.com/espressif/esp32-at/issues/52> , there is no attachment..

mxwang101, 2017-12-30:
[components.zip](https://github.com/espressif/esp32-at/files/1594270/components.zip)

mxwang101, 2017-12-31:
Whole problems project zip. can not be send. It is big 210MB

xcguang, 2018-01-02:
Hi @mxwang101 ,
The libat_core.a you used is compiled based on idf2.1. 
Please clone the latest project by `git clone --recursive https://github.com/espressif/esp32-at.git`, then run `make defconfig` and `make` to compile it.

mxwang101, 2018-01-03:
Hi, xcguang, Thanks, it is work fine now.

 

But I find the file( at_task.c in idf2.1) was miss in idf3.0?  

 

We add program in at_task.c to auto run list of at commands,  So how to do in idf3.0?  Please help & give us examples.

 

Thanks again,

 

Michael Wang

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 xcguang
发送时间: 2018年1月2日 13:41
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] wifi: invalid magic number 1, (#52)

 

Hi @mxwang101 <https://github.com/mxwang101>  ,
The libat_core.a you used is compiled based on idf2.1.
Please clone the latest project by git clone --recursive https://github.com/espressif/esp32-at.git, then run make defconfig and make to compile it.

xcguang, 2018-01-03:
You can add it in `esp32-at/main/interface/uart/at_uart_task.c`.

mxwang101, 2018-01-03:
OK，I will try it. thanks.
On 01/03/2018 20:19, xcguang wrote:

You can add it in esp32-at/main/interface/uart/at_uart_task.c.