espressif/esp-at #53: ESP32-AT receiving UDP

zavovi, 2017-12-06:
Hello,
I want use this application for my project and I cannot receive UDP packet. I tried create server: AT+CIPSERVER=1,1234 and receive UDP packets on the port 123, but no packets received. I see packets on the wireshark. TCP is OK. 

And other question, is it possible send/receive multicast UDP packets? For example in address 224.0.0.251 and port 5353. I know, that there is MDNS service, but I need more than MDNS. 

Thank you,
Vilem

xcguang, 2017-12-07:
Hi @zavovi ,
AT+CIPSERVER is to create TCP server, and you should send AT+CIPSTART="UDP",xxx,xxx to create UDP server or client. 
Now we can support MDNS, and you can send/receive multicast UDP packets.

zavovi, 2017-12-07:
Hi @xcguang,
Thank you, now, UDP is working. But when I tried UDP multicast (AT+CIPSTART="UDP","224.0.0.251",5353,5353,0), and send UDP data from PC to 224.0.0.251, then It nothing received in esp32. What is it needed for receiving multicast?
Thanks.

zavovi, 2017-12-07:
Hi @xcguang,
I discovered new behavior. When I tried use AT+MDNS=1,1,"somethig","any",5353 and after that use UDP multicast on other port, that is ok. When I disabled AT+MDNS=0,1 and tried the same port, everything is OK, why? Is it possible add function for add/remove multicast group? 

Thank you.

xcguang, 2017-12-08:
Hi @zavovi ,
Sorry, I have checked it again,  AT+CIPSTART cannot create UDP multicast. 
But it enables UDP multicast when sending AT+MDNS, so after that , it works well.

zavovi, 2017-12-08:
Hi @xcguang,
Thank you very much! I have one other question. Is it possible change Information elements in wifi AP via AT commands?

xcguang, 2017-12-10:
Hi @zavovi ,
Could you give an example on it?

zavovi, 2017-12-10:
Hi @xcguang 
For example, in API reference manual (http://esp-idf.readthedocs.io/en/latest/api-reference/wifi/esp_wifi.html) function "esp_wifi_set_vendor_ie".

xcguang, 2017-12-12:
Hi @zavovi ，
Thank you very much for your suggestion, we will seriously consider it. If there is any progress, we will   reply here to you.

zavovi, 2017-12-12:
Hi @xcguang ,
Thank you very much. It is the good news. I am now working on integrating the ESP32-AT to my project, and I doscovered two additional questions. 

1. Is it possible increase maximum TCP connections? I am using twu UDP servers and one TCP server. I want accept up to 10 TCP connections. Is it possible?

2. When I accepted socket (0, CONNECT in AT), Is it possible get IP address from remote device? 

For example, this is my project: http://zavavov.vzap.eu/en/tcs-en/

Thank you.

MaJerle, 2017-12-12:
@zavovi and @xcguang,

https://github.com/espressif/ESP8266_NONOS_SDK/issues/63

zavovi, 2017-12-12:
Hi @MaJerle ,
Thank you! Now, question number 1? 

And now, when I received multicast UPD, I see this: +IPD,2,101,224.0.0.251,5353:, Why I not see the right IP address of the client, which sent the data? And in some commands is IP address in quotes and there is not, why?

MaJerle, 2017-12-12:
@zavovi,

Im not ESP guy. But how you know you dont see correct one?

This is my project: https://github.com/MaJerle/ESP_AT_Lib

zavovi, 2017-12-12:
Hi @MaJerle ,
The IP address is incorrect, because there is multicast IP address, but the data was sent from endpoint (e.g. 192.168.0.102). 

Your project looks very good, but I am not using RTOS, my project is bare-metal.

MaJerle, 2017-12-12:
@zavovi,

This is starting point to optimize everything. Later I will release non-RTOS support.

zavovi, 2017-12-14:
Hi all,
one additional question. When I received classic UDP data: "+IPD,3,9,0.0.0.0,21105:", why the IP address is NULL?

xcguang, 2017-12-26:
@zavovi ,
Could you give us more information to reproduce it? Thanks.

zavovi, 2018-01-02:
I am using these commands:
AT+MDNS=1,1,"tcs","_tcs",5353
AT+MDNS=0,1
AT+CIPSTART=1,"UDP","224.0.0.251",5353,5353

After these commands I sent data from Android device to multicast IP address 224.0.0.251 and in ESP32 I received this data: "+IPD,3,9,0.0.0.0,21105:......". I see the IP address "0.0.0.0", why? I want see the IP address of the Android device.

Thank you

zavovi, 2018-01-08:
Hi @xcguang,
I tried UDP again and, when I used this command: 
AT+CIPSTART=1,"UDP","",21105,21105
then UDP is working but I receiving data without IP address: "+IPD,3,9,0.0.0.0,21105:......".

Another question: Is it possible increase maximum connections? I am using 2 UDP connections and I need up to 10 TCP connections.

Thank you.

xcguang, 2018-01-17:
Hi @zavovi , You can set the max connections  by calling `esp_at_module_init`.

zavovi, 2018-01-17:
Hi @xcguang ,
is it possible do it from AT command via UART? If I would change it in code, I would recompile application... 

And what is the real maximum in this function?

Thank you.

zavovi, 2018-03-17:
Hi @xcguang, I am sorry about too many questions now. I am working on implementation ESP32 into my project very hard. I still not received right IP address from UDP received data.

How to reproduce:
AT+CIPSTART=1,"UDP","10.0.0.8",21105,21105

When I sent data from IP 10.0.0.9 I received "+IPD,3,9,10.0.0.8,21105:......", I think, it is not really good, or?

Many thanks for your support. If is it anything, what can I do for you, let's tel me :-)

xcguang, 2018-03-20:
@zavovi ,
Could you give some log? Or how can we reproduce it?

zavovi, 2018-03-20:
@xcguang, reproduce it is very simple. 
1. Connect to an WiFi router with ESP32.
2. Start UDP server (e.g. AT+CIPSTART=1,"UDP","10.0.0.8",21105,21105)
3. Send UDP data from PC or another device, with other IP address (not the same as command above - 10.0.0.8)
4. Received "+IPD,3,9,10.0.0.8,21105:......", but I expected IP address from host, which sent the data.

I am using Hercules Setup Utility for sending UDP data from Windows to ESP32. 
Is it ok?

xcguang, 2018-03-27:
Hi @zavovi ,
Please try it again using the latest version. If it works well for you, please update the issue.

zavovi, 2018-03-27:
Hi @xcguang,
It is working! Thank you very much!

closed 2018-03-27 by zavovi