espressif/esp-at #54: ESP32 as TCP Client in UART-Wi-Fi Passthrough

JPast, 2017-12-13:
Hello,

I want to use the ESP32-AT-Command-Set to set up a WIFI-TCP-Communication with an Android Smartphone. A second ESP32 is configured as Access Point. The connection of the devices works without problems. But in transparent transmission mode I get no response when TCP-Connection gets closed e.g. if the Smartphone shuts down. 
Only the Programming-UART puts messages when WIFI-Connection is closed or opened again.

For TCP-Connection I use AT+CIPSTART="TCP","192.168.101.110",1000,0 so TCP keep-alive is disabled.
But TCP is kept alive and Transmission-UART don´t put messages like CLOSED. Now I don´t know how to detect a missing TCP-Connection occurring while run-time. Is this detection possible with AT-Commands? I don´t want to leave transparrent transmission before detection.

For my application it would be enough to detect WIFI-Connection and let TCP keep-alive enabled.
Is it possible to build in a GPIO-Port which is toggling when the AT-Command-Core puts out the WIFI-Connection messages? I think therefore I need access to the libat_core.a. Why is this closed source?
 

Thank you,

JPast

xcguang, 2017-12-26:
Hi @JPast ,
If the message is printed in transparent transmission mode, It confuses message with data. 
Maybe we can post an event in `at_status_callback`, then you add code to check it, and print a message or control a gpio.

JPast, 2017-12-27:
Hi @xcguang ,

thank you for your answer.
This would be a solution for my problem. It would be nice to have two events to handle in `at_status_callback` . One for a lost WiFi-Connection and another one for a lost TCP-Connection.
How long would it take to implement this events in libat_core.a?

xcguang, 2017-12-29:
Hi @JPast ,
We will think about it after the New Year's day. If there is any update, we will reply here.

xcguang, 2018-01-25:
Hi @JPast ,
We have added some cb for connection, please try it  if possible. The attachment is the at lib for test. 

```c
typedef struct {
    int32_t (*recv_data)(uint8_t*data,int32_t len);
    void (*connect_cb)(void);
    void (*disconnect_cb)(void);
} esp_at_custom_net_ops_struct;

bool esp_at_custom_net_ops_regist (int32_t link_id,esp_at_custom_net_ops_struct* ops);

void test_connect_cb(void)
{
    printf("test_connect_cb\r\n");
}

void test_disconnect_cb(void)
{
    printf("test_disconnect_cb\r\n");
}
```
```c
void at_task_init(void)
{
    ......
    esp_at_custom_net_ops_struct custom_net_ops = {
        .recv_data = NULL,
        .connect_cb = test_connect_cb,
        .disconnect_cb = test_disconnect_cb
    };

    ......
    esp_at_module_init (CONFIG_LWIP_MAX_SOCKETS - 1, version);  // reserved one for server
    free(version);

    esp_at_custom_net_ops_regist(0,&custom_net_ops); // MUST after esp_at_module_init
    ......
}
```

[at_lib_test.zip](https://github.com/espressif/esp32-at/files/1663457/at_lib_test.zip)

JPast, 2018-01-26:
Hi @xcguang,

I implemented this code and the at_lib_test_core in the newest AT-Command-Core-Version. The compilation and flashing worked without failure. 
Unfortunately this solution approach doesn`t fit well. When I start TCP-Connection with AT+CIPSTART="TCP","192.168.101.110",1000,5 and build up a transparent transmission test_connect_cb is printed out. But when I shut up the WIFI-AP or stop the TCP-Server-Socket, nothing happens. The text test_disconnected_cb is first printed out when I trigger to stop transparent transmission by sending “+++”. I want to detect an missing connection without leaving transparent transmission mode. Is it possible to print out test_disconnected_cd when TCP-Connection is lost while transparent transmission mode is enabled? 

Thank you for your help.

JPast

xcguang, 2018-01-26:
Hi @JPast ,
Sorry for inconvenient. 
You are right, it should print disconnect and connect message whether transparent transmission or normal mode.Please try again using the attachment.
[libat_core.a.zip](https://github.com/espressif/esp32-at/files/1667439/libat_core.a.zip)

JPast, 2018-01-29:
Hi @xcguang,
thank you for your help. This libat_core.a is working fine.

closed 2018-01-29 by JPast