espressif/esp-at #57: CPU 0: at_recv_task, - IDLE (CPU 0), Task watchdog got triggered

mxwang101, 2018-01-07:
When the program run to: 

AT+BLESCAN=1,3

rand happen stop at: 

AT+BLESCAN

then " Task watchdog got triggered"

Here is LOG:

[0;32mI (6558) BLE_AT_GATTC: EVT 0, gattc if 5[0m

Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: at_recv_task
Aborting.
abort() was called at PC 0x400d0a4b on core 0

Backtrace: 0x4008a714:0x3ffc05a0 0x4008a813:0x3ffc05c0 0x400d0a4b:0x3ffc05e0 0x40082502:0x3ffc0600 0x4000bfed:0x00000000

Rebooting...
ets Jun  8 2016 00:22:57

Here is outo exec program log:

Need to Parse:+BLESCAN:e7:2b:5c:ca:40:00,-89,0409694354031900000201060303112305ff1ec192f7,
Send command3
Need to Parse:AT+BLESCAN=1,3
Need to Parse:
OK
Send command3
Need to Parse:AT+BLESCAN=1,3
Need to Parse:
OK
Send command3
Need to Parse:AT+BLESCAN
Send command3
Need to Parse:
ready
Send command1
Need to Parse:AT+GMR
Need to Parse:AT version:1.1.b1.0(52f4bbf - Dec 25 2017 02:18:37)
Need to Parse:SDK version:v3.0-rc1-2-g0ebae99a-dirty
Need to Parse:compile time:Jan  6 2018 12:41:01
Need to Parse:
OK

Can see, after exec 2 times AT+BLESCAN=1,3 ,    Stop at AT+BLESCAN and "Task watchdog got triggered"

FayeY, 2018-01-08:
Hi, sorry that we cannot reproduce your problem. 
Could you provide more details about your test steps?
Thanks.

mxwang101, 2018-01-09:
OK, I edited "at_uart_task.c" to auto exec AT commands. 
When run to "AT+BLESCAN=1,3" , It is rand stop at "AT+BLESCAN". 
Please check attan zip file for more detail. And try to reproduce the problem

[at_uart_task.zip](https://github.com/espressif/esp32-at/files/1614327/at_uart_task.zip)

xcguang, 2018-01-09:
Hi,
Please try 
```
// if (at_app_rx_count == 0) {
    vRingbufferReturnItem(at_app_rx_buffer, data);
// }
```
in `at_port_read_data`.

mxwang101, 2018-01-09:
OK,  So far so good.

 

I try it over night, to see what happen. Thanks first!

 

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 xcguang
发送时间: 2018年1月9日 16:33
收件人: espressif/esp32-at
抄送: mxwang101; Author
主题: Re: [espressif/esp32-at] CPU 0: at_recv_task, - IDLE (CPU 0), Task watchdog got triggered (#57)

 

Hi,
Please try

// if (at_app_rx_count == 0) {
    vRingbufferReturnItem(at_app_rx_buffer, data);
// }

in at_port_read_data.

mxwang101, 2018-01-10:
Thanks，fixed！
On 01/09/2018 16:33, xcguang wrote:

Hi,
Please try

// if (at_app_rx_count == 0) {
    vRingbufferReturnItem(at_app_rx_buffer, data);
// }

in at_port_read_data.