espressif/esp-at #58: AT+PING system crash

orshefi, 2018-01-08:
I used the AT+PING command to google servers, every time after the 7 ping fired it crashes the system.
I thought at first it had something to do with the speed of my instructions but when I sent 1 instruction every 5 minutes it still happens.

`WIFI CONNECTED
I (5843) wifi: pm start, type:0

I (6038) event: sta ip: 192.168.1.33, mask: 255.255.255.0, gw: 192.168.1.1
WIFI GOT IP
AT+PING="8.8.8.8"
+PING:77

OK
AT+PING="8.8.8.8"
+PING:68

OK
AT+PING="8.8.8.8"
+PING:79

OK
AT+PING="8.8.8.8"
+PING:65

OK
AT+PING="8.8.8.8"
+PING:66

OK
AT+PING="8.8.8.8"
+PING:65

OK
AT+PING="8.8.8.8"
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction)
. Exception was unhandled.
Register dump:
PC      : 0x4017ee8a  PS      : 0x00060430  A0      : 0x00000000  A1      : 0x3ffbcc20  
0x4017ee8a: ping_thread at /home/or/Workspace/Nayax/esp/esp32-at/esp-idf/components/lwip/apps/ping/ping.c:311

A2      : 0x00000000  A3      : 0xffffffff  A4      : 0x00000003  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x8017edbd  A9      : 0x3ffbcc00  
A10     : 0xffffffff  A11     : 0x00000003  A12     : 0x00000001  A13     : 0x3ffbbdb8  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x4017ee8a:0x3ffbcc20 0x7ffffffd:0x3ffbcc70
0x4017ee8a: ping_thread at /home/or/Workspace/Nayax/esp/esp32-at/esp-idf/components/lwip/apps/ping/ping.c:311

Rebooting...
`

xcguang, 2018-03-27:
Hi @orshefi ,
Please try it again using the latest vesion.

xcguang, 2018-05-22:
Hi, I close this issue firstly, if the problem still persists, please reopen the issue.

closed 2018-05-22 by xcguang