espressif/esp-at #61: AT+BLEGATTCPRIMSRV=0 make "Guru Meditation Error: Core  0 panic'ed (LoadProhibited)"

mxwang101, 2018-01-11:
run at commands:
----------------------------------------------------------------------
AT+RST

OK

ready
WIFI CONNECTED
WIFI GOT IP
AT+GMR
AT version:1.1.b1.0(52f4bbf - Dec 25 2017 02:18:37)
SDK version:v3.0-rc1-2-g0ebae99a-dirty
compile time:Jan 11 2018 18:57:57

OK
AT+BLEINIT=1

OK
AT+BLESCAN=1,3

OK
+BLESCAN:fe:45:9a:8b:5d:1c,-55,06096c696e79380201060303e7fe0cff0001010125fe459a8b5d1c,
+BLESCAN:f4:6e:19:93:f3:57,-48,06096c696e79380201060303e7fe0cff0001010125f46e1993f357,
AT+BLECONN=0,"fe:45:9a:8b:5d:1c"

OK
+BLECONN:0,"fe:45:9a:8b:5d:1c"
+BLECONNPARAM:0,64,96,96,0,400
AT+BLEGATTCPRIMSRV=0

ready
WIFI CONNECTED
WIFI GOT IP
----------------------------------------------------------------------------

LOG:
[0;32mI (30357) BLE_AT_GATTC: EVT 6, gattc if 3

[0m
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x4012d93f  PS      : 0x00060030  A0      : 0x8013112e  A1      : 0x3ffe7df0  
A2      : 0x00000000  A3      : 0x3ffc7780  A4      : 0x3ffd6940  A5      : 0x3ffd6940  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x8012d97c  A9      : 0x3ffe7dd0  
A10     : 0x3ffd6914  A11     : 0x40082d8c  A12     : 0x3fffff60  A13     : 0x3ffe7e28  
A14     : 0x00000000  A15     : 0x3fffff7c  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x4012d93f:0x3ffe7df0 0x4013112b:0x3ffe7e10 0x4012cb51:0x3ffe7e60 0x40134486:0x3ffe7e80

Rebooting...
ets Jun  8 2016 00:22:57

mxwang101, 2018-01-12:
Any info. about fix this problem?

xiewenxiang, 2018-01-12:
@mxwang101 
What is the the remote device you used to test? I want to reproduce this issus.

mxwang101, 2018-01-12:
We tested with 2 remote devices, one our products, one Samsung Gear S2.

 

Result, 

 

SDK version:v2.1-2-g7138fb02-dirty , both work good.

SDK version:v3.0-rc1-99-g6c69d5e6-dirty, one our products not work, Samsung Gear S2 works.

 

one our products got more info: "+BLECONNPARAM:0,64,96,96,0,400"  It is not see in SDK version:v2.1-2-g7138fb02-dirty, maybe the have some problem???

 

OK, Here is  the test.

 

remote device:

 

No-1: our band, Bluetooth address: fe:45:9a:8b:5d:1c

No-2: Gear S2 (E1A9) LE, Bluetooth address: e4:b0:71:33:63:53

 

Test result:

 

TEST 1) 

 

AT version:1.0.b1.0(1e9060d - Aug 25 2017 10:37:43)

SDK version:v2.1-2-g7138fb02-dirty

compile time: Jan 12 2018 19:51:15

 

 

Both No-1 and No-2, work good.

 

No-1 LOG :

 

AT+RST

 

OK

 

ready

 

[... 387 lines cut ...]

+BLECONN:0,"e4:b0:71:33:63:53"

AT+BLEGATTCPRIMSRV=0

+BLEGATTCPRIMSRV:0,1,0x1800,1

+BLEGATTCPRIMSRV:0,2,0x1801,1

 

OK

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 xiewenxiang
发送时间: 2018年1月12日 14:15
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] AT+BLEGATTCPRIMSRV=0 make "Guru Meditation Error: Core 0 panic'ed (LoadProhibited)" (#61)

 

@mxwang101 <https://github.com/mxwang101> 
What is the the remote device you used to test? I want to reproduce this issus.

xiewenxiang, 2018-01-15:
@mxwang101 
What is the ble chip used in your product?

mxwang101, 2018-01-15:
Nordic IC,  We have 3 kinds of products, all same IC.

 

They work under SDK version:v2.1-2-g7138fb02-dirty

 

Not work at  SDK version:v3.0-rc1-99-g6c69d5e6-dirty

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 xiewenxiang
发送时间: 2018年1月15日 9:40
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] AT+BLEGATTCPRIMSRV=0 make "Guru Meditation Error: Core 0 panic'ed (LoadProhibited)" (#61)

 

@mxwang101 <https://github.com/mxwang101> 
What is the ble chip used in your product?

xiewenxiang, 2018-01-15:
Could you provide the model of the nordic ble soc?
nRF52822 or some other?

mxwang101, 2018-01-15:
nRF52822AC，32K SRAM+256K FLASH

| |
汪欣
邮箱：mwang_56@163.com
|

签名由 网易邮箱大师 定制

On 01/15/2018 15:44, xiewenxiang wrote:

Could you provide the model of the nordic ble soc?
nRF52822 or some other?

xiewenxiang, 2018-01-17:
@mxwang101 
We have bought the nRF51822AC, when we received it, we will reproduce this issues and fix it ASAP.

mxwang101, 2018-01-17:
OK, Wait for good news. Old esp32-at works fine.

| |
汪欣
邮箱：mwang_56@163.com
|

签名由 网易邮箱大师 定制

On 01/17/2018 10:28, xiewenxiang wrote:

@mxwang101
We have bought the nRF51822AC, when we received it, we will reproduce this issues and fix it ASAP.

mxwang101, 2018-01-23:
TICC2540 same problem.

mxwang101, 2018-01-25:
Hi，team，how about this problem fix？ We find more divces have same problem。Low Ver. 2.1 works fine

xiewenxiang, 2018-01-29:
Hi, @mxwang101 

I have received the nRF51822 chip, Could you provide your product bin? I want to reproduce it.

mxwang101, 2018-02-01:
OK,  I will get one for your test. I can not give our product bin since it is not standard design

xiewenxiang, 2018-03-13:
@mxwang101 
We have test it with a `PCA10040(n52822)`, but it seems works fine.

FayeY, 2018-03-22:
Closing as can't reproduce. Please feel free to reopen if the issue persists and you can supply more details.

closed 2018-03-22 by FayeY