espressif/esp-at #62: AT lib and Ethernet integration

santanapablo1975, 2018-01-16:
Hi All,

I have the follwing situation.

I am integrating the Ethernet functionality also with the AT lib, but some how when I start the Ethernet interface I got the message "WIFI GOT IP", this message comes from the AT lib.

Then I decide to switch off the wifi with "AT+CWMODE=0", and start the Ethernet interface again but I got the message again.

Looks that the interface is OK but the message should not appear with WIFI, only "GOT IP" is OK, maybe this should be considered for the next release.

Here some logs about it
at+cwmode=3<\r><\n>
<\r><\n>
OK<\r><\n>
at+cwjap="xxx","yyy"<\r><\n>
WIFI CONNECTED<\r><\n>
WIFI GOT IP<\r><\n>
<\r><\n>
OK<\r><\n>
at+cwmode=0<\r><\n>
WIFI DISCONNECT<\r><\n>
<\r><\n>
OK<\r><\n>
at.eth<\r><\n>
<\r><\n>
ETH OK<\r><\n>
<\r><\n>
 ETH is status:<\r><\n>
<\r><\n>
 IP<\r><\n>
192.168.0.75<\r><\n>
 GW<\r><\n>
192.168.0.1<\r><\n>
 MASK<\r><\n>
255.255.255.0# WIFI GOT IP

santanapablo1975, 2018-01-16:
Hi All

I forgot to send the versions

at+gmr<\r><\n>
AT version:1.1.b1.0(52f4bbf - Dec 25 2017 02:18:37)<\r><\n>
SDK version:v3.0-rc1<\r><\n>
compile time:Jan 16 2018 17:09:23

xcguang, 2018-01-17:
Hi @santanapablo1975 , we have added "+ETH_GOT_IP" to address ethernet got ip, but have not released at github. We will synchronize it ASAP.

santanapablo1975, 2018-01-17:
Thanks for the update, I will close the issue!

closed 2018-01-17 by santanapablo1975