espressif/esp-at #63: +NOTIFY:0,3,2,20,  Stop get string with 0x00

mxwang101, 2018-01-20:
TEST under "OLD SDK version:v2.1-2-g7138fb02-dirty"

The string (20) , NOTIFY will send begin with 0x00:

00 10 14 12 01 14 09 29 1B 0E FF 30 30 31 33 23 00 00 00

But +NOTIFY:0,3,2,20, get nothing. 

send at command:

AT+BLEGATTCWR=0,3,1,,3
>
OK
+NOTIFY:0,3,2,20,
+NOTIFY:0,3,2,20,
+NOTIFY:0,3,2,20,

We Sniffer BLE info, sure other and send whole string. See attn file:
[image]

mxwang101, 2018-01-22:
Please，We do need this in our products

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 xiewenxiang
发送时间: 2018年1月22日 10:45
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] +NOTIFY:0,3,2,20, Stop get string with 0x00 (#63)

 

@mxwang101 <https://github.com/mxwang101> 
we will fix it ASAP.

xiewenxiang, 2018-01-22:
@mxwang101 
If you want to get it as soon as possible, please give me the AT version. I will give you a temporary version today.

mxwang101, 2018-01-22:
AT+GMR

AT version:1.0.b1.0(1e9060d - Aug 25 2017 10:37:43)

SDK version:v2.1-2-g7138fb02-dirty

compile time: Jan 20 2018 01:02:26

OK

 

Please send me temporary version Thanks

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 xiewenxiang
发送时间: 2018年1月22日 14:42
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] +NOTIFY:0,3,2,20, Stop get string with 0x00 (#63)

 

@mxwang101 <https://github.com/mxwang101> 
If you want to get it as soon as possible, please give me the AT version. I will give you a temporary version today.

xiewenxiang, 2018-01-22:
Hi @mxwang101 

The attach files is the `esp-at.bin`, please try it. If you have any question, please let me know.
[esp-at.tar.gz](https://github.com/espressif/esp32-at/files/1651497/esp-at.tar.gz)

mxwang101, 2018-01-22:
OK, I will try it, and let you know result. Thanks

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 xiewenxiang
发送时间: 2018年1月22日 17:54
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] +NOTIFY:0,3,2,20, Stop get string with 0x00 (#63)

 

Hi @mxwang101 <https://github.com/mxwang101> 

The attach files is the esp-at.bin, please try it. If you have any question, please let me know.
esp-at.tar.gz <https://github.com/espressif/esp32-at/files/1651497/esp-at.tar.gz>

mxwang101, 2018-01-22:
I try it. But not fix. anything others?

 

I do this: 

 

1. Put new file esp-at.bin  into my working /build/: 

 

/c/esp32-at/esp32-at/esp32-at/build/esp-at.bin

 

2. export IDF_PATH="C:/esp32-at/esp32-at/esp32-at/esp-idf"

 

then follow: 

................................

3. `rm sdkconfig` to remove the old configuration.

4. Set the latest default configuration by `make defconfig`. 

5. `make menuconfig` -> `Serial flasher config` to configure the serial port for downloading.

6. `make flash` to compile the project and download it into the flash.

................................

 

LOG:

 

AT+BLEGATTCRD=0,3,2,1

+BLEGATTCRD:0,2,

 

OK

AT+BLEGATTCWR=0,3,2,1,2

OK

AT+BLEGATTCWR=0,3,1,,3

OK

+NOTIFY:0,3,2,20,

+NOTIFY:0,3,2,20,

+NOTIFY:0,3,2,20, 

 

 

发件人: mwang_56@163.com [mailto:mwang_56@163.com] 
发送时间: 2018年1月22日 18:07
收件人: 'espressif/esp32-at'; 'espressif/esp32-at'
抄送: 'Mention'
主题: 答复: [espressif/esp32-at] +NOTIFY:0,3,2,20, Stop get string with 0x00 (#63)

 

OK, I will try it, and let you know result. Thanks

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 xiewenxiang
发送时间: 2018年1月22日 17:54
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] +NOTIFY:0,3,2,20, Stop get string with 0x00 (#63)

 

Hi @mxwang101 <https://github.com/mxwang101> 

The attach files is the esp-at.bin, please try it. If you have any question, please let me know.
esp-at.tar.gz <https://github.com/espressif/esp32-at/files/1651497/esp-at.tar.gz>

FayeY, 2018-01-23:
Hi, sorry that we cannot reproduce your problem with our new AT bin.
Could you provide the AT version information which can be got by command "AT+GMR"?
Thanks.

mxwang101, 2018-01-23:
AT+GMR

AT version:1.0.b1.0(1e9060d - Aug 25 2017 10:37:43)

SDK version:v2.1-2-g7138fb02-dirty

compile time:Jan 22 2018 19:32:58

OK

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 FayeY
发送时间: 2018年1月23日 10:28
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] +NOTIFY:0,3,2,20, Stop get string with 0x00 (#63)

 

Hi, sorry that we cannot reproduce your problem with our new AT bin.
Could you provide the AT version information which can be got by command "AT+GMR"?
Thanks.

xiewenxiang, 2018-01-23:
@mxwang101 
Hi mxwang101, you don't use the `esp_at.bin` which was I send to you , because the compile time is not correct.

mxwang101, 2018-01-23:
I do not know how to setup use the new esp_at.bin. 

 

I search where is old esp_at.bin,  delete it, and put new one in.

 

the  follow readme file. make clean,  remake all.

 

something not right? 

 

Could please let me know how to make new esp_at.bin work?  Thanks

 

 

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 xiewenxiang
发送时间: 2018年1月23日 14:25
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] +NOTIFY:0,3,2,20, Stop get string with 0x00 (#63)

 

@mxwang101 <https://github.com/mxwang101> 
Hi mxwang101, you don't use the esp_at.bin which was I send to you , because the compile time is not correct.

xiewenxiang, 2018-01-23:
@mxwang101 
First, make clean.
Then, make 
Now, you need to use the new `esp-at.bin` to replace the old `esp-at.bin  (./esp-at/build/esp-at.bin)`, 
In the end, make flash
just make flash

xiewenxiang, 2018-01-23:
@mxwang101 
Does it work now?

mxwang101, 2018-01-23:
OK, I will try it, Thanks

mxwang101, 2018-01-23:
Can we flash only new esp-at.bin to the chip?

mxwang101, 2018-01-23:
OK, Now it is OK,  THANKS!

 

AT+GMR

AT version:1.0.b1.0(1e9060d - Jan 22 2018 17:39:07)

SDK version:v2.1

compile time: Jan 22 2018 17:41:13

OK

 

......................

+NOTIFY:0,3,2,20,"[1]065#

+NOTIFY:0,3,2,20," 

  _____  

098#

+NOTIFY:0,3,2,20,"00013#

................................

 

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 xiewenxiang
发送时间: 2018年1月23日 16:24
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] +NOTIFY:0,3,2,20, Stop get string with 0x00 (#63)

 

@mxwang101 <https://github.com/mxwang101> 
Does it work now?

xiewenxiang, 2018-01-23:
@mxwang101  
:)

closed 2018-01-23 by FayeY