espressif/esp-at #64: can't enter any AT command by Putty

andywang2039, 2018-01-23:
hi all,
i hope can get your help and i appreciate you in advance.
i bought 1 pcs ESP-WROVER-KIT from Espressif and want to develop myself hardware application and design.  i used "flash_download tool 3.6.3 " download whole bin file as below.
[image], after download and open by Putty it shows to me as below, during this, i can't enter any AT commands, i think there is someting wrong else?

on the other hand,i also tried use IDF with toolchain to build AT project, but also is failed.

anybody can tell me an simple   procedure for AT build? my PC is Windows 7, 64bit.

thanks 
B.R
Andy

, 
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:4576
load:0x40078000,len:11640
load:0x40080000,len:896
entry 0x400802b8
I (29) boot: Detected ESP32
I (48) boot: ESP-IDF v2.1-65-g8bca703 2nd stage bootloader
I (48) boot: compile time 07:59:19
I (55) boot: Enabling RNG early entropy source...
I (66) boot: SPI Speed      : 40MHz
I (79) boot: SPI Mode       : DIO
I (91) boot: SPI Flash Size : 8MB
I (104) boot: Partition Table:
I (115) boot: ## Label            Usage          Type ST Offset   Length
I (138) boot:  0 phy_init         RF data          01 01 0000f000 00001000
I (161) boot:  1 otadata          OTA data         01 00 00010000 00002000
I (184) boot:  2 nvs              WiFi data        01 02 00012000 0000e000
I (207) boot:  3 at_customize     unknown          40 00 00020000 000e0000
I (230) boot:  4 ota_0            OTA app          00 10 00100000 00180000
I (254) boot:  5 ota_1            OTA app          00 11 00280000 00180000
I (277) boot: End of partition table
I (290) boot: Disabling RNG early entropy source...
I (307) boot: Loading app partition at offset 00100000
I (1936) boot: segment 0: paddr=0x00100018 vaddr=0x00000000 size=0x0ffe8 ( 65512)
I (1937) boot: segment 1: paddr=0x00110008 vaddr=0x3f400010 size=0x24a58 (150104) map
I (1954) boot: segment 2: paddr=0x00134a68 vaddr=0x3ffc0000 size=0x02e24 ( 11812) load
I (1985) boot: segment 3: paddr=0x00137894 vaddr=0x40080000 size=0x00400 (  1024) load
I (2007) boot: segment 4: paddr=0x00137c9c vaddr=0x40080400 size=0x11878 ( 71800) load
I (2067) boot: segment 5: paddr=0x0014951c vaddr=0x400c0000 size=0x00064 (   100) load
I (2069) boot: segment 6: paddr=0x00149588 vaddr=0x00000000 size=0x06a80 ( 27264)
I (2092) boot: segment 7: paddr=0x00150010 vaddr=0x400d0018 size=0xc9448 (824392) map
I (2118) cpu_start: Pro cpu up.
I (2129) cpu_start: Single core mode
I (2144) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (2166) heap_alloc_caps: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (2187) heap_alloc_caps: At 3FFB3000 len 00005000 (20 KiB): DRAM
I (2208) heap_alloc_caps: At 3FFBBB28 len 00002000 (8 KiB): DRAM
I (2228) heap_alloc_caps: At 3FFD2EF0 len 0000D110 (52 KiB): DRAM
I (2249) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (2271) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (2292) heap_alloc_caps: At 40091C78 len 0000E388 (56 KiB): IRAM
I (2313) cpu_start: Pro cpu start user code
I (2367) cpu_start: Starting scheduler on PRO CPU.
I (2573) uart: queue free spaces: 10
Bin version(Wrover32):1.0.0
I (2576) wifi: wifi firmware version: 621ab6b
I (2577) wifi: config NVS flash: enabled
I (2577) wifi: config nano formating: disabled
I (2581) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (2591) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (2611) wifi: Init dynamic tx buffer num: 32
I (2611) wifi: Init dynamic rx buffer num: 32
I (2612) wifi: wifi driver task: 3ffbd388, prio:23, stack:4096
I (2615) wifi: Init static rx buffer num: 10
I (2618) wifi: Init dynamic rx buffer num: 32
I (2623) wifi: Init rx ampdu len mblock:7
I (2626) wifi: Init lldesc rx ampdu entry mblock:4
I (2631) wifi: wifi power manager task: 0x3ffdbaf8 prio: 21 stack: 2560
I (2638) wifi: wifi timer task: 3ffdcb78, prio:22, stack:3584
I (2671) phy: phy_version: 362.1, 75758b5, Nov  1 2017, 16:02:06, 0, 0
I (2672) wifi: mode : softAP (30:ae:a4:2a:55:59)
I (2678) wifi: mode : sta (30:ae:a4:2a:55:58) + softAP (30:ae:a4:2a:55:59)
I (2681) wifi: mode : softAP (30:ae:a4:2a:55:59)
I (3159) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1
I (3160) wifi: station: 7c:01:91:b8:9b:a6 join, AID=1, g, 20

xcguang, 2018-01-23:
Hi, 
You should enter AT command via UART1,please refer to [esp32_at_instruction_set_and_examples_en.pdf](http://espressif.com/sites/default/files/documentation/esp32_at_instruction_set_and_examples_en.pdf) . By default, AT project is based on ESP WROOM 32, so  you must configure gpio pins for WROVER32  AT port.

andywang2039, 2018-01-23:
Hi 
many thanks for your quickly reply, after configure gpio for WROVER32, the AT command still can't be enter by Putty or msys32. i used 1 week to study this develop kit but unfortunate the result still is unsuccessful. i would like stick out my processed list, hope you can help review which loop is wrong.

reference procedure is https://github.com/espressif/esp32-at
===================================== menuconfig detail
[image]
[image]
==========================================make flash

AndyWang@AndyWang-PC MINGW32 ~
$ cd c:/esp/esp32-at

AndyWang@AndyWang-PC MINGW32 /c/esp/esp32-at
$ make menuconfig
WARNING: Toolchain version is not supported: 1.22.0-80-g6c4433a
Expected to see version: 1.22.0-75-gbaf03c2
Please check ESP-IDF setup instructions and update the toolchain, or proceed at                                                                                                                            your own risk.
MENUCONFIG

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

GENCONFIG
make print_flash_cmd | tail -n 1 > build/download.config

AndyWang@AndyWang-PC MINGW32 /c/esp/esp32-at
$ make print_flash_cmd | tail -n 1 > build/download.config

AndyWang@AndyWang-PC MINGW32 /c/esp/esp32-at
$ make print_flash_cmd | tail -n 1 > build/download.config

AndyWang@AndyWang-PC MINGW32 /c/esp/esp32-at
$ make factory_bin
WARNING: Toolchain version is not supported: 1.22.0-80-g6c4433a
Expected to see version: 1.22.0-75-gbaf03c2
Please check ESP-IDF setup instructions and update the toolchain, or proceed at                                                                                                                            your own risk.
0x1000,bootloader/bootloader.bin
0x20000,at_customize.bin
0x21000,customized_partitions/ble_data.bin
0x24000,customized_partitions/server_cert.bin
0x26000,customized_partitions/server_key.bin
0x28000,customized_partitions/server_ca.bin
0x2a000,customized_partitions/client_cert.bin
0x2c000,customized_partitions/client_key.bin
0x2e000,customized_partitions/client_ca.bin
0xf000,phy_init_data.bin
0x100000,esp-at.bin
0x8000,partitions_at.bin

AndyWang@AndyWang-PC MINGW32 /c/esp/esp32-at
$ make monitor
WARNING: Toolchain version is not supported: 1.22.0-80-g6c4433a
Expected to see version: 1.22.0-75-gbaf03c2
Please check ESP-IDF setup instructions and update the toolchain, or proceed at                                                                                                                            your own risk.
MONITOR
--- idf_monitor on COM75 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5588
load:0x40078000,len:0
load:0x40078000,len:13716
entry 0x40078fb4
I (28) boot: ESP-IDF v3.1-dev-204-g914ab075 2nd stage bootloader
I (28) boot: compile time 17:35:21
[... 938 lines cut ...]
I (593) cpu_start: Starting scheduler on PRO CPU.
I (670) uart: queue free spaces: 10
I (674) wifi: wifi firmware version: 403db1d
I (674) wifi: config NVS flash: enabled
I (674) wifi: config nano formating: disabled
I (676) system_api: Base MAC address is not set, read default base MAC address f
rom BLK0 of EFUSE
I (685) system_api: Base MAC address is not set, read default base MAC address f
rom BLK0 of EFUSE
I (707) wifi: Init dynamic tx buffer num: 32
I (708) wifi: Init data frame dynamic rx buffer num: 32
I (708) wifi: Init management frame dynamic rx buffer num: 32
I (712) wifi: wifi driver task: 3ffdfb84, prio:23, stack:4096
I (717) wifi: Init static rx buffer num: 10
I (721) wifi: Init dynamic rx buffer num: 32
I (725) wifi: wifi power manager task: 0x3ffe3ce4 prio: 21 stack: 2560
I (757) phy: phy_version: 366.0, ba9923d, Dec 29 2017, 14:25:06, 0, 0
I (759) wifi: mode : softAP (30:ae:a4:4b:11:85)
I (765) wifi: mode : sta (30:ae:a4:4b:11:84) + softAP (30:ae:a4:4b:11:85)
I (769) wifi: mode : softAP (30:ae:a4:4b:11:85)
=========================================================
at this stage, i can't enter any AT command by Putty or msys32.

B.R
Andy

xcguang, 2018-01-25:
Hi 
You can try it like this  
[image]

andywang2039, 2018-01-25:
hi, 
thanks you reviewed my quesiton.

today i study the AT start guide from http://blog.csdn.net/espressif/article/details/78772678, i saw he mentioned should use external USB<=> uart module for AT command, i have confuse: whether i really need a external USB<=> uart module? 

i had stick out whole logs and i tested process in previous, am i did is correct?

the terminal display contents is means i previous set up and make flash process are right? and now i trouble just is set up or choose appropriate UART port for AT command enter?

Best Regards,

Andy

[image]

xcguang, 2018-01-25:
Yes, your process is OK.  
Please ensure it is correct the cable connection between usb2uart adapter and esp32 uart pin(pin19 and pin22).

xcguang, 2018-01-28:
Hello，
Does it work well？

andywang2039, 2018-01-31:
hi xcguang,

谢谢您的回复，但是不幸的是至今为止我还无法正常运行ESP-32 AT 在我的ESP-WROVER-KIT板上(其它getting start example都可以运行），它已经困扰我几个星期了，我反反复复的建立编译环境按照官方指引，并且都已用最新的IDF和toolchan等工具，最最糟的是现在连flash都会出错。现在我拷贝出最近一次失败的Log,这里面有warning和error,请帮我我看一下究竟哪里出了问题。我现在急需能够用到AT，然后开发属于我们自己的ESP32模块，但是前提是AT可以在ESP-WROVER-KIT先跑起来。

另外我想知道乐鑫深圳是否技术支持，我想亲自带着电脑到乐鑫工程师帮我调试一下。

Best Regards,

Andy

AndyWang@AndyWang-PC MINGW32 /e/esp/esp32-at
$ make flash
GENCONFIG
CC build/bootloader/bootloader_support/src/bootloader_random.o
CC build/bootloader/bootloader_support/src/flash_encrypt.o
CC build/bootloader/bootloader_support/src/bootloader_sha.o
CC build/bootloader/bootloader_support/src/esp_image_format.o
CC build/bootloader/bootloader_support/src/flash_partitions.o
CC build/bootloader/bootloader_support/src/secure_boot_signatures.o
CC build/bootloader/bootloader_support/src/bootloader_clock.o
CC build/bootloader/bootloader_support/src/secure_boot.o
CC build/bootloader/bootloader_support/src/efuse.o
CC build/bootloader/bootloader_support/src/bootloader_flash.o
AR build/bootloader/bootloader_support/libbootloader_support.a
CC build/bootloader/log/log.o
AR build/bootloader/log/liblog.a
CC build/bootloader/spi_flash/spi_flash_rom_patch.o
AR build/bootloader/spi_flash/libspi_flash.a
CC build/bootloader/soc/esp32/rtc_clk.o
CC build/bootloader/soc/esp32/rtc_sleep.o
CC build/bootloader/soc/esp32/rtc_init.o
CC build/bootloader/soc/esp32/cpu_util.o
AR build/bootloader/soc/libsoc.a
CC build/bootloader/main/flash_qio_mode.o
CC build/bootloader/main/bootloader_start.o
AR build/bootloader/main/libmain.a
LD build/bootloader/bootloader.elf
esptool.py v2.2.1
building AT customized partition table at_customize.csv
building AT customized partition table succeed
generating AT customized partition bins to /e/esp/esp32-at/build/customized_part                                                                                                                           itions
generating ble_data.bin: python /e/esp/esp32-at/tools/BLEService.py -t /e/esp/es                                                                                                                           p32-at/build/customized_partitions/ble_data.bin /e/esp/esp32-at/components/custo                                                                                                                           mized_partitions/raw_data/ble_data/example.csv
Traceback (most recent call last):
  File "E:/esp/esp32-at/tools/BLEService.py", line 29, in <module>
    import yaml
ImportError: No module named yaml
make: *** [/e/esp/esp32-at/components/customized_partitions/Makefile.projbuild:4                                                                                                                           3: /e/esp/esp32-at/build/at_customize.bin] Error 1

AndyWang@AndyWang-PC MINGW32 /e/esp/esp32-at
$ curl https://bootstrap.pypa.io/get-pip.py -o get-pip.py
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1558k  100 1558k    0     0  37102      0  0:00:43  0:00:43 --:--:-- 45662

AndyWang@AndyWang-PC MINGW32 /e/esp/esp32-at
$ python get-pip.py
Requirement already up-to-date: pip in e:/esp/msys32/mingw32/lib/python2.7/site-                                                                                                                           packages

AndyWang@AndyWang-PC MINGW32 /e/esp/esp32-at
$ curl https://bootstrap.pypa.io/get-pip.py -o get-pip.py
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1558k  100 1558k    0     0  45583      0  0:00:35  0:00:35 --:--:-- 18562

AndyWang@AndyWang-PC MINGW32 /e/esp/esp32-at
$ python get-pip.py
Requirement already up-to-date: pip in e:/esp/msys32/mingw32/lib/python2.7/site-                                                                                                                           packages

AndyWang@AndyWang-PC MINGW32 /e/esp/esp32-at
$ python -m pip install -U pip
[... 694 lines cut ...]
CC build/wpa_supplicant/src/crypto/aes-unwrap.o
CC build/wpa_supplicant/src/crypto/aes-internal-dec.o
CC build/wpa_supplicant/src/crypto/sha1.o
CC build/wpa_supplicant/src/crypto/aes-internal.o
CC build/wpa_supplicant/src/crypto/md5-internal.o
CC build/wpa_supplicant/src/crypto/sha256-internal.o
CC build/wpa_supplicant/src/crypto/aes-internal-enc.o
CC build/wpa_supplicant/port/os_xtensa.o
CC build/wpa_supplicant/src/fast_crypto/fast_crypto_internal-cipher.o
CC build/wpa_supplicant/src/fast_crypto/fast_crypto_internal-modexp.o
CC build/wpa_supplicant/src/fast_crypto/fast_aes-unwrap.o
CC build/wpa_supplicant/src/fast_crypto/fast_sha256.o
CC build/wpa_supplicant/src/fast_crypto/fast_crypto_internal.o
CC build/wpa_supplicant/src/fast_crypto/fast_aes-cbc.o
CC build/wpa_supplicant/src/fast_crypto/fast_sha256-internal.o
CC build/wpa_supplicant/src/fast_crypto/fast_aes-wrap.o
AR build/wpa_supplicant/libwpa_supplicant.a
CC build/xtensa-debug-module/trax.o
AR build/xtensa-debug-module/libxtensa-debug-module.a
LD build/esp-at.elf
E:/esp/esp32-at/components/at/lib\libat_core.a(at_wifiCmd.o):(.literal.at_setupC                                                                                                                           mdMdns+0x30): undefined reference to `mdns_set_hostname'
E:/esp/esp32-at/components/at/lib\libat_core.a(at_wifiCmd.o): In function `at_se                                                                                                                           tupCmdMdns':
(.text.at_setupCmdMdns+0x14a): undefined reference to `mdns_set_hostname'
collect2.exe: error: ld returned 1 exit status
make: *** [E:/esp/esp-idf/make/project.mk:389: /e/esp/esp32-at/build/esp-at.elf]                                                                                                                            Error 1

xcguang, 2018-01-31:
Hi @andywang2039 ，
我们在深圳没有开发工程师，你当前这个问题是因为没有使用我们 esp32-at 下面的 esp-idf 所导致的，请参考 https://github.com/espressif/esp32-at/issues/66

andywang2039, 2018-01-31:
HI xcguang,
谢谢你方法，现在我已经可以输入AT命令了。但是又有新问题就是我只能输入基础命令的前三条，从AT+GSLP开始输入后就显示error。我分别试过make flash和用download toolx下载AT固件，结果是一样的。
请问这是哪里有问题？

Best Regards,
Andy

[image]

xcguang, 2018-02-01:
AT+GSLP=<time>
time unit is ms.

andywang2039, 2018-02-01:
hi xcguang,

为什么我每次重新make esp32-at 时都会报以下错误，该如何解决？我用的是Python2.7。
我发现当失败后我再次输入make，编译将继续进行（但似乎跳过了之前失败的步骤），那是不是意味着后续的某些功能会出错?

andy

AndyWang@AndyWang-PC MINGW32 /e/esp/esp32-at
$ make
WARNING: Toolchain version is not supported: 1.22.0-75-gbaf03c2
Expected to see version: 1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at        your own risk.
WARNING: Toolchain version is not supported: 1.22.0-75-gbaf03c2
Expected to see version: 1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at        your own risk.
CC build/bootloader/bootloader_support/src/bootloader_random.o
CC build/bootloader/bootloader_support/src/flash_encrypt.o
CC build/bootloader/bootloader_support/src/bootloader_sha.o
CC build/bootloader/bootloader_support/src/esp_image_format.o
CC build/bootloader/bootloader_support/src/flash_partitions.o
CC build/bootloader/bootloader_support/src/secure_boot_signatures.o
CC build/bootloader/bootloader_support/src/bootloader_clock.o
CC build/bootloader/bootloader_support/src/secure_boot.o
CC build/bootloader/bootloader_support/src/efuse.o
CC build/bootloader/bootloader_support/src/bootloader_flash.o
AR build/bootloader/bootloader_support/libbootloader_support.a
CC build/bootloader/log/log.o
AR build/bootloader/log/liblog.a
CC build/bootloader/spi_flash/spi_flash_rom_patch.o
AR build/bootloader/spi_flash/libspi_flash.a
CC build/bootloader/micro-ecc/micro-ecc/uECC.o
AR build/bootloader/micro-ecc/libmicro-ecc.a
CC build/bootloader/soc/esp32/rtc_clk.o
CC build/bootloader/soc/esp32/rtc_time.o
CC build/bootloader/soc/esp32/rtc_sleep.o
CC build/bootloader/soc/esp32/rtc_init.o
CC build/bootloader/soc/esp32/soc_memory_layout.o
CC build/bootloader/soc/esp32/cpu_util.o
CC build/bootloader/soc/esp32/rtc_pm.o
AR build/bootloader/soc/libsoc.a
CC build/bootloader/main/flash_qio_mode.o
CC build/bootloader/main/bootloader_start.o
AR build/bootloader/main/libmain.a
LD build/bootloader/bootloader.elf
esptool.py v2.1
building AT customized partition table at_customize.csv
building AT customized partition table succeed
generating AT customized partition bins to /e/esp/esp32-at/build/customized_part       itions
generating ble_data.bin: python /e/esp/esp32-at/tools/BLEService.py -t /e/esp/es       p32-at/build/customized_partitions/ble_data.bin /e/esp/esp32-at/components/custo       mized_partitions/raw_data/ble_data/example.csv
Traceback (most recent call last):
  File "E:/esp/esp32-at/tools/BLEService.py", line 29, in <module>
    import yaml
ImportError: No module named yaml
make: *** [/e/esp/esp32-at/components/customized_partitions/Makefile.projbuild:4       3: /e/esp/esp32-at/build/at_customize.bin] Error 1

AndyWang@AndyWang-PC MINGW32 /e/esp/esp32-at
$ python setup.py install
E:\esp\msys32\mingw32\bin\python.exe: can't open file 'setup.py': [Errno 2] No such file or directory

AndyWang@AndyWang-PC MINGW32 /e/esp/esp32-at
$ curl https://bootstrap.pypa.io/get-pip.py -o get-pip.py
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 39 1558k   39  608k    0     0  23959      0  0:01:06  0:00:26  0:00:40 18472

ustccw, 2018-02-01:
@andywang2039 你好, yaml 模块会对低功耗蓝牙(BLE)部分有影响, 对 wifi 和其他模块无影响。建议还是装上 yaml 模块比较好。安装办法你已经提过，见[ESP32 AT 入门指南 Part 5.1](http://blog.csdn.net/espressif/article/details/78772678)。

andywang2039, 2018-02-02:
HI, thanks your help, i had fixed this issue and can run AT comments,cool!

i have another question about AT-->BT part:

1. when i do AT+BLEADDR commands, the terminal feedback error as below( i had closed wi-fi).
[image]
2. which commands should to be implements then i can find my testing DUT's BT equipment name by my iphone? because when i open my iphone BT equipments seaching, the ESP-WROVER-KIT BT name does not list out, it seems i can't search it by my iphone. 

Andy

xiewenxiang, 2018-02-02:
@andywang2039 

The first question:

This command was used to set the static random address, a static address is a 48-bit randomly generated address and shall meet the following
requirements:
• The two most significant bits of the static address shall be equal to ‘1’
• All bits of the random part of the static address shall not be equal to ‘1’
• All bits of the random part of the static address shall not be equal to ‘0’
The example about this command in our document is error, we will modify it as soon as possible. 
You can set the address like this:
`AT+BLEADDR=1,"f8:7f:24:87:1c:f7"`

The second question:

You need to set the adv data. The adv data should follow the BLE specification.
for example:
```
AT+BLEINIT=2
AT+BLEADVDATA="020106060950686f6e65"
AT+BLEADVSTART
```

xcguang, 2018-03-12:
Hi @andywang2039,
Are the issues resolved? Please update them.

closed 2018-08-07 by xcguang