espressif/esp-at #66: undefined reference of mdns_set_hostname

mendesmauricio, 2018-01-24:
Hi guys, i just started to run th examples but the esp32-at i couldn't compile.

i am getting this error

C:/esp32/examples/esp32-at/components/at/lib\libat_core.a(at_wifiCmd.o):(.literal.at_setupCmdMdns+0x30): undefined reference to `mdns_set_hostname'
C:/esp32/examples/esp32-at/components/at/lib\libat_core.a(at_wifiCmd.o): In function `at_setupCmdMdns':
(.text.at_setupCmdMdns+0x14a): undefined reference to `mdns_set_hostname'
collect2.exe: error: ld returned 1 exit status
make: *** [C:/esp32/esp-idf/make/project.mk:389: /c/esp32/examples/esp32-at/build/esp-at.elf] Error 1

i am using https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20180110.zip version

Does anyone have a insight?

Many Thanks

xcguang, 2018-01-25:
Hi,
You can ```git clone --recursive https://github.com/espressif/esp32-at.git```, then ```cd esp32-at``` and ```make```. 
Not set ```export IDF_PATH=xxxx``` and it will be set automatically when compiling AT project.

mendesmauricio, 2018-01-25:
Hey thanks for a fast response. 

I still with problems. i added my console log in attachment. any hints?
[console.txt](https://github.com/espressif/esp32-at/files/1663157/console.txt)

xcguang, 2018-01-25:
Hi,
Does it work well?

mendesmauricio, 2018-01-25:
Hi still not working.

bellow my console messages.

LD build/esp-at.elf
C:/esp32/workspace/esp32-at/build/esp32\libesp32.a(cpu_start.o):(.literal.main_task+0x18): undefined reference to `heap_caps_enable_nonos_stack_heaps'
C:/esp32/workspace/esp32-at/build/esp32\libesp32.a(cpu_start.o):(.iram1.literal+0xc8): undefined reference to `heap_caps_init'
C:/esp32/workspace/esp32-at/build/esp32\libesp32.a(cpu_start.o): In function `main_task':
C:/esp32/esp-idf/components/esp32/cpu_start.c:431: undefined reference to `heap_caps_enable_nonos_stack_heaps'
C:/esp32/workspace/esp32-at/build/esp32\libesp32.a(cpu_start.o): In function `call_start_cpu0':
C:/esp32/esp-idf/components/esp32/cpu_start.c:179: undefined reference to `heap_caps_init'
C:/esp32/workspace/esp32-at/build/esp32\libesp32.a(system_api.o):(.literal.esp_get_free_heap_size+0x4): undefined reference to `heap_caps_get_free_size'
C:/esp32/workspace/esp32-at/build/esp32\libesp32.a(system_api.o): In function `esp_get_free_heap_size':
C:/esp32/esp-idf/components/esp32/system_api.c:378: undefined reference to `heap_caps_get_free_size'
C:/esp32/workspace/esp32-at/build/esp32\libesp32.a(intr_alloc.o):(.literal.get_desc_for_int+0xc): undefined reference to `heap_caps_malloc'
C:/esp32/workspace/esp32-at/build/esp32\libesp32.a(intr_alloc.o): In function `get_desc_for_int':
C:/esp32/esp-idf/components/esp32/intr_alloc.c:773: undefined reference to `heap_caps_malloc'
C:/esp32/workspace/esp32-at/build/esp32\libesp32.a(intr_alloc.o): In function `esp_intr_alloc_intrstatus':
C:/esp32/esp-idf/components/esp32/intr_alloc.c:577: undefined reference to `heap_caps_malloc'
C:/esp32/workspace/esp32-at/build/newlib\libnewlib.a(syscalls.o):(.iram1.literal+0x0): undefined reference to `heap_caps_malloc_default'
C:/esp32/workspace/esp32-at/build/newlib\libnewlib.a(syscalls.o):(.iram1.literal+0x4): undefined reference to `heap_caps_free'
C:/esp32/workspace/esp32-at/build/newlib\libnewlib.a(syscalls.o):(.iram1.literal+0x8): undefined reference to `heap_caps_realloc_default'
C:/esp32/workspace/esp32-at/build/newlib\libnewlib.a(syscalls.o): In function `_malloc_r':
C:/esp32/esp-idf/components/newlib/syscalls.c:37: undefined reference to `heap_caps_malloc_default'
C:/esp32/workspace/esp32-at/build/newlib\libnewlib.a(syscalls.o): In function `_free_r':
C:/esp32/esp-idf/components/newlib/syscalls.c:42: undefined reference to `heap_caps_free'
C:/esp32/workspace/esp32-at/build/newlib\libnewlib.a(syscalls.o): In function `_realloc_r':
C:/esp32/esp-idf/components/newlib/syscalls.c:47: undefined reference to `heap_caps_realloc_default'
C:/esp32/workspace/esp32-at/build/newlib\libnewlib.a(syscalls.o): In function `_calloc_r':
C:/esp32/esp-idf/components/newlib/syscalls.c:52: undefined reference to `heap_caps_malloc_default'
C:/esp32/workspace/esp32-at/build/spi_flash\libspi_flash.a(flash_mmap.o):(.iram1.literal+0x3c): undefined reference to `heap_caps_malloc'
C:/esp32/workspace/esp32-at/build/spi_flash\libspi_flash.a(flash_mmap.o): In function `spi_flash_mmap_pages':
C:/esp32/esp-idf/components/spi_flash/flash_mmap.c:293: undefined reference to `heap_caps_malloc'
C:/esp32/workspace/esp32-at/build/bt\libbt.a(bt.o):(.literal.esp_bt_controller_mem_release+0x10): undefined reference to `heap_caps_add_region'
C:/esp32/workspace/esp32-at/build/bt\libbt.a(bt.o): In function `esp_bt_controller_mem_release':
C:/esp32/esp-idf/components/bt/bt.c:436: undefined reference to `heap_caps_add_region'
C:/esp32/esp-idf/components/bt/bt.c:442: undefined reference to `heap_caps_add_region'
C:/esp32/workspace/esp32-at/build/freertos\libfreertos.a(tasks.o): In function `xTaskCreatePinnedToCore':
C:/esp32/esp-idf/components/freertos/tasks.c:4571: undefined reference to `heap_caps_malloc'
C:/esp32/esp-idf/components/freertos/tasks.c:4571: undefined reference to `heap_caps_malloc'
C:/esp32/esp-idf/components/esp32/lib\libpp.a(wdev.o):(.text.wDev_Rxbuf_Init+0x32): undefined reference to `heap_caps_malloc'
C:/esp32/esp-idf/components/esp32/lib\libnet80211.a(ieee80211_nvs.o):(.text.wifi_nvs_load+0x20): undefined reference to `heap_caps_malloc'
C:/esp32/esp-idf/components/esp32/lib\libpp.a(esf_buf.o):(.text.esf_buf_alloc+0xc): undefined reference to `heap_caps_calloc'
C:/esp32/esp-idf/components/esp32/lib\libpp.a(esf_buf.o): In function `esf_buf_alloc':
(.text.esf_buf_alloc+0x15e): undefined reference to `heap_caps_calloc'
C:/esp32/esp-idf/components/esp32/lib\libpp.a(esf_buf.o): In function `esf_buf_alloc':
(.text.esf_buf_alloc+0x172): undefined reference to `heap_caps_calloc'
C:/esp32/esp-idf/components/esp32/lib\libpp.a(esf_buf.o): In function `esf_buf_alloc':
(.text.esf_buf_alloc+0x18e): undefined reference to `heap_caps_calloc'
C:/esp32/esp-idf/components/esp32/lib\libpp.a(esf_buf.o): In function `esf_buf_alloc':
(.text.esf_buf_alloc+0x1a4): undefined reference to `heap_caps_malloc'
C:/esp32/esp-idf/components/esp32/lib\libpp.a(esf_buf.o): In function `esf_buf_setup':
(.text.esf_buf_setup+0x1c4): undefined reference to `heap_caps_calloc'
C:/esp32/esp-idf/components/esp32/lib\libpp.a(esf_buf.o): In function `esf_buf_setup':
(.text.esf_buf_setup+0x1d7): undefined reference to `heap_caps_calloc'
C:/esp32/esp-idf/components/esp32/lib\libpp.a(esf_buf.o): In function `esf_buf_setup':
(.text.esf_buf_setup+0x1f2): undefined reference to `heap_caps_calloc'
C:/esp32/esp-idf/components/esp32/lib\libpp.a(esf_buf.o): In function `esf_buf_setup':
(.text.esf_buf_setup+0x203): undefined reference to `heap_caps_malloc'
[... cut ...]
C:/esp32/workspace/esp32-at/components/at/lib\libat_core.a(at_wifiCmd.o): In function `at_setupCmdMdns':
(.text.at_setupCmdMdns+0x14a): undefined reference to `mdns_set_hostname'
C:/esp32/esp-idf/components/esp32/lib\libcore.a(misc_nvs.o):(.text.misc_nvs_load+0x14): undefined reference to `heap_caps_malloc'
C:/esp32/esp-idf/components/esp32/lib\libcore.a(misc_nvs.o):(.text.misc_nvs_load+0x2a): undefined reference to `heap_caps_malloc'
collect2.exe: error: ld returned 1 exit status
make: *** [C:/esp32/esp-idf/make/project.mk:389: /c/esp32/workspace/esp32-at/build/esp-at.elf] Error 1

xcguang, 2018-01-27:
Hi @mendesmauricio ,
Maybe the commit id of your esp-idf is not `e7dc749e`, could you confirm it by ```git log``` in esp-idf directory which you use?
We have found ```mdns_set_hostname``` is removed in the latest esp-idf, and we are porting it.

mendesmauricio, 2018-01-27:
Hello @xcguang  thanks for helping. 

The version on esp-idf folder inside of esp-32at folder is 
On esp32-at folder we have a esp-idf folder in that one the version is 

$ git log
commit e7dc749e2f953a976325964b5ac49b53c8bafc42 (HEAD)
Merge: 4a55009f df93f672
Author: Jiang Jiang Jian <jack@espressif.com>
Date:   Tue Jan 16 19:43:52 2018 +0800

i do have other esp-idf folder which version is 
$ git log
commit ffd4187883d69c5c39f2a1961fda06b51ed998fd (HEAD -> master, origin/master,                                                                                             origin/HEAD)
Merge: 84741e85 da1bf609
Author: Angus Gratton <angus@espressif.com>
Date:   Tue Jan 23 08:36:52 2018 +0800
    Merge branch 'feature/esptool_v2_2' into 'master'
    esptool: Update to v2.2
    See merge request !1755

what i should do? revert to the e7dc749e commit?

Many thanks guys,

xcguang, 2018-01-28:
Hi @mendesmauricio ,
You can revert your esp-idf to the e7dc749e commit
or
Open a new shell in esp32-at directory, and run ```make``` directly.

xcguang, 2018-03-27:
Hi @mendesmauricio ,
Please try it again using the latest version.

closed 2018-04-11 by xcguang