espressif/esp-at #67: Filesystem Operations, AT+FS, Not Work

mxwang101, 2018-01-26:


mxwang101, 2018-01-26:
LOG: 

AT+SYSFLASH?
+SYSFLASH:"ble_data",64,1,0x21000,0x3000
+SYSFLASH:"server_cert",64,2,0x24000,0x2000
+SYSFLASH:"server_key",64,3,0x26000,0x2000
+SYSFLASH:"server_ca",64,4,0x28000,0x2000
+SYSFLASH:"client_cert",64,5,0x2a000,0x2000
+SYSFLASH:"client_key",64,6,0x2c000,0x2000
+SYSFLASH:"fatfs",1,129,0x70000,0x90000

OK
AT+SYSFLASH=2,"fatfs",0,20
ERR CODE:0x010a0000

ERROR
AT+FS=0,1,"file1",100,10
ERR CODE:0x01090000

ERROR
AT+FS=0,4,"."
ERR CODE:0x01090000

ERROR

mxwang101, 2018-01-26:
AT version:1.0.b1.0(1e9060d - Jan 22 2018 17:39:07)
SDK version:v2.1
compile time:Jan 22 2018 17:41:13

xcguang, 2018-01-27:
Hi @mxwang101 ，
By default, `fs` is not supported, and you have to enable it in ```make menuconfig``` to regist the  command about FS. 
About ```AT+SYSFLASH```, now we just support the partition which type is 0x40.

mxwang101, 2018-01-27:
Hi，Xcguang，

 

After enable it in make menuconfig, AT+FS Works fine. Under AT version:1.0.b1.0(1e9060d - Aug 25 2017 10:37:43)

 

But, when I reload test ver. AT version:1.0.b1.0(1e9060d - Jan 22 2018 17:39:07)， It is not work again。 

 

Could you please help me make a new esp-at.bin,  with both fix AT+FS, and fix +NOTIFY:0,3,2,20, problem.

 

Thanks,

 

 

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 xcguang
发送时间: 2018年1月27日 11:59
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] Filesystem Operations, AT+FS, Not Work (#67)

 

Hi @mxwang101 <https://github.com/mxwang101>  ，
By default, fs is not supported, and you have to enable it in make menuconfig to regist the command about FS.
About AT+SYSFLASH, now we just support the partition which type is 0x40.

mxwang101, 2018-01-29:
xiewenxiang: work fine! thanks

 

发件人: bounces+848413-3139-mwang_56=163.com@sgmail.github.com [mailto:bounces+848413-3139-mwang_56=163.com@sgmail.github.com] 代表 xiewenxiang
发送时间: 2018年1月29日 13:31
收件人: espressif/esp32-at
抄送: mxwang101; Mention
主题: Re: [espressif/esp32-at] Filesystem Operations, AT+FS, Not Work (#67)

 

@mxwang101 <https://github.com/mxwang101> 
esp-at.tar.gz <https://github.com/espressif/esp32-at/files/1672374/esp-at.tar.gz>

mxwang101, 2018-02-01:
OK, Thanks, It is work now

closed 2018-02-02 by xcguang