espressif/esp-at #68: AT+WAKEUPGPIO not available

Cheong2K, 2018-01-30:
ESP8266 has this AT command for host to wake up the Wi-Fi chip, but it seems that ESP32-AT doesn't support this, how to deal with power down for both host and ESP32?

xcguang, 2018-02-01:
Hi @Cheong2K ,
AT+WAKEUPGPIO is to wakeup from light-sleep, and esp32-at doesn't support light sleep now.

Cheong2K, 2018-02-01:
Thank you. How do I deal with the power management then? Any document?

xcguang, 2018-02-01:
Now esp32-at supports modem sleep and deep sleep.  You can choose one mode of them.

FayeY, 2018-02-09:
Hi, could this [Sleep Modes](https://esp-idf.readthedocs.io/en/latest/api-reference/system/sleep_modes.html#entering-deep-sleep) document help?

Cheong2K, 2018-02-10:
I am using the AT firmware, I assume that I do not need to recompile the AT firmware, would you consider to add that AT command to the ESP32-AT?

xcguang, 2018-02-11:
Hi, 
Which sleep mode do you use? Modem sleep needn't wakeup.
Now  esp32-at doesn't support wakeup from deep-sleep, and we will add it as soon as possible.

Cheong2K, 2018-02-12:
What I want is: in my design, I have a host MCU to control the ESP32 via UART AT commands, the Host MCU sleeps all the time, so if something goes to UART from the ESP32, the Host MCU must be awaken before the ESP32 can send data.

I am not sure I can send data to the ESP32 if it is in modem sleep or deep sleep modes, or I also need to use a GPIO to wake the ESP32 up first?

xcguang, 2018-02-12:
If in modem sleep mode, you can send data and needn't wake the ESP32, but you must do it if in deep sleep mode.
Now we don't support to wake the host mcu in official bin, and is it possible to wake the host mcu  by UART?

xcguang, 2018-06-19:
Hi，
How about this issue? I will close it, and you can reopen it if you have any question.Thanks.

closed 2018-06-19 by xcguang