espressif/esp-at #69: Advertising data with BLE not working with Android

jeremydg71, 2018-01-31:
Having issues when trying to discover ESP32 device running AT program using Android device. My issues only come up when I set the advertising data. When advertising data is set, I can no longer discover the device. It does not show up in my scan list.  Otherwise, if I do not set the advertising data the ESP32 is discoverable and I can connect just fine. I have demonstrated this behavior using the ESP32devKitC.

I have tested on multiple android devices, running android OS v8.1.0

I have tried several bluetooth apps including nRF Connect. I have also written my own tester Android app with the same result as below:

When not using advertising data for the ESP32 in server mode, this sequence of commands works and can discover and then connect to device after the ESP32 device has begun advertising:
AT+BLEINIT=2
AT+BLEADVSTART

When I set the advertising data, I am no longer able to discover the device. It does not show up in scan list for any BLE android test apps or my own android test application.
AT+BLEINIT=2
AT+BLEADVDATA="1122334455"
AT+BLEADVSTART

I have tested these commands of communicating and connecting between two ESP32 devices without issue. Two ESP32devKitC s, one being a server, the other being a client works okay. The server shows up in the scan of the client with the appropriate advertising data.

xiewenxiang, 2018-02-01:
@jeremydg71 
This is because the android phones blocks the BLE package. In fact, the ESP32 device is broadcasting. You can use a BLE sniffer to verify this.
I think, maybe the broadcast package accepted by android phones must meet certain rules.
You can try this adv data
 `AT+BLEADVDATA="0201061aff4c000215fda50693a4e24fb1afcfc6eb0764782527b7f206c5"`

Please refer to the documentation `CSS v7.pdf`
[CSS v7.pdf](https://github.com/espressif/esp32-at/files/1684396/CSS.v7.pdf)

jeremydg71, 2018-02-01:
Thank you for help! Still new to bluetooth and BLE. We're integrating this chip into our product so this help is much appreciated. Didn't realize there was a specific format in the BLE specification for how things should be structured in the advertising packet payload. I was thinking whatever was in the advertising data would show up as the name when scanning/discovering devices.

For others information:

Any advertising packet payload should follow the BLE specification to be useful. This is not an Android specific thing, but a universal part of the specification.

The advertising payload is basically just a series of advertising structures:
[Length (1byte)][Type (1 byte)][ data (N bytes)]

Length specifies length of the type byte + data, so N+1.

For example, using a single advertising structure to display a unique name "test" with advertising data:
Type of 0x09 indicates that this advertising structure corresponds to the full name of device. See specifications for list of other types.

"test" equals "74 65 73 74" in hex, so:

[05][09][74 65 73 74]
 or just, 050974657374
AT+BLEADVDATA="050974657374"

Specifically for android, broadcast advertising packets are rejected if the payload is not well formatted. If the length byte does not correctly indicate the number of bytes that follow, it will be rejected. You can even have an incorrect type field and it will still pass through filters.

xcguang, 2018-06-30:
Hi @jeremydg71 ,
I will close this issue. Please create the new issue, if have any question.

closed 2018-06-30 by xcguang