espressif/esp-at #70: Notify for BLE Characteristic will not pass through complete data string with 0x0

bedbugboxes, 2018-02-14:
When a BLE characteristic notification is triggered, the ESP sends a "+NOTIFY:0,4,7,6,XXXXXX".  
-- 0 is the device
-- 4 is the Service Index
-- 7 is the Characteristic Index of the 
-- 6 is the length of the string received in the characteristic
-- XXXXXX is the string received

If there's a 0x0 in the data string, then the UART terminates the message.  It looks like the UART routine uses a standard len to determine how many bytes to transmit out.

In my particular case, I have the ESP32 setup as a BLE client and communicating with a Nordic UART Service (NUS), the notify characteristic is set when a response is received.  The Tx protocol to query the server device is "0x41 <address> 0x0 0x0 0x0 <crc>".  The expected response is six bytes "0x41 <address> <data1> <data2> <data3> <crc>"  After the message is sent, the notification is received but if any field contains a zero, the received transmission is truncated and doesn't display.

xiewenxiang, 2018-02-22:
Please update the esp-at to the latest commit.

bedbugboxes, 2018-02-22:
Thank you for the quick update.  I will check it out and hopefully close the issue next week.

bedbugboxes, 2018-03-02:
Issue is fixed in the latest commit - Thank You!

closed 2018-03-02 by bedbugboxes