espressif/esp-at #72: Any information when client connected to ESP32 AP?

zavovi, 2018-02-20:
Hi all,
I am working with ESP32-wroom module and I tried make WiFi AP with ESP32. I see the WiFi in my smartphone, I can connect to this AP, but I didn't see any text on UART, that client (smartphone) is connected or something like that. 

Is it possible allow something like this?

Thank you

CoryFF, 2018-03-10:
I think what you need to do is send the AT+CWLIF command. That will list out which client IPs are connected.

schwac, 2018-03-10:
You need to send AT+CWLIF periodically. That will list out the connected clients IPs.

xcguang, 2018-03-12:
Hi ,
We will add it ASAP.

zavovi, 2018-03-13:
Hi, thank you for all answers, if I want use AT+CWLIF, it should be send in poll and it is wasting time. But now, it may be solution. 

@xcguang, I will wait for your solution :+1: Thank you

MaJerle, 2018-03-13:
Should be solved the same way as in ESP8266

xcguang, 2018-03-13:
@MaJerle Yes, we will add the same message as ESP8266.

closed 2020-03-18 by igrr