espressif/esp-at #73: Overlap with at_customize.bin

kealist, 2018-02-26:
```
Flashing binaries to serial port COM6 (app at offset 0x10000)...
usage: esptool write_flash [-h] [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x20000 for file: C:/msys32/home/kealist/esp32-at/build/at_customize.bin
make: *** [/home/kealist/esp32-at/esp-idf/components/esptool_py/Makefile.projbuild:55: flash] Error 2
```

With the default settings + bluetooth  removing customized AT, I can flash successfully, but I get the above error when I try to make flash with customized AT in the build.  Is their a "force" mode I am missing?  I didn't see anything in the documentation

kealist, 2018-02-26:
I have tried the following without success:

1. Enabling custom partition using menuconfig and renaming partitions_at.csv to partitions.csv (same error)
2.  Changed the NVS partition size from 56K to E000 in the partitions.csv file; changed it to DFFF (same error both ways)

xcguang, 2018-03-12:
Hi，
The log is ```esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x20000 for file: C:/msys32/home/kealist/esp32-at/build/at_customize.bin
make: *** [/home/kealist/esp32-at/esp-idf/components/esptool_py/Makefile.projbuild:55: flash] Error 2``` .  
Please check the partition size and offset.

xcguang, 2018-03-24:
Hi @kealist .
Is this issue resolved?

kealist, 2018-03-25:
As I said in my last post, I got it working.  

I assume the default settings are still broken though.

pieterpotgieter, 2018-05-16:
HI Whatdid you do to get it working - I am having the same issue?

Wihan, 2018-06-09:
@pieterpotgieter Did you ever manage to get it working, we are also trying to get this running, having the same problem?

kealist, 2018-06-09:
@pieterpotgieter @Wihan  Looking at this again after doing some more custom ESP32 firmware.  I think that the problem may be that (starting from a the cloned repo) you need to run `make menuconfig` and then choose load and choose `sdkconfig.defaults` and then save it.  Otherwise it will probably use the wrong partition config.  It's a guess, but I think that could be the issue.

comcomservices, 2018-06-20:
I experienced this same issue and it is the automatically generated parameters that the esptool.py is given to flash to mcu which in the latest IDF release seems to get confused when multiple OTA definitions are used but no factory image partition 

as an example we had the following output from the make print_flash_cmd command

make print_flash_cmd 
--flash_mode dio --flash_freq 40m --flash_size detect 0x1000 bootloader/bootloader.bin 0x0 main.bin 0x8000 partitions.bin

you notice it's trying to flash the main.bin image to 0x0, this would overwrite the bootloader at 0x1000.  I manually modified the paramaters and executed directly with pytool to write to the image location which in my case is 0x10000 for OTA_0 partition where I want to write the app too..

See my partitions.csv file below

# Name,         Type    SubType,Offset,         Size
nvs,            data,   nvs,    0x9000,         16K,
otadata,        data,   ota,    0xd000,         8K,
phy_init,       data,   phy,    0xf000,         4K,
**ota_0,          app,    ota_0,  0x10000,        3M,**
ota_1,          app,    ota_1,  0x31000,        3M,
storage,        data,   ,       0x61000,        1500K,

xcguang, 2018-06-21:
Hi,
The offset of `ota_1` and `storage` is incorrect.
ota_0, app, ota_0, 0x10000, 3M,
ota_1, app, ota_1, **0x310000**, 3M,
storage, data, , **0x610000,** 1500K,

xcguang, 2018-06-21:
1MB=1024*1024 Byte=1048576 Byte = 0x100000 Byte, and 3MB=0x300000.

`ota_0, app, ota_0, 0x10000, 3M,`
So the offset of ota_1 is 0x10000+3M, and it is 0x310000.

comcomservices, 2018-06-21:
@xcguang Right! Thanks for pointing that out..  you got some skills man to spot that! It got messed up while coping it somehow..

I have the following in my file

# Name,		Type	SubType,Offset,		Size
nvs,		data,	nvs,	0x9000,		16K,
otadata,	data,	ota,	0xd000,		8K,
phy_init,	data,	phy,	0xf000,		4K,
ota_0,		app,	ota_0,	0x10000,	3M,
ota_1,		app,	ota_1,	0x310000,	3M,
storage,	data,	,	0x610000,	1500K,

But still get the error when flashing with make flash

esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x1000 for file: bootloader.bin

Any ideas on why the make flash isn't writing the image to the correct location?

xcguang, 2018-06-21:
Hi,
I can't reproduce it. 
Could you provide your sdkconfig and the commit id of IDF and esp-at which you use?

comcomservices, 2018-06-21:
4b91c82cc447640e5b61407e810f1d6f3eabd233 (Just pulled from master 5m ago)

[sdkconfig.txt](https://github.com/espressif/esp32-at/files/2121873/sdkconfig.txt)

xcguang, 2018-06-30:
@kealist I think I misunderstood you at first, and sorry for that.

@kealist @comcomservices @pieterpotgieter @Wihan 
Please try it again using the latest AT project. If the problem persists, please just feel free to let me know.

comcomservices, 2018-07-03:
@xcguang I can confirm that this issue is resolved with the changes you made in the latest commit

qt1, 2018-08-05:
Similar issue here, I think it started after update to esp-idf 
Partition and flash used to work.

According to the messages it seems like the program wants to write some code to 0xe000 while that address space should be in the middle  of otadata

please advise

*partition table:
```
#default.csv from arduino config
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x4000,
otadata,  data, ota,     0xd000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x140000,
app1,     app,  ota_1,   0x150000,0x140000,
eeprom,   data, 0x99,    0x290000,0x1000,
spiffs,   data, spiffs,  0x291000,0x16F000,
```
*error
```
$ make flash
Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000 )...
usage: esptool write_flash [-h] [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0xe000 for file: /home/baruch/esp/rmt_sender/components/arduino/tools/partitions/boot_app0.bin
/home/baruch/esp/esp-idf/components/esptool_py/Makefile.projbuild:62: recipe for target 'flash' failed
make: *** [flash] Error 2
```
*additional info
```
$ make print_flash_cmd
--flash_mode dio --flash_freq 40m --flash_size detect 0xe000 ota_data_initial.bin 0xe000 /home/baruch/esp/rmt_sender/components/arduino/tools/partitions/boot_app0.bin 0x1000 bootloader/bootloader.bin 0x10000 rmt_sender.bin 0x8000 partitions.bin
```

```
$ make
To flash all build output, run 'make flash' or:
python /home/baruch/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0xe000 /home/baruch/esp/rmt_sender/build/ota_data_initial.bin 0xe000 /home/baruch/esp/rmt_sender/components/arduino/tools/partitions/boot_app0.bin 0x1000 /home/baruch/esp/rmt_sender/build/bootloader/bootloader.bin 0x10000 /home/baruch/esp/rmt_sender/build/rmt_sender.bin 0x8000 /home/baruch/esp/rmt_sender/build/partitions.bin
```

xcguang, 2018-08-07:
Hi @qt1 ,
The information is `0xe000 /home/baruch/esp/rmt_sender/build/ota_data_initial.bin 0xe000 /home/baruch/esp/rmt_sender/components/arduino/tools/partitions/boot_app0.bin`, but I can't found them in partition table you provided. Are they defined in your project? It seems these partitions are overlap with `otadata`, please change them.

closed 2018-09-29 by xcguang