espressif/esp-at #74: AT+BLEGATTSSETATTR takes approximately 1 second to process

kealist, 2018-03-02:
I'm not sure if there is a way to optimize processing, but it takes a long time for the AT commands image to process the data from `AT+BLEGATTSSETATTR`.

```
AT+BLEGATTSSETATTR=1,12,,4
>
0xFF FF FF FF (wait one sec)
OK
```

Is there a way to improve this? It seems abnormally long...

xiewenxiang, 2018-03-05:
@kealist Thanks for your suggestion, We will check it.

kealist, 2018-03-07:
[image]

Here is a diagram from the logic analyzer showing time for reference.   If I need to update 7 characteristics (each 4 bytes), this takes approximately 7 seconds.

xiewenxiang, 2019-01-22:
We have fixed it, does this problem still exist?

closed 2019-01-23 by FayeY